openwrt/openwrt #18011: Upnp record duplicates after update from 23.05.5 to 24.10

abubaca4, 2025-02-17:
### Describe the bug

In log error like `miniupnpd get_timestamp no entry found`

[image]

And one more error report https://community.wd.com/t/why-my-cloud-home-make-so-much-upnp-forwarding-up-to-115/473182

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-18:
Last 3 patces are the difference, helps to report in proper repo andpoint out you  using iptable or nftable kind of daemon, tagging maintainers and guilty committers if you know which pr is at fault.

https://github.com/openwrt/packages/commits/openwrt-24.10/net/miniupnpd

howto (you need a linux installation, like in a vm)
https://openwrt.org/docs/guide-developer/toolchain/single.package

brada4, 2025-02-20:
You can download package makefile and patches from github, on top of sdk.

abubaca4, 2025-02-22:
I use nftables version
I compile
4c934aea718ceb682551b9bab0a8618e4b12920d (miniupnpd-nftables_2.3.3-r2_aarch64_cortex-a53, 23.05.5 last version). Has no problem
c9a170578ef70d216900d4bb680000d62b2c88f1 (miniupnpd-nftables_2.3.6-r1_aarch64_cortex-a53). Has problem
7058ca424add74534248bdf197722ea02be3d3b7 (miniupnpd-nftables_2.3.7-r1_aarch64_cortex-a53). Has problem
So problem started after https://github.com/openwrt/packages/commit/c9a170578ef70d216900d4bb680000d62b2c88f1

My upnp clients zerotier 1.14.2 and µTorrent 2.2.1(if need for test)

It's take a lot of time
How I do it
1) Download dir with miniupnpd Makefile from commit using https://download-directory.github.io/
2) git clone https://git.openwrt.org/openwrt/openwrt.git
3) cd openwrt
4) git pull
5) git checkout v24.10.0
5.5) install gawk and ncurses-dev
6) ./scripts/feeds update -a
7) ./scripts/feeds install -a
8) make menuconfig select platforme and select miniupnpd in net firewall
9) make tools/install
10) make toolchain/install
11) make toolchain/compile
12) make target/compile
(has [error](https://github.com/openwrt/openwrt/issues/10254) without 11 and 12)
13) replace miniupnpd makefile and patches file with dir content from 1
14) make package/miniupnpd/clear
15) make package/miniupnpd/download
16) make package/miniupnpd/prepare
17) make package/miniupnpd/compile
18) copy result ipk(file removed in next build) and repeat 1,13-17 for other commits

This sequence of actions was obtained by trial and error and took a long time to complete

brada4, 2025-02-22:
@neheb @self hosting-group may look at this, should have been faster if you reported in correct repo tagging authors.

Neustradamus, 2025-02-23:
To follow this ticket.

abubaca4, 2025-03-03:
Look like has no problem

closed 2025-12-24 by ldir-EDB0

abubaca4, 2025-12-24:
> I am not sure, it has been fixed?

A newer version should fix the issue. But it is not yet available at https://mirror-03.infra.openwrt.org/releases/24.10.5/packages/aarch64_cortex-a53/packages/. As far as I understand, the packages are built at certain intervals, and after the next build, the updated version should appear.