openwrt/openwrt #18019: Zerotier segmentation fault on startup.

looselyrigorous, 2025-02-18:
### Describe the bug

The Zerotier service crashes on startup.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R6S

### Image kind

Official downloaded image

### Steps to reproduce

1. I created the image using the imagebuilder. The command I used was:
```
make image ROOTFS_PARTSIZE="512" PROFILE="friendlyarm_nanopi-r6s" PACKAGES="parted losetup resize2fs irqbalance luci-app-irqbalance luci-proto-wireguard iperf3 tcpdump drill htop luci luci-ssl luci-app-uhttpd luci-app-wol luci-app-sqm luci-app-ddns ddns-scripts ddns-scripts-cloudflare ca-certificates acme-acmesh acme-acmesh-dnsapi luci-app-acme zerotier dockerd docker luci-app-dockerman
```
2. Flashed the image to my device (NanoPi R6S).

3. I then created a minimal 1 network configuration:

`/etc/config/zerotier`:
``` 
config zerotier 'global'
        option enabled '1'
        option secret 'REDACTED'

config network
        option id 'REDACTED'
```

4. And finally started zerotier:

```
root@speedtail:~# zerotier-one
```

### Actual behaviour

Zerotier tries to initialize but segfaults after a few seconds.
```
root@speedtail:/tmp# zerotier-one
Starting Control Plane...
Starting V6 Control Plane...
Segmentation fault
```

### Expected behaviour

Zerotier should start and work normally.

### Additional info

As a low-effort test I thought I'd try using the `23.05.5` binaries, and to my surprise they worked as expected.

I uninstalled the packages pre-installed in the image:
```
opkg remove --force-depends libminiupnpc zerotier
```

Downloaded the older versions:
```
cd /tmp
curl -LO https://downloads.openwrt.org/releases/23.05.5/packages/aarch64_generic/packages/libminiupnpc_2.2.3-1_aarch64_generic.ipk
curl -LO https://downloads.openwrt.org/releases/23.05.5/packages/aarch64_generic/packages/zerotier_1.14.1-1_aarch64_generic.ipk
```

And then installed them:
```
opkg install zerotier_1.14.1-1_aarch64_generic.ipk
opkg install libminiupnpc_2.2.3-1_aarch64_generic.ipk
```

Afterwards, launching either `zerotier-one` directly or using the initscript worked as intended.

The only difference I can immediately notice is that the `libminiupnpc` dependency is an older version, `2.2.3` on `23.05.5` versus `2.2.8` on `24.10.0`

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

pcislocked, 2025-02-18:
Yep, can confirm the same

```Wed Feb 19 01:40:02 2025 daemon.err zerotier-one[15537]: Starting Control Plane...
Wed Feb 19 01:40:02 2025 daemon.err zerotier-one[15537]: Starting V6 Control Plane...
Wed Feb 19 01:40:02 2025 daemon.notice netifd: Interface 'zerotier' is enabled
Wed Feb 19 01:40:02 2025 daemon.notice netifd: Network device 'zt44xlzd3z' link is up
Wed Feb 19 01:40:02 2025 daemon.notice netifd: Interface 'zerotier' has link connectivity
Wed Feb 19 01:40:02 2025 daemon.notice netifd: Interface 'zerotier' is setting up now
Wed Feb 19 01:40:02 2025 daemon.notice netifd: Interface 'zerotier' is now up
Wed Feb 19 01:40:05 2025 user.notice mwan3-hotplug[15580]: mwan3 hotplug on zerotier not called because interface disabled
Wed Feb 19 01:40:05 2025 user.notice firewall: Reloading firewall due to ifup of zerotier (zt44xlzd3z)
Wed Feb 19 01:40:08 2025 kern.info kernel: [  475.494808] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
Wed Feb 19 01:40:08 2025 kern.info kernel: [  475.504039] epc = 77e468a8 in libc.so[77dc0000+b2000]
Wed Feb 19 01:40:08 2025 kern.info kernel: [  475.509337] ra  = 77e47288 in libc.so[77dc0000+b2000]
Wed Feb 19 01:40:08 2025 daemon.notice netifd: Network device 'zt44xlzd3z' link is down
Wed Feb 19 01:40:08 2025 daemon.notice netifd: Interface 'zerotier' has link connectivity loss
Wed Feb 19 01:40:08 2025 daemon.notice netifd: Interface 'zerotier' is now down
Wed Feb 19 01:40:08 2025 daemon.info procd: Instance zerotier::instance1 s in a crash loop 6 crashes, 6 seconds since last crash
```

Ilya87, 2025-02-23:
I also can confirm segfault in OpenWrt 24.10 on Asus RT-N14U.

eurodrew555, 2025-02-24:
I've got another workaround for this issue which I'm using on my OpenWRT Wireless APs.
The workaround I'm using is to disable port mapping (which I don't need on my WAPs)

First create a new json conf file /etc/config/zerotier.conf

#### /etc/config/zerotier.conf

```
{ "settings": { "portMappingEnabled": false }}
```

Now reconfigure ZeroTier to use this local configuration file:

```
uci set zerotier.global.local_conf_path='/etc/config/zerotier.conf'
uci commit zerotier
```

Now restart ZeroTier and the issue should be resolved.

```
service zerotier enable
service zerotier restart
```

miyatsu, 2025-02-25:
Looks like miniupnp's API change cause this problem. See:
(1). https://github.com/zerotier/ZeroTierOne/issues/2390
(2). https://github.com/zerotier/ZeroTierOne/issues/2332

rabbitrogi, 2025-02-25:
In 23.05.5, miniupnpc is 2.2.3, but in 24.10.0, miniupnpc is 2.2.8.

In miniupnpc-2.2.3/include/miniupnpc.h, line 119 to 123 is:
```
MINIUPNP_LIBSPEC int
UPNP_GetValidIGD(struct UPNPDev * devlist,
                 struct UPNPUrls * urls,
                                 struct IGDdatas * data,
                                 char * lanaddr, int lanaddrlen);
```

In miniupnpc-2.2.8/include/miniupnpc.h, line 121 to 126 is:
```
MINIUPNP_LIBSPEC int
UPNP_GetValidIGD(struct UPNPDev * devlist,
                 struct UPNPUrls * urls,
                 struct IGDdatas * data,
                 char * lanaddr, int lanaddrlen,
                 char * wanaddr, int wanaddrlen);
```

But in ZeroTierOne-1.14.1/osdep/PortMapper.cpp, line 233 is:
`                                        if ((foundValidIGD = UPNP_GetValidIGD(devlist,&urls,&data,lanaddr,sizeof(lanaddr)))&&(lanaddr[0])) {
`

So we need to define a new 'wanaddr' to fix this problem.

miyatsu, 2025-02-26:
ping @mwarning

mwarning, 2025-02-26:
I will try to add and upstream a patch.

sppmasterspp, 2025-03-02:
Are you guys compiling with GCC14. I have a segfault when I compile for ipq806x (R7800) but it works OK with ipq807x.

mwarning, 2025-03-03:
Zerotier compiled for OpenWrt 24.10 (6df0e3d02a) works on my device (ramips/mt7620).
A test with the imagebuilder still needs to be done. 

Nevertheless, I have made a MR: https://github.com/openwrt/packages/pull/26088

sppmasterspp, 2025-03-06:
> Nevertheless, I have made a MR: [openwrt/packages#26088](https://github.com/openwrt/packages/pull/26088)
I still get segmentation fault at start. 
I compile for ipq806x (R7800) using main OpenWrt. How can I debug this? 
My experience compiling for 23.05 is that if I changed the OpenWrt menuconfig default settings, zerotier would give segfault at start. When I used the default settings zerotier run OK.
Especially these settings - GCC14 (instead of default 13), compile with PIE, 

```
CONFIG_TARGET_ipq806x=y
CONFIG_TARGET_ipq806x_generic=y
CONFIG_TARGET_ipq806x_generic_DEVICE_netgear_r7800=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_ATH10K_LEDS=y
CONFIG_ATH10K_THERMAL=y
# CONFIG_BINUTILS_USE_VERSION_2_42 is not set
CONFIG_BINUTILS_USE_VERSION_2_43=y
CONFIG_BINUTILS_VERSION="2.43.1"
CONFIG_BINUTILS_VERSION_2_43=y
CONFIG_BUILD_PATENTED=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_EXPERIMENTAL=y
**CONFIG_GCC_DEFAULT_PIE=y**
**CONFIG_GCC_USE_GRAPHITE=y**
# CONFIG_GCC_USE_VERSION_13 is not set
CONFIG_GCC_USE_VERSION_14=y
CONFIG_GCC_VERSION="14.2.0"
CONFIG_GCC_VERSION_14=y
# CONFIG_GDB is not set
CONFIG_IMAGEOPT=y
CONFIG_INCLUDE_CONFIG=y
CONFIG_KERNEL_ARM_PMU=y
CONFIG_KERNEL_ARM_PMUV3=y
CONFIG_KERNEL_DYNAMIC_DEBUG=y
CONFIG_KERNEL_IO_URING=y
CONFIG_KERNEL_PERF_EVENTS=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_THREADED_RESOLVER=y
CONFIG_LIBCURL_ZLIB=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_TAR_BZIP2=y
CONFIG_PACKAGE_TAR_GZIP=y
CONFIG_PACKAGE_TAR_POSIX_ACL=y
CONFIG_PACKAGE_TAR_XATTR=y
CONFIG_PACKAGE_TAR_XZ=y
CONFIG_PACKAGE_TAR_ZSTD=y
# CONFIG_PACKAGE_apk-mbedtls is not set
CONFIG_PACKAGE_apk-openssl=y
CONFIG_PACKAGE_ath10k-firmware-qca9984=y
# CONFIG_PACKAGE_ath10k-firmware-qca9984-ct is not set
CONFIG_PACKAGE_zerotier=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_PREINITOPT=y
CONFIG_TARGET_OPTIMIZATION="-O2 -pipe -mfpu=neon-vfpv4 -mtune=cortex-a15 -ffunction-sections -fdata-sections -mfloat-abi=hard"
CONFIG_TARGET_OPTIONS=y
# CONFIG_TARGET_PREINIT_SUPPRESS_STDERR is not set
CONFIG_TARGET_PREINIT_TIMEOUT=5
CONFIG_USE_GC_SECTIONS=y
CONFIG_USE_LTO=y
CONFIG_USE_MOLD=y
CONFIG_WIFI_SCRIPTS_UCODE=y
# CONFIG_WPA_MBO_SUPPORT is not set
CONFIG_ZSTD_OPTIMIZE_O3=y
CONFIG_KERNEL_WERROR=y
```

miyatsu, 2025-03-11:
> Nevertheless, I have made a MR: [openwrt/packages#26088](https://github.com/openwrt/packages/pull/26088)
Will this be merged/backport before next `24.10.1` release?

miyatsu, 2025-03-15:
> Zerotier compiled for OpenWrt 24.10 ([6df0e3d](https://github.com/openwrt/openwrt/commit/6df0e3d02ac89d3ad7da1ade7fc51b15163bd136)) works on my device (ramips/mt7620).

ImmortalWrt 24.10.0 release (https://github.com/immortalwrt/immortalwrt/tree/v24.10.0) works on my ramips/mt7621 device (D-Team Newifi D2). Configuring via `luci-app-zerotier` ONLY, no extra workaround at all, work as expected. OpenWrt however, I've never tested.

My test env:

Router A (ImmortalWrt 24.10.0, mt7621) have a private IPv4 wan address (NAT4),  router B (ImmortalWrt 23.05.1, x86) also have a private IPv4 address (NAT1). Those two routers are approximately 350km (217mi) away from each other. Ping from A to B works as expected, vice versa.

P.S. First 1-2 mins have large rtt (around 300ms, must be forward by zerotier global moon server), then I've got normal rtt (around 12ms, best guss is zerotier got UDP hole punching success, both router can talk to each other directly).

sppmasterspp, 2025-03-31:
@mwarning 
Please look at my findings bellow.
Currently when I compile a main OpenWrt build for ipq806x (R7800), if I use `GCC_DEFAULT_PIE` and/or `USE_MOLD` in menuconfig it always results in Segmentation fault at zerotier startup.
Can you look at this issue?
Using `GCC_DEFAULT_PIE` and/or `USE_MOLD` when I compile main OpenWrt for ipq807x doesn't result in that error and zerotier runs OK.

mwarning, 2025-03-31:
@sppmasterspp interesting, but I do not have enough spare time for testing right now.

mwarning, 2025-04-10:
The MR has been merged to master. But I was not able to confirm if it works. @looselyrigorous can you check if it is fixed?

looselyrigorous, 2025-04-11:
It's working correctly now, on openwrt master.

mwarning, 2025-04-17:
OK, then let's close this as fixed unless someone proves the opposite.

Ilya87, 2025-05-15:
@mwarning 
ASUS RT-N14U Openwrt 24.10.1. Pinging from Windows - ping loss started from 172 packets. zerotier 1.14.1-r2. Error still here.
```
[  294.746430] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  294.764232] epc = 77e27830 in libc.so[77da1000+b2000]
[  294.774583] ra  = 77e28210 in libc.so[77da1000+b2000]
[  304.588696] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  304.606695] epc = 77e98830 in libc.so[77e12000+b2000]
[  304.616839] ra  = 77e99210 in libc.so[77e12000+b2000]
[  314.254300] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  314.272276] epc = 77e15830 in libc.so[77d8f000+b2000]
[  314.282460] ra  = 77e16210 in libc.so[77d8f000+b2000]
[  323.792005] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  323.809800] epc = 77ddb830 in libc.so[77d55000+b2000]
[  323.819957] ra  = 77ddc210 in libc.so[77d55000+b2000]
[  332.853486] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  332.871305] epc = 77e58830 in libc.so[77dd2000+b2000]
[  332.881434] ra  = 77e59210 in libc.so[77dd2000+b2000]
[  342.016545] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  342.034517] epc = 77e91830 in libc.so[77e0b000+b2000]
[  342.044657] ra  = 77e92210 in libc.so[77e0b000+b2000]
```

pcislocked, 2025-05-15:
can confirm, stable on 24.10

Ilya87, 2025-05-15:
I done clear config (without previous) to check:
```
[  395.484861] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  395.502669] epc = 77e8b830 in libc.so[77e05000+b2000]
[  395.512799] ra  = 77e8c210 in libc.so[77e05000+b2000]
[  404.421022] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  404.438981] epc = 77e22830 in libc.so[77d9c000+b2000]
[  404.449122] ra  = 77e23210 in libc.so[77d9c000+b2000]
[  412.746789] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  412.764594] epc = 77df9830 in libc.so[77d73000+b2000]
[  412.774722] ra  = 77dfa210 in libc.so[77d73000+b2000]
[  428.461635] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  428.479621] epc = 77ddd830 in libc.so[77d57000+b2000]
[  428.489772] ra  = 77dde210 in libc.so[77d57000+b2000]
[  436.764498] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  436.782277] epc = 77e62830 in libc.so[77ddc000+b2000]
[  436.792408] ra  = 77e63210 in libc.so[77ddc000+b2000]
[  445.391020] do_page_fault(): sending SIGSEGV to zerotier-one for invalid write access to 00000002
[  445.408812] epc = 77e0b830 in libc.so[77d85000+b2000]
[  445.418945] ra  = 77e0c210 in libc.so[77d85000+b2000]
```

Error still here.

looselyrigorous, 2025-05-16:
I recompiled openwrt master again and the issue is still there. Looks like I didn't test it properly on my end, sorry about that.

Ilya87, 2025-05-16:
@mwarning I think – this is the correct patch
https://gitlab.archlinux.org/archlinux/packaging/packages/zerotier-one/-/blob/main/0001-miniupnpc.patch?ref_type=heads

Ilya87, 2025-05-22:
Ping @mwarning

Ilya87, 2025-05-29:
> When the GCC of openwrt is 13, it is normal to run the zerotier-one command by making zerotier or apk add zerotier. When the GCC is 14, it is not possible to run after making zerotier or apk add zerotier.

I just now rebuild openwrt in Ubuntu with gcc 13. Result is the same.

xuanranran, 2025-05-29:
> > When the GCC of openwrt is 13, it is normal to run the zerotier-one command by making zerotier or apk add zerotier. When the GCC is 14, it is not possible to run after making zerotier or apk add zerotier.
> 
> I just now rebuild openwrt in Ubuntu with gcc 13. Result is the same.

armv8?

Ilya87, 2025-05-29:
> > > When the GCC of openwrt is 13, it is normal to run the zerotier-one command by making zerotier or apk add zerotier. When the GCC is 14, it is not possible to run after making zerotier or apk add zerotier.
> > 
> > 
> > I just now rebuild openwrt in Ubuntu with gcc 13. Result is the same.
> 
> armv8?

Asus RT-N14U, ramips mt7620

jjm2473, 2025-06-09:
This exception can be triggered when there is UPnP service in the network

jjm2473, 2025-06-10:
I'm pretty sure zerotier is using the wrong `miniupnpc.h`.

Add these line to `osdep/PortMapper.cpp:207` :
```cpp
#if MINIUPNPC_API_VERSION < 18
#define MINIUPNPC_API_VERSION 0
#error "MINIUPNPC_API_VERSION < 18"
#endif
```

then make, will product a error:
```
ccache aarch64-openwrt-linux-musl-g++ -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/mnt/data/build/istoreos-build/24.10/armsr/openwrt/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.14.1=ZeroTierOne-1.14.1 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wl,-z,noexecstack -ffunction-sections -fdata-sections  -I/mnt/data/build/istoreos-build/24.10/armsr/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/usr/include -I/mnt/data/build/istoreos-build/24.10/armsr/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include -I/mnt/data/build/istoreos-build/24.10/armsr/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include/fortify  -Wall -Wno-deprecated
-std=c++17 -pthread -Irustybits/target -isystem ext -Iext/prometheus-cpp-lite-1.0/core/include -Iext-prometheus-cpp-lite-1.0/3rdparty/http-client-lite/include -Iext/prometheus-cpp-lite-1.0/simpleapi/include -DNDEBUG  -DZT_USE_MINIUPNPC -DZT_USE_SYSTEM_MINIUPNPC -DZT_USE_SYSTEM_NATPMP -DZT_NO_TYPE_PUNNING -DZT_ARCH_ARM_HAS_NEON -march=armv8-a+crypto -mtune=generic -mstrict-align -DZT_BUILD_PLATFORM=1 -DZT_BUILD_ARCHITECTURE=4 -DZT_SOFTWARE_UPDATE_DEFAULT="\"disable\"" -D_MT_ALLOCATOR_H -D_POOL_ALLOCATOR_H -D_EXTPTR_ALLOCATOR_H -D_DEBUG_ALLOCATOR_H   -c -o osdep/PortMapper.o osdep/PortMapper.cpp
osdep/PortMapper.cpp:208: warning: "MINIUPNPC_API_VERSION" redefined
  208 | #define MINIUPNPC_API_VERSION 0
      |
In file included from osdep/PortMapper.cpp:47:
ext/miniupnpc/miniupnpc.h:23: note: this is the location of the previous definition
   23 | #define MINIUPNPC_API_VERSION   16
      |
osdep/PortMapper.cpp:209:2: error: #error "MINIUPNPC_API_VERSION < 18"
  209 | #error "MINIUPNPC_API_VERSION < 18"
      |  ^~~~~
make[3]: *** [<builtin>: osdep/PortMapper.o] Error 1

```

Even `-DZT_USE_SYSTEM_MINIUPNPC -DZT_USE_SYSTEM_NATPMP` is used, but `ext/miniupnpc/miniupnpc.h` still included, it is wrong!

Let's try modify `ext/miniupnpc/miniupnpc.h` to see where include it, add these line to `ext/miniupnpc/miniupnpc.h:24` : 
```
#ifdef ZT_USE_SYSTEM_MINIUPNPC
#error "should not include 'ext/miniupnpc/miniupnpc.h' on ZT_USE_SYSTEM_MINIUPNPC !!!"
#endif
```

Make again, see what happen:
```
ccache aarch64-openwrt-linux-musl-g++ -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/mnt/data/build/istoreos-build/24.10/armsr/openwrt/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.14.1=ZeroTierOne-1.14.1 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wl,-z,noexecstack -ffunction-sections -fdata-sections  -I/mnt/data/build/istoreos-build/24.10/armsr/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/usr/include -I/mnt/data/build/istoreos-build/24.10/armsr/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include -I/mnt/data/build/istoreos-build/24.10/armsr/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include/fortify  -Wall -Wno-deprecated -std=c++17 -pthread -Irustybits/target -isystem ext -Iext/prometheus-cpp-lite-1.0/core/include -Iext-prometheus-cpp-lite-1.0/3rdparty/http-client-lite/include -Iext/prometheus-cpp-lite-1.0/simpleapi/include -DNDEBUG  -DZT_USE_MINIUPNPC -DZT_USE_SYSTEM_MINIUPNPC -DZT_USE_SYSTEM_NATPMP -DZT_NO_TYPE_PUNNING -DZT_ARCH_ARM_HAS_NEON -march=armv8-a+crypto -mtune=generic -mstrict-align -DZT_BUILD_PLATFORM=1 -DZT_BUILD_ARCHITECTURE=4 -DZT_SOFTWARE_UPDATE_DEFAULT="\"disable\"" -D_MT_ALLOCATOR_H -D_POOL_ALLOCATOR_H -D_EXTPTR_ALLOCATOR_H -D_DEBUG_ALLOCATOR_H   -c -o osdep/PortMapper.o osdep/PortMapper.cpp
In file included from osdep/PortMapper.cpp:47:
ext/miniupnpc/miniupnpc.h:25:2: error: #error "should not include 'ext/miniupnpc/miniupnpc.h' on ZT_USE_SYSTEM_MINIUPNPC !!!"
   25 | #error "should not include 'ext/miniupnpc/miniupnpc.h' on ZT_USE_SYSTEM_MINIUPNPC !!!"
      |  ^~~~~
osdep/PortMapper.cpp:208: warning: "MINIUPNPC_API_VERSION" redefined
  208 | #define MINIUPNPC_API_VERSION 0
      | 
ext/miniupnpc/miniupnpc.h:23: note: this is the location of the previous definition
   23 | #define MINIUPNPC_API_VERSION   16
      | 
osdep/PortMapper.cpp:209:2: error: #error "MINIUPNPC_API_VERSION < 18"
  209 | #error "MINIUPNPC_API_VERSION < 18"
      |  ^~~~~
make[3]: *** [<builtin>: osdep/PortMapper.o] Error 1

```

it says `osdep/PortMapper.cpp:47` include `ext/miniupnpc/miniupnpc.h`, let's check `osdep/PortMapper.cpp`:
```bat
...
46 #ifdef ZT_USE_SYSTEM_MINIUPNPC
47 #include <miniupnpc/miniupnpc.h>
48 #include <miniupnpc/upnpcommands.h>
49 #else
...
```

so `#include <miniupnpc/miniupnpc.h>` actually include `ext/miniupnpc/miniupnpc.h`?

is g++ flags `-isystem ext` cause the problem? try to remove it.

I found that `-isystem ext` is defined at `make-linux.mk:12`, remove it and see what happen in make:
```
ccache aarch64-openwrt-linux-musl-g++ -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/mnt/data/build/istoreos-build/24.10/armsr/openwrt/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.14.1=ZeroTierOne-1.14.1 -ffunction-sections -fda [... cut ...]
make[3]: *** [<builtin>: controller/EmbeddedNetworkController.o] Error 1
```

It' works, `-isystem ext` is removed from g++ flags, but cause other errors.

add back `-isystem ext`, but prepend `-isystem $(STAGING_DIR)/usr/include`, so `make-linux.mk:12` looks like:
```
INCLUDES?=-Irustybits/target -isystem $(STAGING_DIR)/usr/include -isystem ext -Iext/prometheus-cpp-lite-1.0/core/include -Iext-prometheus-cpp-lite-1.0/3rdparty/http-client-lite/include -Iext/prometheus-cpp-lite-1.0/simpleapi/include
```

Make again, finally succeeded!!

jjm2473, 2025-06-10:
Does this mean that there is a bug in openwrt's build system? `$(STAGING_DIR)/usr/include` is not used as a system include path.

jjm2473, 2025-06-10:
I have PR'd the patch and have tested that zerotier does not crash anymore. https://github.com/openwrt/packages/pull/26730

Ilya87, 2025-06-13:
@looselyrigorous I also can confirm new patch work, issue is solved.

closed 2025-06-16 by thess