openwrt/openwrt #18021: Xiaomi AX3600 - Attended Sysupgrade - Unsupported target: ipq807x/generic

Code-TT, 2025-02-18:
I've tried to use Attended Sysupgrade on luci to upgrade openwrt from v23.05.5 to v24.10.0 without success. 
```

Error building the firmware image
Server response: Unsupported target: ipq807x/generic. The requested target was either dropped, is still being built or is not supported by the selected version. Please check the forums or try again later.
```

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "xiaomi,ax3600",
    "target": "ipq807x/generic",
    "version": "24.10.0",
    "packages": [
        "ath10k-firmware-qca9887-ct",
        "ath11k-firmware-ipq8074",
        "base-files",
        "batctl-tiny",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fping",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ipq-wifi-xiaomi_ax3600",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct-smallbuffers",
        "kmod-ath11k",
        "kmod-ath11k-ahb",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-michael-mic",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-fs-ext4",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc16",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-libphy",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-phy-aquantia",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-qca-nss-dp",
        "kmod-qca-ssdk",
        "kmod-qrtr",
        "kmod-qrtr-smd",
        "kmod-slhc",
        "kmod-thermal",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-xhci-hcd",
        "kmod-usb3",
        "libblkid",
        "libblobmsg-json",
        "libc",
        "libcomerr",
        "libext2fs",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libopenssl",
        "libopenssl-conf",
        "libopenssl-legacy",
        "libsmartcols",
        "libss",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "libuuid",
        "logd",
        "losetup",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-usteer",
        "luci-base",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-batman-adv",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "nano-full",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-openssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-25.037.68351-2ca3f19"
}
```

pcislocked, 2025-02-18:
the firmware-selector website generates it just fine tho. huh. weird.

as a temporary workaround; 
just go here: https://firmware-selector.openwrt.org/?version=24.10.0&target=qualcommax%2Fipq807x&id=xiaomi_ax3600

(language selector is on top right corner)

under "Customize installed packages and/or first boot script", just paste below to the packages (took your json out and put the packages there, but check them again yourself as well)

`ath11k-firmware-ipq8074 base-files ca-bundle dnsmasq dropbear e2fsprogs firewall4 fstools kmod-ath11k-ahb kmod-fs-ext4 kmod-gpio-button-hotplug kmod-leds-gpio kmod-nft-offload kmod-phy-aquantia kmod-qca-nss-dp kmod-usb-dwc3 kmod-usb-dwc3-qcom kmod-usb3 libc libgcc libustream-mbedtls logd losetup mtd netifd nftables odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe procd-ujail uboot-envtools uci uclient-fetch urandom-seed urngd ipq-wifi-xiaomi_ax3600 kmod-ath10k-ct-smallbuffers ath10k-firmware-qca9887-ct luci ath10k-firmware-qca9887-ct ath11k-firmware-ipq8074 base-files batctl-tiny busybox ca-bundle cgi-io dnsmasq dropbear e2fsprogs firewall4 fping fstools fwtool getrandom hostapd-common ipq-wifi-xiaomi_ax3600 iw iwinfo jansson jshn jsonfilter kernel kmod-ath kmod-ath10k-ct-smallbuffers kmod-ath11k kmod-ath11k-ahb kmod-cfg80211 kmod-crypto-acompress kmod-crypto-aead kmod-crypto-ccm kmod-crypto-cmac kmod-crypto-crc32c kmod-crypto-ctr kmod-crypto-gcm kmod-crypto-gf128 kmod-crypto-ghash kmod-crypto-hash kmod-crypto-hmac kmod-crypto-manager kmod-crypto-michael-mic kmod-crypto-null kmod-crypto-rng kmod-crypto-seqiv kmod-crypto-sha512 kmod-fs-ext4 kmod-gpio-button-hotplug kmod-hwmon-core kmod-leds-gpio kmod-lib-crc-ccitt kmod-lib-crc16 kmod-lib-crc32c kmod-lib-lzo kmod-libphy kmod-mac80211 kmod-nf-conntrack kmod-nf-conntrack6 kmod-nf-flow kmod-nf-log kmod-nf-log6 kmod-nf-nat kmod-nf-reject kmod-nf-reject6 kmod-nfnetlink kmod-nft-core kmod-nft-fib kmod-nft-nat kmod-nft-offload kmod-nls-base kmod-phy-aquantia kmod-ppp kmod-pppoe kmod-pppox kmod-qca-nss-dp kmod-qca-ssdk kmod-qrtr kmod-qrtr-smd kmod-slhc kmod-thermal kmod-usb-core kmod-usb-dwc3 kmod-usb-dwc3-qcom kmod-usb-xhci-hcd kmod-usb3 libblkid libblobmsg-json libc libcomerr libext2fs libiwinfo libiwinfo-data libjson-c libjson-script liblucihttp liblucihttp-ucode libmbedtls libmnl libnftnl libnl-tiny libopenssl libopenssl-conf libopenssl-legacy libsmartcols libss libubox libubus libuci libuclient libucode libustream-mbedtls libuuid logd losetup luci luci-app-attendedsysupgrade luci-app-firewall luci-app-opkg luci-app-usteer luci-base luci-light luci-mod-admin-full luci-mod-network luci-mod-status luci-mod-system luci-proto-batman-adv luci-proto-ipv6 luci-proto-ppp luci-ssl luci-theme-bootstrap mtd nano-full netifd nftables-json odhcp6c odhcpd-ipv6only openwrt-keyring opkg ppp ppp-mod-pppoe procd procd-seccomp procd-ujail px5g-mbedtls rpcd rpcd-mod-file rpcd-mod-iwinfo rpcd-mod-luci rpcd-mod-rrdns rpcd-mod-ucode ubi-utils uboot-envtools ubox ubus ubusd uci uclient-fetch ucode ucode-mod-fs ucode-mod-html ucode-mod-math ucode-mod-nl80211 ucode-mod-rtnl ucode-mod-ubus ucode-mod-uci ucode-mod-uloop uhttpd uhttpd-mod-ubus urandom-seed urngd usign wireless-regdb wpad-openssl`

and it generates an image like this one: https://sysupgrade.openwrt.org/store/4e7abd28e6883532f9caa7551108254d046172479716e69fba87f2892ccbc0c5/openwrt-24.10.0-69391a0151c5-qualcommax-ipq807x-xiaomi_ax3600-squashfs-sysupgrade.bin

edrikk, 2025-02-19:
`    "target": "ipq807x/generic",`

This target was changed to `qualcommax/ipq807x` as part of https://github.com/openwrt/openwrt/commit/f02f6aaa8d4e1025ab4aa9f569123e57f689f4e5

ASU is still looking at the old target given your device's configuration.
I'm fairly sure (but can't remember for certain) - when it happened, I updated the 

/etc/openwrt_release  ---> DISTRIB_TARGET='qualcommax/ipq807x'
/etc/os-release ---> OPENWRT_BOARD="qualcommax/ipq807x"

I don't recall any other ones.  If that doesn't "work" you may want to reconfigure and start afresh.

migueldeblas, 2025-03-12:
Will there be an update to fix this without having to edit anything by hand?