openwrt/openwrt #18022: AttendedSystemUpgrade fails on 24.10

Gamerayers, 2025-02-19:
### Describe the bug

make: *** [Makefile:331: image] Error 2 occurs when you try to do an attendedsystemupgrade. The rest of the upgrade seems MUCH better than a week ago, but still fails.

### OpenWrt version

r28800-04570f5ee2

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R5S

### Image kind

Official downloaded image

### Steps to reproduce

Simply tried to do an attendedsystemupgrade.
I've been testing with my R5S to determine if dockers get blown away during an update prior to doing so on my x86. BUT, I can't get it to actually update.

### Actual behaviour

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r28800-04570f5ee2",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "c9213c3d7cb8cdae30ea60f1b84f1a0735bae9b09be4551193e69d13683083d0",
    "sha256_unsigned": "",
    "client": "luci/25.041.69577~2e9ffc6",
    "packages": {
        "adguardhome": "0.107.53-r1",
        "apk-mbedtls": "3.0.0_pre20241130-r2",
        "base-files": "1654~04570f5ee2",
        "ca-bundle": "20240203-r1",
        "dnsmasq": "2.90-r4",
        "docker": "27.3.1-r1",
        "docker-compose": "2.32.4-r1",
        "dockerd": "27.3.1-r1",
        "dropbear": "2024.86-r1",
        "e2fsprogs": "1.47.2-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "fstools": "2024.12.02~49d36ba2-r1",
        "hostapd": "2024.09.15~5ace39b0-r3",
        "kernel": "6.6.74~5e38b7d693c5cbf5e311103c5e070bd8-r1",
        "kmod-gpio-button-hotplug": "6.6.74-r5",
        "kmod-mt7915-firmware": "6.6.74.2025.01.22~a22d59e4-r1",
        "kmod-nft-offload": "6.6.74-r1",
        "kmod-r8169": "6.6.74-r1",
        "libc": "1.2.5-r4",
        "logd": "2024.04.26~85f10530-r1",
        "losetup": "2.40.2-r1",
        "lsblk": "2.40.2-r1",
        "luci": "25.041.69577~2e9ffc6",
        "luci-app-attendedsysupgrade": "25.041.69577~2e9ffc6",
        "luci-app-ddns": "25.041.69577~2e9ffc6",
        "luci-app-dockerman": "0.5.13.20241008-r1",
        "luci-app-usteer": "25.041.69577~2e9ffc6",
[... 409 lines cut ...]
WARNING: /builder/repositories:9: unrecogized keyword: packages/packages.adb
ERROR: failed to read repositories: /builder/repositories: unsupported repositories file keyword
/builder/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/usteer: line 10: /lib/functions/network.sh: No such file or directory
/builder/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/usteer: line 11: /usr/share/libubox/jshn.sh: No such file or directory
/builder/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/usteer: line 12: /lib/functions.sh: No such file or directory
./etc/init.d/usteer: line 10: /lib/functions/network.sh: No such file or directory
./etc/init.d/usteer: line 11: /usr/share/libubox/jshn.sh: No such file or directory
./etc/init.d/usteer: line 12: /lib/functions.sh: No such file or directory
Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-aarch64_generic_musl/root-rockchip/dev".
Ignoring, exclude it (-e/-ef) to override.
error: ext4_allocate_best_fit_partial: failed to allocate 386 blocks, out of space?
make[3]: *** [/builder/include/image.mk:397: /builder/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.ext4] Error 1
make[2]: *** [Makefile:262: build_image] Error 2
make[1]: *** [Makefile:158: _call_image] Error 2
make: *** [Makefile:331: image] Error 2

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-19:
Repeat attempt in hour/day. Basically error whatever the cause is cached and given back to you repeatedly without attempting new build.

Gamerayers, 2025-02-26:
I tried a number of times, and continue to get the errors. As I saw some errors about an unrecognized package for usteer, I removed that just to see if it would help. It didn't. I am still getting an error about too many blocks. So I removed the package of Adguardhome, and now the number of too many packages is even higher.

```
Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-aarch64_generic_musl/root-rockchip/dev".
Ignoring, exclude it (-e/-ef) to override.
error: ext4_allocate_best_fit_partial: failed to allocate 1050 blocks, out of space?
make[3]: *** [/builder/include/image.mk:397: /builder/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.ext4] Error 1
make[2]: *** [Makefile:263: build_image] Error 2
make[1]: *** [Makefile:159: _call_image] Error 2
make: *** [Makefile:332: image] Error 2
```

This all seems to fail with the ext4_allocate_best_fit_partial. Is there something I can do to fix this??