openwrt/openwrt #18034: [Raspberry Pi 5] SPI seems to be not working with an SPI Ethernet chip (ENC28J60)

AlbrechtL, 2025-02-19:
### Describe the bug

I'm working on a custom Raspberry Pi 5 image with ENC28J60 support [1]. The ENC28J60 chip is connected via SPI to the Pi and I added the ENC28J60 driver to OpenWrt [2]. In Raspberry OS the ENC28J60 is working fine. So my hardware setup is correct. Even when I compile my custom image for a Raspberry Pi Zero 1 the ENC28J60 is working fine. Therefore, I'm thinking that my configuration in my custom image is correct as well.

After some investigations, it seems that there are some SPI issues because the ENC28J60 is not detected. But I couldn't find the root cause (maybe because I'm not an SPI Linux driver expert).

[1] Custom Raspberry Pi 5 image with ENC28J60 support: https://github.com/AlbrechtL/openwrt-rtl8367n/tree/rtl8367n_support
[2] ENC28J60 package support: https://github.com/openwrt/openwrt/commit/9d11ec48d5a52960deaf84dd61b4f3025c5be5be.

### OpenWrt version

r28807-18e967e806

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B Rev 1.0

### Image kind

Self-built image

### Steps to reproduce

1. Connect a ENC28J60 module to the Raspberry Pi 5
2. Compile the custom image https://github.com/AlbrechtL/openwrt-rtl8367n with the diffconfig https://github.com/AlbrechtL/openwrt-rtl8367n/blob/18e967e80605d74b864953d5391002bef0cca571/diffconfig.rpi5
3. Boot the Pi
4. Watch for `eth1`

### Actual behaviour

```
root@OpenWrt:~# ls /dev/spi*
/dev/spidev10.0
```
 `spidev10.0` is the bus master not the ENC28J60. See `tree`:

```
root@OpenWrt:~# tree /sys/bus/spi/
/sys/bus/spi/
├── devices
│   └── spi10.0 -> ../../../devices/platform/soc/107d004000.spi/spi_master/spi10/spi10.0
├── drivers
│   ├── enc28j60
│   │   ├── bind
│   │   ├── module -> ../../../../module/enc28j60
│   │   ├── uevent
│   │   └── unbind
│   └── spidev
│       ├── bind
│       ├── module -> ../../../../module/spidev
│       ├── spi10.0 -> ../../../../devices/platform/soc/107d004000.spi/spi_master/spi10/spi10.0
│       ├── uevent
│       └── unbind
├── drivers_autoprobe
├── drivers_probe
└── uevent

9 directories, 9 files
```

### Expected behaviour

Recorded with Raspberry OS on the same hardware.

```
albrecht@albrecht-rpi5:~$ ls /dev/spi*
/dev/spidev0.1  /dev/spidev10.0
```

```
albrecht@albrecht-rpi5:~$ tree /sys/bus/spi/
/sys/bus/spi/
├── devices
│   ├── spi0.0 -> ../../../devices/platform/axi/1000120000.pcie/1f00050000.spi/spi_master/spi0/spi0.0
│   ├── spi0.1 -> ../../../devices/platform/axi/1000120000.pcie/1f00050000.spi/spi_master/spi0/spi0.1
│   └── spi10.0 -> ../../../devices/platform/soc/107d004000.spi/spi_master/spi10/spi10.0
├── drivers
│   ├── enc28j60
│   │   ├── bind
│   │   ├── module -> ../../../../module/enc28j60
│   │   ├── spi0.0 -> ../../../../devices/platform/axi/1000120000.pcie/1f00050000.spi/spi_master/spi0/spi0.0
│   │   ├── uevent
│   │   └── unbind
│   ├── spidev
│   │   ├── bind
│   │   ├── module -> ../../../../module/spidev
│   │   ├── spi0.1 -> ../../../../devices/platform/axi/1000120000.pcie/1f00050000.spi/spi_master/spi0/spi0.1
│   │   ├── spi10.0 -> ../../../../devices/platform/soc/107d004000.spi/spi_master/spi10/spi10.0
│   │   ├── uevent
│   │   └── unbind
│   └── stmpe-spi
│       ├── bind
│       ├── uevent
│       └── unbind
├── drivers_autoprobe
├── drivers_probe
└── uevent

14 directories, 12 files

```

### Additional info

```
root@OpenWrt:~# cat /boot/config.txt 
################################################################################
# Bootloader configuration - config.txt
################################################################################

################################################################################
# For overclocking and various other settings, see:
# https://www.raspberrypi.com/documentation/computers/config_txt.html
################################################################################

# OpenWrt config
include distroconfig.txt

[all]
# Place your custom settings here.

# For switch
dtparam=spi=on
#dtoverlay=rtl8365mb-enc28j60-switch
dtoverlay=enc28j60

# For UART
dtparam=uart0
dtparam=uart0_console
enable_rp1_uart=1
root@OpenWrt:~# 
```

### Diffconfig

```text
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2712=y
CONFIG_TARGET_bcm27xx_bcm2712_DEVICE_rpi-5=y
CONFIG_PACKAGE_kmod-dsa=y
CONFIG_PACKAGE_kmod-dsa-realtek=y
CONFIG_PACKAGE_kmod-dsa-rtl8365mb=y
CONFIG_PACKAGE_kmod-enc28j60=y
CONFIG_PACKAGE_kmod-fixed-phy=y
CONFIG_PACKAGE_kmod-lib-crc-itu-t=y
CONFIG_PACKAGE_kmod-lib-crc7=y
CONFIG_PACKAGE_kmod-libphy=y
CONFIG_PACKAGE_kmod-mdio-devres=y
CONFIG_PACKAGE_kmod-mii=m
CONFIG_PACKAGE_kmod-net-selftests=y
CONFIG_PACKAGE_kmod-of-mdio=y
CONFIG_PACKAGE_kmod-phy-realtek=y
CONFIG_PACKAGE_kmod-phylink=y
CONFIG_PACKAGE_kmod-regmap-core=y
CONFIG_PACKAGE_kmod-spi-bcm2835=y
CONFIG_PACKAGE_kmod-spi-dev=y
CONFIG_PACKAGE_r8169-firmware=m
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brocaar, 2025-06-03:
I'm struggling with the same issue. I have a custom built of OpenWrt 24.10.1, with SPI enabled in the `config.txt` like you have, the only thing I can see is `/dev/spidev10.0`. Did you make any progress?

TimVosch, 2025-06-04:
I expect that will be the issue. I see in `make kernel_menuconfig` that the designware modules are disabled. I am now building an image with those set to `built_in`. I'll report back

brocaar, 2025-06-04:
@AlbrechtL just to let you know, for my project we have solved this issue by enabling the following additional kernel modules (see also the above link):

```
CONFIG_SPI_DESIGNWARE=y
CONFIG_SPI_DW_MMIO=y
```

AlbrechtL, 2025-06-05:
Thanks for figuring it out!

Honestly, I'm a little bit lost. When I add the following to `target/linux/bcm27xx/bcm2712/config-6.6` and I recompile, nothing is changed, or I don't see any effect. SPI is still not working. When I run `make kernel_menuconfig CONFIG_TARGET=subtarget` after editing, my changes will be deleted in `target/linux/bcm27xx/bcm2712/config-6.6`.

```
CONFIG_SPI_DESIGNWARE=y
CONFIG_SPI_DW_MMIO=y
CONFIG_SPI_DW_DMA=y
# CONFIG_SPI_DW_PCI is not set
```

@brocaar Do you have an idea what is going at my side?

AlbrechtL, 2025-06-06:
> What if you set it through kernel_menuconfig? I don't have the repo setup now, but IIRC its under driver > SPI > DesignWare.

Correct. But enabling the features manually creates tons of issues before the kernel starts compiling because of questions of new kernel feature that need so be answered manually. But I'm pretty sure that in my kernel the SPI feature is enabled.

Maybe I miss something else. Can you share your `diffconfig` as well as your `config.txt `content?

Furthermore, I just saw that OpenWrt switched to Linux 6.12. Did you already test with 6.12?

TimVosch, 2025-06-06:
@AlbrechtL I don't have the laptop with the OpenWRT repo at hand, but check out this commit by brocaar, it should contain the new kernel config values:
https://github.com/chirpstack/chirpstack-gateway-os/commit/277dcbf635451e391a2c25df2fbe996f3a4c9b64

AlbrechtL, 2025-06-06:
Success! Tested with OpenWrt SNAPSHOT, r29961-79f097faae and kernel 6.12.32!

```
root@OpenWrt:~# tree /sys/bus/spi/
/sys/bus/spi/
├── devices
│   ├── spi0.0 -> ../../../devices/platform/axi/1000120000.pcie/1f00050000.spi/spi_master/spi0/spi0.0
│   └── spi0.1 -> ../../../devices/platform/axi/1000120000.pcie/1f00050000.spi/spi_master/spi0/spi0.1
├── drivers
│   └── enc28j60
│       ├── bind
│       ├── module -> ../../../../module/enc28j60
│       ├── spi0.0 -> ../../../../devices/platform/axi/1000120000.pcie/1f00050000.spi/spi_master/spi0/spi0.0
│       ├── uevent
│       └── unbind
├── drivers_autoprobe
├── drivers_probe
└── uevent

8 directories, 6 files

```

@TimVosch Thanks for the link - it helped a lot!

I added the following lines to `target/linux/bcm27xx/bcm2712/config-6.12`
```
CONFIG_SPI=y
# CONFIG_SPI_BCM2835AUX is not set
CONFIG_SPI_DESIGNWARE=y
# CONFIG_SPI_DW_DMA is not set
CONFIG_SPI_DW_MMIO=y
# CONFIG_SPI_DW_PCI is not set
CONFIG_SPI_DYNAMIC=y
CONFIG_SPI_MASTER=y
```

The issue was that I also enabled these kernel modules in `.config`. After removing, SPI was working fine!
```
CONFIG_PACKAGE_kmod-spi-bcm2835=y
CONFIG_PACKAGE_kmod-spi-dev=y
```

@TimVosch Do you think it makes sense providing a pull request to add this to OpenWrt?

AlbrechtL, 2025-06-10:
Thanks @brocaar and @TimVosch for figuring out the correct kernel modules!

brocaar, 2025-06-11:
Thanks @AlbrechtL for integrating this into OpenWrt 👍