openwrt/openwrt #18035: Recent changes break the build of libffi

graysky2, 2025-02-19:
### Describe the bug

Recent changes leave the build system unable to compile libffi, see error below confirmed by myself and @hnyman 

### OpenWrt version

N/A

### OpenWrt release

N/A

### OpenWrt target/subtarget

confirmed on x86/64 and bcm27xx/bcm2712

### Device

N/A

### Image kind

Self-built image

### Steps to reproduce

Checkout HEAD then try to build libffi

### Actual behaviour

Here is an example of the error:
```
...
CFLAGS="-mcpu=cortex-a76 -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6=libffi-3.4.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/usr/include -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/include -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/include/fortify " CXXFLAGS="-mcpu=cortex-a76 -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6=libffi-3.4.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/usr/include -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/include -I/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/include/fortify " LDFLAGS="-L/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/usr/lib -L/scratch/openwrt/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/lib -fuse-ld=bfd -znow -zrelro " make --jobserver-auth=fifo:/tmp/GMfifo3545584  -C /scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -mcpu=cortex-a76 -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6=libffi-3.4.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="aarch64-openwrt-linux-musl-ld.bfd" NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;
make[3]: Entering directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6'
MAKE x86_64-pc-linux-musl : 0 * all-all
make[4]: Entering directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu'
make  all-recursive
make[5]: Entering directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu'
Making all in include
make[6]: Entering directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu/include'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu/include'
Making all in testsuite
make[6]: Entering directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu/testsuite'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu/testsuite'
Making all in man
make[6]: Entering directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu/man'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu/man'
make[6]: Entering directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu'
make[6]: Leaving directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu'
make[5]: Leaving directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu'
make[4]: Leaving directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu'
MAKE x86_64-pc-linux-musl : 0 * all-configured
make[4]: Entering directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu'
make[4]: *** No rule to make target 'all-configured'.  Stop.
make[4]: Leaving directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/aarch64-openwrt-linux-gnu'
make[3]: *** [Makefile:3826: all-configured] Error 2
make[3]: Leaving directory '/scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6'
make[2]: *** [Makefile:92: /scratch/openwrt/build_dir/target-aarch64_cortex-a76_musl/libffi-3.4.6/.built] Error 2
make[2]: Leaving directory '/scratch/openwrt/feeds/packages/libs/libffi'
time: package/feeds/packages/libffi/compile#0.13#0.04#0.16
    ERROR: package/feeds/packages/libffi failed to build.
make[1]: *** [package/Makefile:189: package/feeds/packages/libffi/compile] Error 1
make[1]: Leaving directory '/scratch/openwrt'
make: *** [/scratch/openwrt/include/toplevel.mk:233: package/libffi/compile] Error 2
```

[... 429 lines cut ...]
CONFIG_PACKAGE_zsh=y
CONFIG_PACKAGE_zstd=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_PERL_NOCOMMENT=y
CONFIG_PERL_THREADS=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_TARGET_EXT4_RESERVED_PCT=1
CONFIG_TARGET_KERNEL_PARTSIZE=64
CONFIG_TARGET_OPTIMIZATION="-march=x86-64-v3 -O3 -pipe"
CONFIG_TARGET_OPTIONS=y
CONFIG_TARGET_ROOTFS_PARTSIZE=512
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_ZSTD_OPTIMIZE_O3=y
CONFIG_KERNEL_WERROR=y
# CONFIG_PACKAGE_dnsmasq_full_dhcpv6 is not set
# CONFIG_RSYNC_acl is not set
# CONFIG_RSYNC_xattr is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

hnyman, 2025-02-19:
I wonder if it could be due to today's large changes to automake, autoconf, bc, mtools

At least one old relocatability patch from @jow- was removed from autoconf in these version upgrade.

asvio, 2025-02-19:
Reverting commit [b930ce3](https://github.com/openwrt/openwrt/commit/b930ce3bf7aa350cf0252dbf95c9447e51be4de9) seems to solve the problem, at least for nbg7815

robimarko, 2025-02-20:
The issue is that we are using autoreconf on libffi for no reason since the release tarballs already contain the configure script and libffi probably was not updated against newer automake.

closed 2025-02-20 by robimarko