openwrt/openwrt #18042: mediatek: filogic: 2.5G WAN port LED on EX5601 remains off at power on or reboot

stepd88, 2025-02-20:
### Describe the bug

the led indicating a 2.5 link on the WAN port remains off when the device is power on or rebooted
issue started from: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=9772ca190ee45d33129ed2e0df3b3b7bd014a860

### OpenWrt version

r28840-31dc43daf5

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Zyxel EX5601-T0 (stock layout)

### Image kind

Official downloaded image

### Steps to reproduce

Power on, or simply reboot the device.

### Actual behaviour

Power on or simply reboot the device with an active 2.5 link connected to the WAN port, the 2.5G WAN LED will remain off, to turn it on, you need to disconnect/reconnect the cable from the WAN port, or turn off/reboot the other device connected to the other end of the WAN.eboot device with 

### Expected behaviour

Revert or fix issue

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

abajk, 2025-02-22:
@stepd88 On OpenWRT 24.10, the behavior is the same. My commit does not affect the LED on the front panel. Maybe you are talking about a different LED?

stepd88, 2025-02-22:
ok