openwrt/openwrt #18044: Truly poor PPPoE performance after installing 24.10 on Cudy WR1300 v2

ogarcia, 2025-02-20:
### Describe the bug

The truth is that there is not much more to tell. With OpenWrt version 23.05.5 r24106-10cc5fcd00 it works perfect but from the moment you upgrade to 24.10.0 the PPPoE performance drops to less than half. If you downgrade everything works perfectly again.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

Cudy WR1300 v2

### Image kind

Official downloaded image

### Steps to reproduce

1. Upgrade or install `openwrt-24.10.0-ramips-mt7621-cudy_wr1300-v2-squashfs-sysupgrade.bin` in a Cudy WR1300 v2.
2. Configure the wan as PPPoE client.
3. Perform a speed test (at most it will reach 150/160mps).

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ogarcia, 2025-02-20:
Neither option does anything.

With `ethtool -K wan rx-gro-list off` or `ethtool -K wan gro off` the same results are obtained. It doesn't seem to have anything to do with that.

aimacintyre, 2025-02-23:
Try setting packet steering to `Enabled (All CPUs)` instead of `Enabled`?

ogarcia, 2025-03-06:
First of all my apologies for being late, I have had some complicated days and I have not been able to do the test until now.

@aimacintyre I have made the change and it solves the problem. I don't know if I should document this or do some major revision to see why with the previous version this setting was not necessary. Anything else let me know.

aimacintyre, 2025-03-07:
In 24.10 and later the packet steering option value `Enabled (All CPUs)` has the same effect as the option value `Enabled` had in 23.05 and earlier; the option value `Enabled` now has a more limited effect - hence change the setting restores the 23.05 behaviour/performance in your case.

Since 23.05 the `mt7621` subtarget has had `Enabled` as the default packet steering option value while nearly all other targets had `Disabled` as the default value; in 24.10 and later `Enabled` is now the default value for all targets.

ogarcia, 2025-03-07:
> In 24.10 and later the packet steering option value `Enabled (All CPUs)` has the same effect as the option value `Enabled` had in 23.05 and earlier;

Ok. I take note of the change. Thank you!

closed 2025-03-07 by ogarcia

d3rsch0rsch, 2025-07-02:
OMG, thank you so much for this hint!!!