openwrt/openwrt #18052: Linksys EA8500 IPQ806x router gets wrong WAN port MAC address assignment in 24.10.0

rwl408, 2025-02-21:
In 24.10.0 firmware release Linksys EA8500, and possibly all of IPQ806x routers, has its wan port MAC address assigned to be the lan port's MAC address plus one. Unfortunately that particular address is used by phy0-ap0 (2.4 GHz radio 0). As such, when the 2.4 GHz radio is enabled, the wan port loses internet access.

EngineerGuy314, 2025-02-22:
I ran into the same exact problem with 24.10.0 on my Linksys EA8500. Changed the MAC address of the WAN to a completely different random value, and now it works fine.

v12mike, 2025-03-03:
> I ran into the same exact problem with 24.10.0 on my Linksys EA8500. Changed the MAC address of the WAN to a completely different random value, and now it works fine.

ME too ;)
In Luci, navigate to network/interfaces/devices/wan and edit the MAC address

amenos93, 2025-04-12:
Thank you man . i also change tehe 2.4 wifi settings mac adress randomly every think is fine

tom887812, 2025-10-06:
[IPQ806x] 24.10.0-3: QCA8K DSA driver breaks Layer 2 communication on WAN port - ARP/NDP completely fails
Device Information
Platform: IPQ806x
OpenWrt Version: 24.10.0 / 24.10.1 / 24.10.2 / 24.10.3 (all affected)
Kernel: 6.6.104
Architecture: arm_cortex-a15_neon-vfpv4
DSA Driver: qca8k (QCA8337 switch)

Issue Summary
WAN port cannot establish Layer 2 connectivity with upstream router when using OpenWrt 24.10.x on IPQ806x devices. Both IPv4 ARP and IPv6 NDP fail to resolve the gateway MAC address, making the WAN interface completely non-functional despite physical link being up.

Working versions: 23.05.0 - 23.05.6 (same hardware, same configuration)
Broken versions: 24.10.0, 24.10.1, 24.10.2, 24.10.3

Network Topology
Primary Router (23.05.5)
├─ LAN: 192.168.1.1/24
└─ WAN port connected to ↓

Secondary Router (24.10.3) ← AFFECTED
├─ WAN: 192.168.1.2/24 (static IP)
│  └─ Gateway: 192.168.1.1
└─ LAN: 192.168.2.1/24
Symptoms
1. ARP Resolution Fails
# ip neigh show 192.168.1.1
192.168.1.1 dev wan ref 1 used 0/0/0 probes 5 INCOMPLETE
# After multiple retries:
192.168.1.1 dev wan used 0/0/0 probes 6 FAILED
2. IPv6 NDP Also Fails
Bash

# ip -6 neigh show dev wan
fe80::9e3d:cfff:fe04:af72 router used 0/0/0 probes 3 FAILED
fd66:9ab6:2980::1 ref 1 used 0/0/0 probes 5 INCOMPLETE
3. Ping Completely Fails
# ping -c 3 192.168.1.1
PING 192.168.1.1: 56 data bytes
--- 192.168.1.1 ping statistics ---
3 packets transmitted, 0 packets received, 100% packet loss
4. Can Only Ping Self
✅ Ping to WAN IP (192.168.1.2): Works
✅ Ping to own IPv6 link-local: Works
❌ Ping to gateway (192.168.1.1): 100% loss
❌ Ping to any external host: 100% loss
5. Physical Link is UP
# ip link show wan
8: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP
    link/ether 58:ef:ff:ff:ff:ff brd ff:ff:ff:ff:ff:ff

# dmesg | grep wan
qca8k 37000000.mdio-mii:10 wan: Link is Up - 1Gbps/Full - flow control rx/tx
6. IPv6 Address Acquired (but unusable)
# ip addr show wan
inet6 fd66:9ab6:2980:0:5aef:68ff:fe2d:5b95/64 scope global noprefixroute
inet6 260e:2a0:d6s:f1:ae:6f:f2:56e3/64 scope global dynamic
Note: IPv6 addresses are acquired via SLAAC, proving some Layer 2 communication exists, but directed communication to gateway fails.

Configuration
Network Configuration
# uci show network.wan
network.wan=interface
network.wan.device='wan'
network.wan.proto='static'
network.wan.ipaddr='192.168.1.2'
network.wan.netmask='255.255.255.0'
network.wan.gateway='192.168.1.1'
network.wan.dns='1.1.1.1 1.0.0.1'

# uci show network.@device[5]
[... 77 lines cut ...]
System Information Dump

=== OpenWrt Release ===
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='24.10.3'
DISTRIB_REVISION='r28872-daca7c049b'
DISTRIB_TARGET='ipq806x/generic'

=== Kernel ===
Linux 6.6.104 #0 SMP armv7l GNU/Linux

=== DSA Configuration ===
# lsmod | grep dsa
(Module built into kernel)

=== QCA8K Driver ===
qca8k 37000000.mdio-mii:10 wan: PHY [qca8k-0.0:04] 
Qualcomm Atheros 8337 internal PHY (irq=POLL)

=== Network Interfaces ===
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1502
    link/ether 62:cd:ff:ff:ff:ff
8: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500
    link/ether 58:ef:ff:ff:ff:ff
    inet 192.168.1.2/24 brd 192.168.1.255 scope global wan