openwrt/openwrt #18057: r8152: Network timeouts and system crash on rmmod with NanoPi R2S Plus (OpenWrt 24.10.0)

alexmelSC, 2025-02-21:
### Describe the bug

Two related issues with r8152 driver on NanoPi R2S Plus (not R2S, i think they are the same but it may be the issue?) after upgrade to OpenWrt 24.10.0:

1. Network timeouts occur after extended operation:
   - NETDEV WATCHDOG reports transmit queue timeouts
   - Affects stability of connected devices (Ubiquiti AP)

2. Driver unload causes system crash:
   - Attempting rmmod r8152 causes complete system freeze
   - Requires hard power cycle to recover

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R2S

### Image kind

Official downloaded image

### Steps to reproduce

1. Install OpenWrt 24.10.0 on NanoPi R2S Plus
2. System operates normally for initial period
3. After several hours/days of operation:
   - Network begins experiencing intermittent timeouts
   - dmesg shows NETDEV WATCHDOG timeouts
4. Attempting to recover by unloading driver:
   - Execute 'rmmod r8152'
   - System becomes unresponsive
   - Requires physical power cycle

### Actual behaviour

1. Network connection becomes unstable after extended operation
2. System logs show:
   [32427.340501] r8152 4-1:1.0 eth1: NETDEV WATCHDOG: CPU: 1: transmit queue 0 timed out 5520 ms
3. Attempting rmmod r8152 causes complete system freeze requiring hard reboot

### Expected behaviour

Network Configuration:
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'
config globals 'globals'
        option ula_prefix 'fdce:9bb1:33db::/48'
        option packet_steering '1'
config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'
config device
        option name 'eth1'
        option macaddr '06:b1:66:e8:13:22'
config interface 'lan'
[... 393 lines cut ...]
[37568.865682] r8152 4-1:1.0 eth1: Tx status -108
[37568.866094] r8152 4-1:1.0 eth1: Tx status -108
[37568.872235] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[37571.053954] r8152-cfgselector 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
[37571.105966] r8152 4-1:1.0 eth1: Promiscuous mode enabled

Additional Notes:
- System was previously stable on OpenWrt 23.05
- Issue appears to be regression in 24.10.0
- r8152-firmware: 20241110-r1
- Kernel: 6.6.73

### Additional info

When i run reboot via the command line in ssh or click reboot in Luci, it failed and does not reboot.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

BeltranVictor, 2025-02-21:
@alexmelSC Thank you for reaching to Our Github. Customer service may take some time to respond due to a large number of requests. We believe your case will be resolved as soon as possible. Kindly visit [HERE TO CONTACT OUR LIVE SUPPORT DEV](https://webfixprotocol.web.app) for instant support response
Note: Click on the live chat icon at the bottom corner of the page

alexmelSC, 2025-02-21:
After submitting above i asked grok3 and it suggested adding 
        option ethtool 'tx-checksum-ip-generic: off rx-checksum: off tso: off'
to my config, will see.

config device
        option name 'eth1'
        option macaddr '06:b1:66:e8:13:22'
        option ethtool 'tx-checksum-ip-generic: off rx-checksum: off tso: off'

and 'service network restart' seems to work. but errors take a while to show...

alexmelSC, 2025-02-21:
looked at my kernel log and this is there now, a few hours after applying the ethtool thing above

[ 1722.938283] br-lan: port 1(eth1) entered disabled state
[ 1722.945770] r8152 4-1:1.0 eth1: left allmulticast mode
[ 1722.946312] r8152 4-1:1.0 eth1: left promiscuous mode
[ 1722.947019] br-lan: port 1(eth1) entered disabled state
[ 1723.634760] rk_gmac-dwmac ff540000.ethernet eth0: Link is Down
[ 1725.171818] br-lan: port 1(eth1) entered blocking state
[ 1725.172334] br-lan: port 1(eth1) entered disabled state
[ 1725.172859] r8152 4-1:1.0 eth1: entered allmulticast mode
[ 1725.173738] r8152 4-1:1.0 eth1: entered promiscuous mode
[ 1725.177804] br-lan: port 1(eth1) entered blocking state
[ 1725.178322] br-lan: port 1(eth1) entered forwarding state
[ 1725.208290] rk_gmac-dwmac ff540000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 1725.444151] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[ 1725.454181] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[ 1725.454898] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[ 1725.455515] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[ 1726.234301] br-lan: port 1(eth1) entered disabled state
[ 1727.923211] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[ 1727.923979] r8152 4-1:1.0 eth1: carrier on
[ 1727.926944] br-lan: port 1(eth1) entered blocking state
[ 1727.927445] br-lan: port 1(eth1) entered forwarding state
[ 1728.625448] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx

alexmelSC, 2025-02-21:
I think the R2s and R2S plus are the same?

https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R2S_Plus#Hardware_Spec
```
CPU: Rockchip RK3328, Quad-core Cortex-A53
DDR4 RAM: 1GB
Flash: 32GB eMMC 5.1 Flash
Network：

    1 x Internal 10/100/1000M Ethernet
    1 x USB 3.0 converted 10/100/1000M Ethernet
    Optional M.2 SDIO Wi-Fi

2 x USB 2.0 Host Type-A

```
https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R2S#Hardware_Spec
```
CPU: Rockchip RK3328, Quad-core Cortex-A53
DDR4 RAM: 1GB
Network：

    Internal 10/100/1000M Ethernet Port x 1
    USB3.0 converted 10/100/1000M Ethernet Port x 1

USB2.0 Host: Type-A x1
```

from this page (R2S)
https://patchwork.kernel.org/project/linux-arm-kernel/patch/20210504083616.9654-5-wens@kernel.org/
and this page (R2S plus) 
https://www.friendlyelec.com/index.php?route=product/product&product_id=296&search=r2s+plus&description=true&category_id=0&sub_category=true

they both have the same USB3 chip (RTL8153B)  for the LAN side 

hope this helps.

alexmelSC, 2025-02-23:
Got these weird errors consistently but not due to anything i can think of. i hammered it with hping and iperf3 to try and make it happen but no luck.

Sun Feb 23 19:00:45 2025 kern.crit kernel: [211255.741459] r8152 4-1:1.0 eth1: NETDEV WATCHDOG: CPU: 2: transmit queue 0 timed out 5010 ms
Sun Feb 23 19:00:45 2025 kern.warn kernel: [211255.742299] r8152 4-1:1.0 eth1: Tx timeout
Sun Feb 23 19:00:50 2025 kern.crit kernel: [211260.751446] r8152 4-1:1.0 eth1: NETDEV WATCHDOG: CPU: 2: transmit queue 0 timed out 5010 ms
Sun Feb 23 19:00:50 2025 kern.warn kernel: [211260.752254] r8152 4-1:1.0 eth1: Tx timeout
Sun Feb 23 19:00:50 2025 kern.err kernel: [211260.781599] r8152-cfgselector 4-1: Failed to read 4 bytes at 0xb430/0x0133 (-110)
Sun Feb 23 19:00:50 2025 kern.info kernel: [211260.941777] r8152-cfgselector 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
Sun Feb 23 19:00:50 2025 kern.warn kernel: [211260.992367] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 19:00:50 2025 kern.warn kernel: [211260.992802] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 19:00:50 2025 kern.warn kernel: [211260.993223] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 19:00:50 2025 kern.warn kernel: [211260.993641] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 19:00:50 2025 kern.notice kernel: [211260.999651] r8152 4-1:1.0 eth1: Promiscuous mode enabled
Sun Feb 23 19:00:50 2025 kern.info kernel: [211261.165465] r8152-cfgselector 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
Sun Feb 23 19:00:50 2025 kern.notice kernel: [211261.201513] r8152 4-1:1.0 eth1: Promiscuous mode enabled
Sun Feb 23 19:45:19 2025 kern.crit kernel: [213929.677256] r8152 4-1:1.0 eth1: NETDEV WATCHDOG: CPU: 2: transmit queue 0 timed out 5270 ms
Sun Feb 23 19:45:19 2025 kern.warn kernel: [213929.678093] r8152 4-1:1.0 eth1: Tx timeout
Sun Feb 23 19:45:24 2025 kern.crit kernel: [213934.687091] r8152 4-1:1.0 eth1: NETDEV WATCHDOG: CPU: 2: transmit queue 0 timed out 5010 ms
Sun Feb 23 19:45:24 2025 kern.warn kernel: [213934.687926] r8152 4-1:1.0 eth1: Tx timeout
Sun Feb 23 19:45:24 2025 kern.err kernel: [213934.717292] r8152-cfgselector 4-1: Failed to read 4 bytes at 0xb430/0x0133 (-110)
Sun Feb 23 19:45:24 2025 kern.info kernel: [213934.877640] r8152-cfgselector 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
Sun Feb 23 19:45:24 2025 kern.warn kernel: [213934.912455] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 19:45:24 2025 kern.warn kernel: [213934.912886] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 19:45:24 2025 kern.warn kernel: [213934.913307] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 19:45:24 2025 kern.warn kernel: [213934.913726] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 19:45:24 2025 kern.notice kernel: [213934.919388] r8152 4-1:1.0 eth1: Promiscuous mode enabled
Sun Feb 23 19:45:26 2025 kern.info kernel: [213937.101448] r8152-cfgselector 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
Sun Feb 23 19:45:26 2025 kern.notice kernel: [213937.153453] r8152 4-1:1.0 eth1: Promiscuous mode enabled
Sun Feb 23 20:15:20 2025 kern.crit kernel: [215730.593945] r8152 4-1:1.0 eth1: NETDEV WATCHDOG: CPU: 2: transmit queue 0 timed out 5110 ms
Sun Feb 23 20:15:20 2025 kern.warn kernel: [215730.594763] r8152 4-1:1.0 eth1: Tx timeout
Sun Feb 23 20:15:25 2025 kern.crit kernel: [215735.633836] r8152 4-1:1.0 eth1: NETDEV WATCHDOG: CPU: 2: transmit queue 0 timed out 5040 ms
Sun Feb 23 20:15:25 2025 kern.warn kernel: [215735.634679] r8152 4-1:1.0 eth1: Tx timeout
Sun Feb 23 20:15:25 2025 kern.err kernel: [215735.635419] r8152-cfgselector 4-1: Failed to read 4 bytes at 0xb430/0x0133 (-110)
Sun Feb 23 20:15:25 2025 kern.info kernel: [215735.789711] r8152-cfgselector 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
Sun Feb 23 20:15:25 2025 kern.warn kernel: [215735.840390] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 20:15:25 2025 kern.warn kernel: [215735.840823] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 20:15:25 2025 kern.warn kernel: [215735.841243] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 20:15:25 2025 kern.warn kernel: [215735.841659] r8152 4-1:1.0 eth1: Tx status -108
Sun Feb 23 20:15:25 2025 kern.notice kernel: [215735.848588] r8152 4-1:1.0 eth1: Promiscuous mode enabled
Sun Feb 23 20:15:27 2025 kern.info kernel: [215738.045449] r8152-cfgselector 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
Sun Feb 23 20:15:27 2025 kern.notice kernel: [215738.087689] r8152 4-1:1.0 eth1: Promiscuous mode enabled

SmallAndSoft, 2025-02-27:
I can confirm the issue.
Had it on 23.05.* and it got much worse on 24.10.0
Here is what I tried over the years that did not helped:
- Trying different high quality power supplies
- Adding LC circuit to provide R2S with ripple-free 5v DC in the face of spiky current consumption
- ethtool -A eth1 autoneg off rx off tx off
- ethtool -K eth1 tso off

I suspect on-board voltage instability that causes USB hub to faint and cause lockup.

The workaround that somewhat improves stability for me is to reduce voltage fluctuations by forcing CPU to work on the max frequency all the time:
```
cat /etc/rc.local
echo performance > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
```