openwrt/openwrt #18059: Autoconf and automake upgrades caused trouble for packages buildbot

hnyman, 2025-02-21:
### Describe the bug

Looks like the recent upgrade of autoconf by #17825 and automake by #17827 (and maybe some other toolchain tools at the same time) have caused major trouble for the packages buildbot. The current faillogs list is 400+ packages, including procd, firewall, etc. basic OpenWrt packages.

The errors seem to relate the autoconf/automake components, or finding libtool etc.

Examples from buildbot faillogs today:

https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a76/
https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a7/base/
https://downloads.openwrt.org/snapshots/faillogs/mips64_mips64r2/

wolfssl:
```
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /builder/shared-workdir/build/sdk/staging_dir/host/bin/aclocal -I /builder/shared-workdir/build/sdk/staging_dir/target-arm_cortex-a7_musl_eabi/host/share/aclocal -I /builder/shared-workdir/build/sdk/staging_dir/hostpkg/share/aclocal -I /builder/shared-workdir/build/sdk/staging_dir/target-arm_cortex-a7_musl_eabi/usr/share/aclocal -I m4 -I . --force -I m4
/builder/shared-workdir/build/sdk/staging_dir/host/bin/aclocal: 2: /builder/shared-workdir/build/sdk/staging_dir/host/bin/aclocal.real: not found
autoreconf: error: /builder/shared-workdir/build/sdk/staging_dir/host/bin/aclocal failed with exit status: 127
/builder/shared-workdir/build/sdk/staging_dir/host/bin/sed: can't read /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7_musl_eabi/wolfssl-regular/wolfssl-5.7.6-stable/libtool: No such file or directory
```

libmnl
```
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash '/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a7_musl_eabi/libmnl-1.0.5/build-aux/missing' aclocal-1.16 -I m4
/builder/shared-workdir/build/sdk/staging_dir/host/bin/aclocal-1.16: 2: /builder/shared-workdir/build/sdk/staging_dir/host/bin/aclocal.real: not found
WARNING: 'aclocal-1.16' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <https://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <https://www.gnu.org/software/autoconf>
         <https://www.gnu.org/software/m4/>
         <https://www.perl.org/>
make[4]: *** [Makefile:433: aclocal.m4] Error 127
```

asterisk
```
Can't locate Autom4te/ChannelDefs.pm in @INC (you may need to install the Autom4te::ChannelDefs module) (@INC contains: /builder/shared-workdir/build/staging_dir/host/share/autoconf /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.32.1 /usr/local/share/perl/5.32.1 /usr/lib/x86_64-linux-gnu/perl5/5.32 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl-base /usr/lib/x86_64-linux-gnu/perl/5.32 /usr/share/perl/5.32 /usr/local/lib/site_perl) at /builder/shared-workdir/build/sdk/staging_dir/host/bin/autoconf line 42.
BEGIN failed--compilation aborted at /builder/shared-workdir/build/sdk/staging_dir/host/bin/autoconf line 42.
Please install autoconf and run bootstrap.sh again!
```

cc @Ansuel 

### OpenWrt version

main

### OpenWrt release

main

### OpenWrt target/subtarget

all

### Device

toolchain

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

kuanyili, 2025-02-21:
Changing `PERL` from `/usr/bin/env perl` to `$(shell command -v perl)` in b930ce3bf7aa might be the culprit.

The modification seems to be driven by upstream commit introduced in v1.17
https://git.savannah.gnu.org/cgit/automake.git/commit/?id=7f33a3b841b4339093f6a72de030e49e053fc447
but new shebang also makes autotools in SDK (used by buildbot) non-relocatable.

There is currently a commit on master branch which may solve this issue.
https://git.savannah.gnu.org/cgit/automake.git/commit/?id=5beb339d15db32dd4df3ddc711adb3ece2851e42

Issue #18035 is probably related.

hnyman, 2025-02-21:
You should probably revert both until somebody with expertise analyses those upgrades.

robimarko, 2025-02-21:
Most likely its due to autoreconf being called which makes regenerates the otherwise shipped configure scripts and/or makefile.

And the newer automake is incompatible or its being misconfigured in our host build.

robimarko, 2025-02-21:
You are probably onto something as autoconf dropped the `autoconf.as` shell script and rewrote it as `autoconf.in` in Perl so we probably gotta patch that one to look at `STAGING_DIR_HOST`

Ansuel, 2025-02-21:
Ok my bad for not noticing dropping of relocation patch... That is probably
the culprit

hnyman, 2025-02-22:
Could the solution be something like this?

I have no perl skills, but based on other changes in 000-relocatable patch, I might have suucceeeded to re-create the same functionality that we had in the bin/autoconf.as before e0f5ce974 removed it.  
(So long lines that I split them heavily.)

```
-my $autom4te = $ENV{'AUTOM4TE'} || '@bindir@/@autom4te-name@';
-my $trailer_m4 = $ENV{'trailer_m4'} || '@pkgdatadir@/autoconf/trailer.m4';
+my $autom4te = $ENV{'AUTOM4TE'} ||
+	($ENV{'STAGING_DIR_HOST'} ?
+		$ENV{'STAGING_DIR_HOST'} . '/bin/@autom4te-name@' :
+		'@bindir@/@autom4te-name@');
+my $trailer_m4 = $ENV{'trailer_m4'} ||
+	($ENV{'STAGING_DIR_HOST'} ?
+		$ENV{'STAGING_DIR_HOST'} . '/share/autoconf/autoconf/trailer.m4' :
+		'@pkgdatadir@/autoconf/trailer.m4');
```

Added into 000-relocateable.patch 
```
--- a/tools/autoconf/patches/000-relocatable.patch
+++ b/tools/autoconf/patches/000-relocatable.patch
@@ -1,3 +1,22 @@
+--- a/bin/autoconf.in
++++ b/bin/autoconf.in
+@@ -44,8 +44,14 @@
+ use Autom4te::General;
+ 
+ # Lib files.
+-my $autom4te = $ENV{'AUTOM4TE'} || '@bindir@/@autom4te-name@';
+-my $trailer_m4 = $ENV{'trailer_m4'} || '@pkgdatadir@/autoconf/trailer.m4';
++my $autom4te = $ENV{'AUTOM4TE'} ||
++	($ENV{'STAGING_DIR_HOST'} ?
++		$ENV{'STAGING_DIR_HOST'} . '/bin/@autom4te-name@' :
++		'@bindir@/@autom4te-name@');
++my $trailer_m4 = $ENV{'trailer_m4'} ||
++	($ENV{'STAGING_DIR_HOST'} ?
++		$ENV{'STAGING_DIR_HOST'} . '/share/autoconf/autoconf/trailer.m4' :
++		'@pkgdatadir@/autoconf/trailer.m4');
+ 
+ # $HELP
+ # -----
 --- a/bin/autoheader.in
 +++ b/bin/autoheader.in
 @@ -30,9 +30,12 @@ use 5.006;
```

Autoconf compiles ok with it, and I am currently trying to test compile a firmware.

ptpt52, 2025-02-22:
mark

dmascord, 2025-02-24:
Looks like we also need https://github.com/autoconf-archive/autoconf-archive/pull/313/commits/e23f281c036f0c2b4a1b462ace02f9afd70c045d so that LUAJIT support is correctly handled

robimarko, 2025-02-24:
I have been digging a bit more and the issue with `aclocal.real` falling is because its now Perl but the Perl shebang is pointing to the original full path of the host staging dir Perl binary which then off course as soon as you change the directory name is wrong and thus no Perl is found and `aclocal.real` is not executed.

That is issue number 1, but even after that building will still fail though with a more mysterious errors

robimarko, 2025-02-24:
I think I will just revert Automake update for now since fixing the Perl issue requires backporting of the error to the warning patch, but that then requires `autoreconf` to be done so `configure` gets updated but that will fail since we dont yet have automake built at that time so we get the chicken and egg issue.

robimarko, 2025-02-24:
I reverted the automake bump as we cannot really fix it due to being unable to autoreconf at that stage

closed 2025-02-24 by openwrt-bot

lucize, 2025-02-25:
@hnyman @robimarko 
tried to compile locally, but at least for frr-host I still have this issue

```

configure.ac:378: error: AM_COND_IF: no such condition "LUAJIT"
/home/proxy/staging_dir/host/share/aclocal-1.16/cond-if.m4:23: AM_COND_IF is expanded from...
/home/proxy/staging_dir/host/share/aclocal/ax_lua.m4:199: AX_PROG_LUA is expanded from...
configure.ac:378: the top level
autom4te: error: /home/proxy/staging_dir/host/bin/m4 failed with exit status: 1
aclocal.real: error: /home/proxy/staging_dir/host/bin/autom4te failed with exit status: 1
autoreconf: error: /home/proxy/staging_dir/host/bin/aclocal failed with exit status: 1

```

robimarko, 2025-02-25:
@lucize Yeah, it seems to be a known bug in 2024.10.16
https://lists.gnu.org/archive/html/autoconf-archive-maintainers/2024-11/msg00000.html

So, reverted it for now

lorand-horvath, 2025-02-26:
@hnyman @robimarko Building a snapshot image still won't work for me, should I wait more or is there another issue?

robimarko, 2025-02-26:
@lorand-horvath With what error it fails?

Also, make sure to completely clean everything as just rebuilding autoconf-archive for example does not work

hnyman, 2025-02-26:
> Building a snapshot image still won't work for me, 

Does not compile from sources, with the full toolchain?
That should work.

Or do you mean some of the buildbot dependent cook-the-image methods?
The buildbot is still recovering from the automake/autoconf mishap plus in addition from a out-of-space issue in some of the workers. It will still take 1-2 days until all packages have been rebuilt and available for cooking the image fro ready components. Not yet green for all architectures. https://buildbot.openwrt.org/main/packages/#/builders

commit e03676a9 (2025-02-24): tools/autoconf: Fix relocatable patch

Fix the 000-relocatable.patch broken by e0f5ce9. The patch segment about
detecting STAGING_DIR_HOST was erroneously removed, as upstream had
deleted the previous bin/autoconf.as and had implemented it in perl
in bin/autoconf.in. Re-create the previous functionality in that.

Fixes: #18059
Fixes: e0f5ce974 ("tools/autoconf: bump to 2.72")

Link: https://github.com/openwrt/openwrt/pull/18073
PR #18073 (2025-02-24): tools/autoconf: Fix 000-relocatable patch
cc @robimarko @Ansuel 

Fix the 000-relocatable.patch broken by e0f5ce9. The patch segment about detecting `STAGING_DIR_HOST` was erroneously removed, as upstream had deleted the previous `bin/autoconf.as` and had implemented it in perl in `bin/autoconf.in.` Re-create the previous functionality in that.

The erroneously removed section was: https://github.com/openwrt/openwrt/commit/e0f5ce9746323f0c38e49594411dbeb5394e2f15#diff-3a6566ac67b53ff73ff207e25f03fd126675208f4cd8c9882219fa074da91047L178-L195

Fixes: #18059
Fixes: e0f5ce974 ("tools/autoconf: bum [... cut]