openwrt/openwrt #18061: Can't build 24.10.0 release from sources for NanoPi devices

TyroSpy, 2025-02-21:
### Describe the bug

I get following error:
```
make -f ./scripts/Makefile.build obj=scripts/dtc/pylibfdt
  cat scripts/dtc/pylibfdt/libfdt.i_shipped > scripts/dtc/pylibfdt/libfdt.i
  unset CROSS_COMPILE; unset CFLAGS; CC="/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/host/bin/gcc" LDSHARED="/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/host/bin/gcc -shared " LDFLAGS="-L/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/host/lib -L/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/hostpkg/lib -L/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/target-aarch64_generic_musl/host/lib" VERSION="2024.10" CPPFLAGS="-O2 -I/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/host/include -I/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/hostpkg/include -I/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/target-aarch64_generic_musl/host/include -I/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/host/include -I/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/hostpkg/include -I/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/target-aarch64_generic_musl/host/include -std=gnu11 -I/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt" OBJDIR=scripts/dtc/pylibfdt SOURCES="/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt/fdt.c /home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt/fdt_ro.c /home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt/fdt_wip.c /home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt/fdt_sw.c /home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt/fdt_rw.c /home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt/fdt_strerror.c /home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt/fdt_empty_tree.c /home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt/fdt_addresses.c /home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt/fdt_overlay.c scripts/dtc/pylibfdt/libfdt.i" SWIG_OPTS="-I/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt -I/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/scripts/dtc/libfdt/.." python3 scripts/dtc/pylibfdt/setup.py --quiet build_ext --inplace
scripts/dtc/pylibfdt/libfdt_wrap.c:171:11: fatal error: Python.h: No such file or directory
  171 | # include <Python.h>
      |           ^~~~~~~~~~
compilation terminated.
error: command '/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/host/bin/gcc' failed with exit code 1
make[6]: *** [scripts/dtc/pylibfdt/Makefile:33: rebuild] Error 1
make[5]: *** [scripts/Makefile.build:397: scripts/dtc/pylibfdt] Error 2
make[4]: *** [Makefile:2055: scripts_dtc] Error 2
make[4]: Leaving directory '/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10'
make[3]: *** [Makefile:353: /home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/build_dir/target-aarch64_generic_musl/u-boot-nanopc-t4-rk3399/u-boot-2024.10/.built] Error 2
make[3]: Leaving directory '/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopc-t4-rk3399/compile#8.76#3.89#11.74
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopc-t4-rk3399).
make[2]: *** [package/Makefile:179: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10'
make[1]: *** [package/Makefile:173: /home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10'
make: *** [/home/ast/build/openwrt/NanoPi_Build_Official/build/openwrt-24.10/include/toplevel.mk:233: world] Error 2

```

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R4S

### Image kind

Self-built image

### Steps to reproduce

https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

closed 2025-02-21 by TyroSpy

peterwillcn, 2025-06-14:
@1715173329 Can we remove the dependency of pylibfdt from u-boot?

1715173329, 2025-06-14:
no