openwrt/openwrt #18062: RPCError RPC call to luci/getFeatures failed with error -32000: Object not found

LaierX, 2025-02-21:
### Describe the bug

RPCError
RPC call to luci/getFeatures failed with error -32000: Object not found
  at ClassConstructor.handleCallReply (http://192.168.1.1/luci-static/resources/rpc.js?v=25.049.66344~2b8e93c:11:3)

[image]

### OpenWrt version

r28470-315facfce6

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

VMware, Inc. VMware Virtual Platform

### Image kind

Official downloaded image

### Steps to reproduce

The system uses ubuntu and debian
Version using 24.10, 23.05
The result is the same after multiple compilation
All of the above problems

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dawnfrost, 2025-02-27:
I got same error after I upgrade some luci packages on 2025/2/27

[image]

LaierX, 2025-03-01:
> I got same error after I upgrade some luci packages on 2025/2/27
> 
> [image]

Reinstall the system. Do not use the docker version of the system, this problem will occur

closed 2025-03-01 by LaierX