openwrt/openwrt #18068: Banana PI-BR4: mt7530 upload speed tricks

andrey-le, 2025-02-21:
### Describe the bug

Hi, seems like mt7530 driver has a bug.

So, here’s the situation:

LAN1 is connected to an access point (ASUS TUF AX6000)
LAN2 is connected to an access point (ASUS TUF AX6000)
LAN3 is connected to a PC
LAN 1-2-3 are bridged

When the PC is turned off and in Wake-On-Lan state, LAN3 switches to 10Mbits mode, and (attention!) incoming connections to LAN1 and LAN2 also implicitly switch to this mode, even though ethtool shows 1Gbits Full-Duplex. As soon as I turn on the PC, everything works at full speed again.

I'm not sure about this fix in linux kernel, maybe something simular should be done in openwrt code:
https://github.com/torvalds/linux/commit/dfa988b4c7c3a48bde7c2713308920c7741fff29

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Steps to reproduce

Run from the access point to BPI-R4 when LAN3 is sleeping: 

root@AX6000_AP_ZAL:~# iperf3 -c 192.168.100.1

Connecting to host 192.168.100.1, port 5201
[  5] local 192.168.100.8 port 59410 connected to 192.168.100.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  2.25 MBytes  18.9 Mbits/sec    0    204 KBytes
[  5]   1.00-2.00   sec  1.62 MBytes  13.6 Mbits/sec    0    264 KBytes
[  5]   2.00-3.00   sec  1.25 MBytes  10.5 Mbits/sec    0    315 KBytes
[  5]   3.00-4.00   sec  1.62 MBytes  13.6 Mbits/sec    0    375 KBytes
....
[  5]   9.00-10.00  sec  1.12 MBytes  9.44 Mbits/sec    0    537 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  14.0 MBytes  11.7 Mbits/sec    0             sender
[  5]   0.00-10.31  sec  11.6 MBytes  9.46 Mbits/sec                  receiver

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

andrey-le, 2025-02-21:
Same situation here:
https://forum.banana-pi.org/t/mt7530-1000-100-mix-strange-behavior/21983

danpawlik, 2025-02-25:
Check https://github.com/openwrt/openwrt/issues/15348

andrey-le, 2025-11-17:
Hi everybody, seems like this bug is fixed somehow, at least for 24.10.4 version.