openwrt/openwrt #18079: bpi-r4: 6ghz not working with US country code

wberrier, 2025-02-23:
### Describe the bug

The forums are reporting that 6ghz works by setting the country code to GB or JP (and to set all of the radios to the same country code).

I can confirm that this works.

Setting to US does not work.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Self-built image

### Steps to reproduce

Set the 6ghz radio to:

`        option country 'US'`

Valid parameters, etc...

### Actual behaviour

Doing a scan from a 6ghz client: the SSID does not show up.

### Expected behaviour

I would expect to be able to set the country code to US, and be able to connect.  Again, GB and JP work as expected.

### Diffconfig

```text
I used the image builder to create this image.
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-23:
What does `iw list` say about channel availability in 6GHz band?
Note switching ETSI/FCC/JP usually disables radar detection in 5GHz, so you usually need to power router down in between.

brada4, 2025-02-23:
It is client-only.
https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/tree/db.txt#n1955

testuser7, 2025-02-23:
Yes, you can't run AP for US because of `NO-IR`.
You can do it for Qualcomm chips because they use internal regdb database.

brada4, 2025-02-23:
You have to campaign to regdb. Reference for 6ghz says indoor operation is permitted at levels from Chapter 59 without extra coordination.

danpawlik, 2025-02-23:
@wberrier you did not provide enough details for your issue.
Please share what wireless card you are using and share also `/etc/config/wireless` config file.

testuser7, 2025-02-23:
@danpawlik Some more details:
- https://forum.openwrt.org/t/adding-openwrt-support-for-zyxel-ex5601-t0/155914/298
- https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/commit/db.txt?id=888da5f48030e45ce14f43a2bf7be07ec79b60d4

brada4, 2025-02-23:
Everything correct, no 6GHz in US.

wberrier, 2025-02-23:
> What does `iw list` say about channel availability in 6GHz band? Note switching ETSI/FCC/JP usually disables radar detection in 5GHz, so you usually need to power router down in between.

This is on the banana pi-r4:

```
# iw list
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 2131 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0xff RX 0xff
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 2 usec (0x04)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: managed
			HE MAC Capabilities (0x08011a000040):
				+HTC HE Supported
				Trigger Frame MAC Padding Duration: 2
				OM Control
				Maximum A-MPDU Length Exponent: 3
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x2270ce920d01f30e4e3f00):
				HE40/2.4GHz
				242 tone RUs/2.4GHz
				Device Class: 1
				LDPC Coding in Payload
				HE SU PPDU with 1x HE-LTF and 0.8us GI
				NDP with 4x HE-LTF and 3.2us GI
				STBC Tx <= 80MHz
				STBC Rx <= 80MHz
				Full Bandwidth UL MU-MIMO
				Partial Bandwidth UL MU-MIMO
				DCM Max Constellation: 2
				DCM Max Constellation Rx: 2
				SU Beamformer
				SU Beamformee
				Beamformee STS <= 80Mhz: 3
				Sounding Dimensions <= 80Mhz: 1
				Codebook Size SU Feedback
				Codebook Size MU Feedback
				Triggered CQI Feedback
[... 998 lines cut ...]
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ FILS_DISCOVERY ]: FILS discovery frame transmission support
		* [ UNSOL_BCAST_PROBE_RESP ]: unsolicated broadcast probe response transmission support
		* [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
		* [ BSS_COLOR ]: BSS coloring support
		* [ POWERED_ADDR_CHANGE ]: can change MAC address while up
	wiphy radio 0:
		freq range: 2400.0 MHz - 2500.0 MHz
		valid interface combinations:
			 * #{ AP, mesh point } <= 16, #{ managed } <= 19,
			   total <= 19, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }

	wiphy radio 1:
		freq range: 5000.0 MHz - 5900.0 MHz
		valid interface combinations:
			 * #{ AP, mesh point } <= 16, #{ managed } <= 19,
			   total <= 19, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }

	wiphy radio 2:
		freq range: 5900.0 MHz - 7200.0 MHz
		valid interface combinations:
			 * #{ AP, mesh point } <= 16, #{ managed } <= 19,
			   total <= 19, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }
```

wberrier, 2025-02-23:
> Everything correct, no 6GHz in US.

Help me understand... (Meaning, I can go down to the electronics store down the street with the option to buy several APs that support 6ghz).

What am I missing?

danpawlik, 2025-02-23:
> > [@wberrier](https://github.com/wberrier) you did not provide enough details for your issue. Please share what wireless card you are using and share also `/etc/config/wireless` config file.
> 
> I'm trying 3 6ghz capable wifi cards:
> 
> * Intel AX210
> * Intel BE200
> * AMD/Mediatek RZ616

Those are clients or you are putting those into bpi-r4? If the second, Intel cards will just work as a client, dunno what about rz616.

brada4, 2025-02-23:
Read the reference document. You have to use authorized authorisation scheme.

rmandrad, 2025-02-26:
This is an upstream issue as US has NO-IR  for the 6g band - I would suggest to create a PR to remove NO-IR 

use https://www.federalregister.gov/documents/2020/05/26/2020-11236/unlicensed-use-of-the-6-ghz-band section 59 to create the PR and provide evidence

https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/log/

[image]

rmandrad, 2025-02-26:
I created an upstream PR to remove the NO-IR, unfortunately there is a technical limitation that doesn't allow to increase the power to 30dBM as 20MHz is not allowed to go over 12dBM!

https://lore.kernel.org/linux-wireless/001e01db8832$844ef130$8cecd390$@gmail.com/T/#t

rmandrad, 2025-03-06:
just a quick update unfortunately the patch will be declined - more details on this link  [https://lists.infradead.org/pipermail/wireless-regdb/2025-March/001698.html](url)

brada4, 2025-03-07:
Ancien kernel is not if concern here.

rmandrad, 2025-03-08:
> Ancien kernel is not if concern here.

agree ... @hauke @nbd168 what is your view ? would an openwrt patch be accepted in this case even if upstream rejects it due to older kernels supportability

gsustek, 2025-03-26:
does anybody know how to solve when 6GHZ is Disabled?

```
		Frequencies:
			* 5935.0 MHz [2] (disabled)
			* 5955.0 MHz [1] (disabled)
			* 5975.0 MHz [5] (disabled)
			* 5995.0 MHz [9] (disabled)
			* 6015.0 MHz [13] (disabled)
			* 6035.0 MHz [17] (disabled)
			* 6055.0 MHz [21] (disabled)
			* 6075.0 MHz [25] (disabled)
			* 6095.0 MHz [29] (disabled)
			* 6115.0 MHz [33] (disabled)
			* 6135.0 MHz [37] (disabled)
			* 6155.0 MHz [41] (disabled)
			* 6175.0 MHz [45] (disabled)
			* 6195.0 MHz [49] (disabled)
			* 6215.0 MHz [53] (disabled)
			* 6235.0 MHz [57] (disabled)
			* 6255.0 MHz [61] (disabled)
			* 6275.0 MHz [65] (disabled)
			* 6295.0 MHz [69] (disabled)
			* 6315.0 MHz [73] (disabled)
			* 6335.0 MHz [77] (disabled)
			* 6355.0 MHz [81] (disabled)
			* 6375.0 MHz [85] (disabled)
			* 6395.0 MHz [89] (disabled)
			* 6415.0 MHz [93] (disabled)
			* 6435.0 MHz [97] (disabled)
			* 6455.0 MHz [101] (disabled)
			* 6475.0 MHz [105] (disabled)
			* 6495.0 MHz [109] (disabled)
			* 6515.0 MHz [113] (disabled)
			* 6535.0 MHz [117] (disabled)
			* 6555.0 MHz [121] (disabled)
			* 6575.0 MHz [125] (disabled)
			* 6595.0 MHz [129] (disabled)
			* 6615.0 MHz [133] (disabled)
			* 6635.0 MHz [137] (disabled)
			* 6655.0 MHz [141] (disabled)
			* 6675.0 MHz [145] (disabled)
			* 6695.0 MHz [149] (disabled)
			* 6715.0 MHz [153] (disabled)
			* 6735.0 MHz [157] (disabled)
			* 6755.0 MHz [161] (disabled)
			* 6775.0 MHz [165] (disabled)
			* 6795.0 MHz [169] (disabled)
			* 6815.0 MHz [173] (disabled)
			* 6835.0 MHz [177] (disabled)
			* 6855.0 MHz [181] (disabled)
			* 6875.0 MHz [185] (disabled)
			* 6895.0 MHz [189] (disabled)
			* 6915.0 MHz [193] (disabled)
			* 6935.0 MHz [197] (disabled)
			* 6955.0 MHz [201] (disabled)
			* 6975.0 MHz [205] (disabled)
			* 6995.0 MHz [209] (disabled)
			* 7015.0 MHz [213] (disabled)
			* 7035.0 MHz [217] (disabled)
			* 7055.0 MHz [221] (disabled)
			* 7075.0 MHz [225] (disabled)
			* 7095.0 MHz [229] (disabled)
			* 7115.0 MHz [233] (disabled)
	valid interface combinations:
		 * #{ managed } <= 1, #{ AP, mesh point } <= 16,
		   total <= 16, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz }

```

brada4, 2025-03-26:
You can discuss your config issues in forum. No globally permitted 6ghz channel, apparently because of your country.

gsustek, 2025-03-26:
i'm in / from Europe. which country has it allowed?

i'm using this reg.bin 
 - https://git.codelinaro.org/clo/ath-firmware/ath12k-firmware/-/blob/main/QCN9274/hw2.0/regdb.bin?ref_type=heads

brada4, 2025-03-27:
Stop stealing others' threads. Ask linaro to support their blobs, we dont do that here.

andrewjlamarche, 2025-03-28:
Hitting this roadblock now in https://github.com/openwrt/openwrt/pull/17869
I can't test 6GHz because of this... so I think a patch is necessary.

brada4, 2026-02-07:
https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/tree/db.txt#n2055
You have to patch away no-ir and rebuild regdb, check it is operational and submit a PR here ;-)