openwrt/openwrt #18083: wireguard-tools: Required dependency package ip is not available in any repository

iDevOps-pl, 2025-02-23:
### Describe the bug

`Details for package wireguard-tools

    Version: 1.0.20210914-r4
    Size: ~84.00 KiB installed
    Dependencies:
        ip Not available
        kmod-wireguard (104.00 KiB) Installed
            kernel (24.00 KiB) Installed
            kmod-crypto-lib-chacha20poly1305 (16.00 KiB) Installed
                kmod-crypto-lib-chacha20 (24.00 KiB) Installed
                kmod-crypto-lib-poly1305 (20.00 KiB) Installed
                    kmod-crypto-hash (8.00 KiB) Installed
            kmod-crypto-lib-curve25519 (36.00 KiB) Installed
                kmod-crypto-kpp (8.00 KiB) Installed
            kmod-udptunnel4 (32.00 KiB) Installed
            kmod-udptunnel6 (16.00 KiB) Installed

Description

WireGuard is a novel VPN that runs inside the Linux Kernel and utilizes state-of-the-art cryptography. It aims to be faster, simpler, leaner, and more useful than IPSec, while avoiding the massive headache. It intends to be considerably more performant than OpenVPN. WireGuard is designed as a general purpose VPN for running on embedded interfaces and super computers alike, fit for many different circumstances. It uses UDP. This package provides the userspace control program for WireGuard, `wg(8)`, a netifd

    Required dependency package ip is not available in any repository.`

### OpenWrt version

r28869-6936deac67

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Steps to reproduce

apk add wireguard-tools
WARNING: customfeeds.list:4: unrecogized keyword: src/gz
ERROR: unable to select packages:
  ip (no such package):
    required by: wireguard-tools-1.0.20210914-r4[ip]

### Actual behaviour

Missing dependency

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

iDevOps-pl, 2025-02-23:
If I wouldnt have it wireguard-tools installed on SNAPSHOT, whats the point to bother anyone. Apparently I had revert from SNAPSHOT to 24.10 due to this bug.

aimacintyre, 2025-02-24:
Probably related to #18059

hnyman, 2025-02-24:
There has been lots of package build breakage since last week. Reported as #18059 

But now fixes have been merged: PR #18073 as fix for autoconf #17825 and automake upgrade  #17827 has been reverted with commit 87f3365d9

It will likely take a day before new SDKs are available for all targets, and then 1-2 additional days to compile packages for all package architectures.