openwrt/openwrt #18085: [mt76x8 | Netgear R6120] Wi-Fi crash or hang during large file transfer through Ethernet, Wi-Fi and refuse connection until router is rebooted

uncharted-user, 2025-02-23:
### Describe the bug

Wi-Fi keeps crashing or hanging while transferring or downloading large files over Ethernet (when Wi-Fi is enabled) and Wi-Fi. Then the router refuses Wi-Fi connection until it is rebooted. I discussed this issue with the community in a [post](https://forum.openwrt.org/t/wifi-is-crashing-during-large-file-transfer-through-ethernet-or-wi-fi-and-refusing-connection-until-router-is-rebooted/224720) which shed some light into the problem. I downgraded to 23.05.4 and discovered that the issue is solved. Furthermore, I experienced a somewhat similar https://github.com/openwrt/openwrt/issues/10074 about a year ago, but it was later solved.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt76x8

### Device

Netgear R6120

### Image kind

Official downloaded image

### Steps to reproduce

**Test 1:** 
1. Connect the testing device through Wi-Fi
2. Visit https://fast.com/
3. Let the first test finish
4. Click on Show more info
5. Click on the Settings
6. Set Test duration (seconds) Min: 300 and Max: 600
7. Click on Save and let the test run
**Test 2:**
1. Connect the testing device through Wi-Fi
2. Transfer multiple files at the same time in total of more than 5 GB using [Local Send](https://localsend.org/) from Device A to Device B. 

### Actual behaviour

The Wi-Fi is crashing or hanging to a state that a full reboot is required.

### Expected behaviour

The Wi-Fi shall not crash or hang, no matter what the load is.

### Additional info

**Logs:
System:**
```
Thu Feb 13 06:37:45 2025 kern.err kernel: [ 2896.523406] mt76x2e 0000:01:00.0: MCU message 1f (seq 10) timed out
Thu Feb 13 06:37:45 2025 kern.info kernel: [ 2896.722688] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Thu Feb 13 06:37:45 2025 kern.info kernel: [ 2896.733677] mt76x2e 0000:01:00.0: Build: 1
Thu Feb 13 06:37:45 2025 kern.info kernel: [ 2896.741793] mt76x2e 0000:01:00.0: Build Time: 201607111443____
Thu Feb 13 06:37:45 2025 kern.info kernel: [ 2896.773470] mt76x2e 0000:01:00.0: Firmware running!
Thu Feb 13 06:37:45 2025 kern.info kernel: [ 2896.786785] ieee80211 phy1: Hardware restart was requested
Thu Feb 13 06:37:46 2025 kern.err kernel: [ 2897.803405] mt76x2e 0000:01:00.0: MCU message 02 (seq 13) timed out
Thu Feb 13 06:37:47 2025 kern.err kernel: [ 2898.603371] mt76x2e 0000:01:00.0: MCU message 1f (seq 14) timed out
Thu Feb 13 06:37:47 2025 kern.info kernel: [ 2898.774712] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Thu Feb 13 06:37:47 2025 kern.info kernel: [ 2898.785677] mt76x2e 0000:01:00.0: Build: 1
Thu Feb 13 06:37:47 2025 kern.info kernel: [ 2898.793859] mt76x2e 0000:01:00.0: Build Time: 201607111443____
Thu Feb 13 06:37:47 2025 kern.info kernel: [ 2898.823402] mt76x2e 0000:01:00.0: Firmware running!
Thu Feb 13 06:37:47 2025 kern.info kernel: [ 2898.834506] ieee80211 phy1: Hardware restart was requested
Thu Feb 13 06:37:49 2025 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED [REDACTED]
```
**Kernel:**
```
[ 2896.523406] mt76x2e 0000:01:00.0: MCU message 1f (seq 10) timed out
[ 2896.722688] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 2896.733677] mt76x2e 0000:01:00.0: Build: 1
[ 2896.741793] mt76x2e 0000:01:00.0: Build Time: 201607111443____
[ 2896.773470] mt76x2e 0000:01:00.0: Firmware running!
[ 2896.786785] ieee80211 phy1: Hardware restart was requested
[ 2897.803405] mt76x2e 0000:01:00.0: MCU message 02 (seq 13) timed out
[ 2898.603371] mt76x2e 0000:01:00.0: MCU message 1f (seq 14) timed out
[ 2898.774712] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 2898.785677] mt76x2e 0000:01:00.0: Build: 1
[ 2898.793859] mt76x2e 0000:01:00.0: Build Time: 201607111443____
[ 2898.823402] mt76x2e 0000:01:00.0: Firmware running!
[ 2898.834506] ieee80211 phy1: Hardware restart was requested
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

IMEVER, 2025-03-02:
The same problem

wenqingj, 2025-03-13:
The log tells phy1 (instead of phy0). Is it possible that the old ticket (#10071) fixes phy0 (2.5 GHz), and this ticket happens only on phy1 (5 GHz)?

uncharted-user, 2025-03-14:
> The log tells phy1 (instead of phy0). Is it possible that the old ticket ([#10074](https://github.com/openwrt/openwrt/issues/10074)) fixes phy0 (2.5 GHz), and this ticket happens only on phy1 (5 GHz)?

I disabled that intensionally, and it happens on both frequency.

IMEVER, 2025-03-18:
I config maximum txpower to 20db，now it works someday ok. Maybe you can try it.

uncharted-user, 2025-03-19:
> I config maximum txpower to 20db

Thank you for the workaround. 

> now it works someday ok

So it's like hit or miss? Did you observe any pattern when it crashes?

IMEVER, 2025-03-22:
> So it's like hit or miss? Did you observe any pattern when it crashes?

Before that setting,  when i backup my phone to cloud with large data upload, the 5g wifi lost
very soon.
But now I make a test to backup my phone, everything works fine.

uncharted-user, 2025-03-24:
> > So it's like hit or miss? Did you observe any pattern when it crashes?
> 
> Before that setting, when i backup my phone to cloud with large data upload, the 5g wifi lost very soon. But now I make a test to backup my phone, everything works fine.

Interesting. So, now you are using 24.10.0.

Vera2016, 2025-04-08:
same problem,  large data upload or download, 5g wifi lost

wenqingj, 2025-04-08:
It appears that MT76x8 is 802.11n on 2.5G, while MT76x2E is 802.11ac on 5G. So if this happens only on 5G then it's related to MT76x2E. Not sure if it's the same cause found (and fixed) in MT76x8 before in ticket #10074.

uncharted-user, 2025-04-08:
> So if this happens only on 5G then it's related to MT76x2E. Not sure if it's the same cause found (and fixed) in MT76x8 before in ticket [#10074](https://github.com/openwrt/openwrt/issues/10074).

It happens on both.

wenqingj, 2025-04-08:
> It happens on both.
Are you able to disable (or remove) all the 5G wifi and test on 2.5G (802.11n) only?
I ask because I've a router with MT76x8 802.11n only and it runs a whole month now with OpenWRT 24.10 without the slightest issue (and also without any issue previously with OpenWRT 23.5.05). I'm speculating on 5GHz heat (or overheat) ...

uncharted-user, 2025-04-08:
> Are you able to disable (or remove) all the 5G wifi and test on 2.5G (802.11n) only? I ask because I've a router with MT76x8 802.11n only and it runs a whole month now with OpenWRT 24.10 without the [...]

Yes, and I can guarantee you that it is not a heating issue on my end ([post](https://forum.openwrt.org/t/wifi-is-crashing-during-large-file-transfer-through-ethernet-or-wi-fi-and-refusing-connection-until-router-is-rebooted/224720))

wenqingj, 2025-04-08:
>Yes, and I can guarantee you that it is not a heating issue on my end

So that implies that with 2.5GHz only (and 5GHz band disabled), the issue still happens (due to some cause not related to heating)?

uncharted-user, 2025-04-08:
> So that implies that with 2.5GHz only (and 5GHz band disabled), the issue still happens (due to some cause not related to heating)?

Exactly, I disassembled it and checked manually. All the heat pads are good as new.

closed 2025-04-19 by bulshittttttttttttttttttttttttttttttt