openwrt/openwrt #18087: Can not boot OpenWrt 24.10.0 on OrangePi Zero3

liaoya, 2025-02-24:
### Describe the bug

I try both `openwrt-24.10.0-sunxi-cortexa53-xunlong_orangepi-zero3-ext4-sdcard.img.gz` and `openwrt-24.10.0-sunxi-cortexa53-xunlong_orangepi-zero3-squashfs-sdcard.img.gz`, it can not start properly.

I get the following with UART

```text
U-Boot SPL 2024.01-OpenWrt-r28427-6df0e3d02a (Feb 03 2025 - 23:09:37 +0000)
DRAM: 2048 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.10.0  (release):OpenWrt v2.10-1 (sunxi-h616)
NOTICE:  BL31: Built : 23:09:37, Feb  3 2025
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0b19a0, model: OrangePi Zero3
ERROR:   RSB: set run-time address: 0x10003

U-Boot 2024.01-OpenWrt-r28427-6df0e3d02a (Feb 03 2025 - 23:09:37 +0000) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: OrangePi Zero3
DRAM:  2 GiB (effective 15.5 EiB)
```

But my board has 1.5G memory, it works well with Armbian. The following is Armbian boot message

```text
U-Boot SPL 2024.01-armbian-2024.01-S866c-P667c-Ha9af-V5c1c-Bda0a-R448a (Dec 07 2024 - 04:18:43 +0000)
DRAM base address is defined as 0x40000000
DRAM has 16 b/raw, 10 b/col, 4 B/width, 1 #rank and 8 #bank
DRAM top address must be less than 0x80000000
DRAM top address must be less than 0x60000000
DRAM: 1536 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.10.2(debug):armbian
NOTICE:  BL31: Built : 04:18:32, Dec  7 2024
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0b2eb8, model: OrangePi Zero3
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
ERROR:   RSB: set run-time address: 0x10003
INFO:    Could not init RSB: -65539
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for erratum 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for erratum 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.
ns16550_serial serial@5000000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19
```

### OpenWrt version

24.10.0

### OpenWrt release

6df0e3d02a

### OpenWrt target/subtarget

sunxi-cortexa53

### Device

xunlong_orangepi-zero3

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

liaoya, 2025-02-26:
I change the invalid parameter and how can it be changed to valid

wigyori, 2025-02-26:
It should be noted that this is linked to the 1.5Gbyte/12Gbit memory mis-detection issue. This seems to be fixed in later u-boot versions (i.e. 2025.01), but on the H618 boards that's suffering from an issue described in https://github.com/openwrt/openwrt/pull/17993 . I do not have an H618 board at the moment to test on but will try to source one.

liaoya, 2025-02-27:
@wigyori I can test it on my board

TMTBO, 2025-03-30:
I have the same problem. Is there any new progress?

wigyori, 2025-03-31:
I have just got my hands on a Zero2W (H618, 4Gb), so stay tuned for a few days.

wigyori, 2025-04-01:
@TMTBO - do you also have the one with 1.5GByte memory or some other config?

For the record, my 2W with uboot-2025.01 also fails to boot (2024.01 was fine), so we'll take it from there.

wigyori, 2025-04-02:
@TMTBO / @liaoya - please test this branch: https://github.com/wigyori/openwrt/tree/h618-202504 , thanks.

TMTBO, 2025-04-07:
> [@TMTBO](https://github.com/TMTBO) / [@liaoya](https://github.com/liaoya) - please test this branch: https://github.com/wigyori/openwrt/tree/h618-202504 , thanks.

Thank you for your outstanding work. Is it possible to compile an image that can be written directly? I have no compilation environment here. If possible, I can test it on my zero3 1.5G on this Saturday.

wigyori, 2025-04-12:
If this still applies I can build you an image with this u-boot later today.

wigyori, 2025-04-12:
> Thank you for your outstanding work. Is it possible to compile an image that can be written directly? I have no compilation environment here. If possible, I can test it on my zero3 1.5G on this [...]

Images for the zero2w / zero3:
https://drive.google.com/file/d/1oRI6sgCBBHp1of9uB3yTNzuBCghu0-jc/view?usp=sharing
https://drive.google.com/file/d/1-h8aNUYyCWeRilV9t_YcCOAYuBYvjKT6/view?usp=sharing

TMTBO, 2025-04-13:
> Images for the zero2w / zero3: https://drive.google.com/file/d/1oRI6sgCBBHp1of9uB3yTNzuBCghu0-jc/view?usp=sharing https://drive.google.com/file/d/1-h8aNUYyCWeRilV9t_YcCOAYuBYvjKT6/view?usp=sharing

It still doesn't seem to be working

[image]

liaoya, 2025-04-14:
@wigyori Thank you for your hard work. I am afraid that official uboot `2025.01` does not solve this problem. I also tried https://sd-card-images.johang.se/ before. I succeeded boot zero2 image but failed on zero3 image. It use official uboot `2025.01` also
But the armbian can start

```
U-Boot SPL 2024.01-armbian-2024.01-S866c-P4a40-H8869-V380a-Bb703-R448a (Mar 03 2025 - 03:55:00 +0000)
DRAM base address is defined as 0x40000000
DRAM has 16 b/raw, 10 b/col, 4 B/width, 1 #rank and 8 #bank
DRAM top address must be less than 0x80000000
DRAM top address must be less than 0x60000000
DRAM: 1536 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.10.2(debug):armbian
NOTICE:  BL31: Built : 03:54:47, Mar  3 2025
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0b2eb8, model: OrangePi Zero3
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
ERROR:   RSB: set run-time address: 0x10003
INFO:    Could not init RSB: -65539
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for erratum 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for erratum 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.
ns16550_serial serial@5000000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19

U-Boot 2024.01-armbian-2024.01-S866c-P4a40-H8869-V380a-Bb703-R448a (Mar 03 2025 - 03:55:00 +0000) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: OrangePi Zero3
DRAM:  1.5 GiB
Core:  57 devices, 25 uclasses, devicetree: separate
WDT:   Not starting watchdog@30090a0
MMC:   mmc@4020000: 0
Loading Environment from FAT... Unable to use mmc 0:1...
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Allwinner mUSB OTG (Peripheral)
Net:   Unsupported value 13, using default (13)
Unsupported value 13, using default (13)
eth0: ethernet@5020000using musb-hdrc, OUT ep1out IN ep1in STATUS ep2in
MAC de:ad:be:ef:00:01
HOST MAC de:ad:be:ef:00:00
RNDIS ready
, eth1: usb_ether
starting USB...
Bus usb@5200000: USB EHCI 1.00
Bus usb@5200400: USB OHCI 1.0
```

cc @TMTBO

TMTBO, 2025-04-14:
I find something about this, maybe it will be helpful for your guys.

wigyori, 2025-04-14:
Thanks for confirming it doesn't work on the 1.5Gb boards. Yeah, I came across that TRIM_SIZE patch earlier, but IIRC it was dropped.

There are two patches out there which are supposed to improve the DRAM scan procedure on H616 (and H618), let me make a build with those included and ask you to test.

wigyori, 2025-04-14:
Round 2:
https://drive.google.com/file/d/1-Hf3M-Qz-tPiMrzGCbpB1UFtYgZRzGej/view?usp=sharing
https://drive.google.com/file/d/1QQb_MaaINp0oAhbGC55tiEMyVC0fq9iI/view?usp=sharing

Let's see how these work.

liaoya, 2025-04-15:
@wigyori 
It does not work and the same error. Can you check the uboot is updated. It print UBoot build time is old.

```
U-Boot SPL 2025.01-OpenWrt-r29139-fef6a3957b (Apr 02 2025 - 10:44:05 +0000)
DRAM: 2048 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.10.0  (release):OpenWrt v2.10-1 (sunxi-h616)
NOTICE:  BL31: Built : 10:44:05, Apr  2 2025
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0b5348, model: OrangePi Zero3
ERROR:   RSB: set run-time address: 0x10003

U-Boot 2025.01-OpenWrt-r29139-fef6a3957b (Apr 02 2025 - 10:44:05 +0000) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: OrangePi Zero3
DRAM:  2 GiB (effective 14.6 EiB)
```

WangBen137, 2025-08-22:
Is there any new progress?My zero3 1.5G has same question.

liaoya, 2026-01-29:
> https://github.com/armbian/build/blob/main/patch/u-boot/u-boot-sunxi/opizero3-1.5GB-trim-from-u-boot-v2024.01.patch here is the patch for u-boot-v2024.01
> 
> [immortalwrt-sunxi-cortexa53-xunlong_orangepi-zero3-squashfs-sdcard.img.gz](https://github.com/user-attachments/files/21948291/immortalwrt-sunxi-cortexa53-xunlong_orangepi-zero3-squashfs-sdcard.img.gz) for test

<https://github.com/user-attachments/files/21948291/immortalwrt-sunxi-cortexa53-xunlong_orangepi-zero3-squashfs-sdcard.img.gz> does not work, the following is the serial output, why the kernel is `4.9.170`

```text
[5954]HELLO! BOOT0 is starting!
[5957]BOOT0 commit : 3ae35eb
[5960]set pll start
[5963]periph0 has been enabled
[5966]set pll end
[5969]unknow PMU
[5971]unknow PMU
[5975]PMU: AXP1530
[5980]dram return write ok
[5982]board init ok
[5984]DRAM BOOT DRIVE INFO: V0.651
[5988]the chip id is 0x2000
[5991]chip id check OK
[6001]DRAM_VCC set to 1100 mv
[6004]DRAM CLK =792 MHZ
[6007]DRAM Type =8 (3:DDR3,4:DDR4,7:LPDDR3,8:LPDDR4)
[6018]Actual DRAM SIZE =1536 M
[6021]DRAM SIZE =1536 MBytes, para1 = 310a, para2 = 6000000, dram_tpr13 = 2056c61
[6034]DRAM simple test OK.
[6037]rtc standby flag is 0x0, super standby flag is 0x0
[6042]dram size =1536
[6045]Use rtc to store dram tuning para
[6050]spinor id is: 5e 40 18, read cmd: 0b
[6055]Succeed in reading toc file head.
[6059]The size of toc is 9c000.
[6216]Entry_name        = u-boot
[6222]Entry_name        = monitor
[6226]Entry_name        = dtb
[6230]Jump to second Boot.
NOTICE:  BL3-1: v1.0(debug):05d6c57
NOTICE:  BL3-1: Built : 13:35:35, 2021-10-28
NOTICE:  BL3-1 commit: 8
NOTICE:  cpuidle init version V2.0
ERROR:   Error initializing runtime service tspd_fast
NOTICE:  BL3-1: Preparing for EL3 exit to normal world
NOTICE:  BL3-1: Next image address = 0x4a000000
▒OTICE:  BL3-1: Next image spsr = 0x1d3

U-Boot 2018.05 (Jul 10 2023 - 10:32:18 +0800) Allwinner Technology

[06.308]CPU:   Allwinner Family
[06.311]Model: sun50iw9
I2C:   ready
[06.316]DRAM:  1.5 GiB
[06.319]Relocation Offset is: 55f5d000
[06.339]secure enable bit: 0
[06.341]pmu_axp152_probe pmic_bus_read fail
[06.345]PMU: AXP1530
[06.350]CPU=1008 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=400Mhz
[06.357]gic: sec monitor mode
[06.362]flash init start
[06.364]workmode = 0,storage type = 3
SF: Detected zb25vq128as with page size 256 Bytes, erase size 64 KiB, total 16 MiB
[06.381]sunxi flash init ok
[06.385]Loading Environment from SUNXI_FLASH... OK
[06.414]get secure storage map err
[06.417]sunxi secure storage is not supported
[06.421]usb burn from boot
delay time 0
weak:otg_phy_config
[06.435]usb prepare ok
[06.649]usb sof ok
[06.651]usb probe ok
[06.653]usb setup ok
set address 0x2
set address 0x2 ok
[09.659]do_burn_from_boot usb : have no handshake
[09.663]get secure storage map err
[09.666]secure storage init fail
[09.671]update dts
[09.675]update part info
[09.679]update bootcmd
[09.681]No ethernet found.
Hit any key to stop autoboot:  0
Android's image name: sun50i_arm64
[13.216]Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.170 (orangepi@ubuntu) (gcc version 5.3.1 20160412 (Linaro GCC 5.3-2016.05) ) #15 SMP PREEMPT Thu Jun 29 17:36:03 CST 2023
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] bootconsole [earlycon0] enabled
```

leesonaa, 2026-02-02:
> [@leesonaa](https://github.com/leesonaa) It works!!
>
> Has Immortalwrt fixed this issue already?

ImmortalWRT does not have this patch. This patch comes from the Armbian community; if the author commit it to the upstream U-Boot, it can be merged.

leesonaa, 2026-04-24:
@TMTBO 我设备没了.你ttl看看跑的uboot是不是我前面图中那样的吧.

TMTBO, 2026-04-24:
> [@TMTBO](https://github.com/TMTBO) 橙子家的zero3属于薛定谔的设备 一会儿能起来 一会儿起不来 . 起不来的话凉会儿又能起来了.重要的还是接个ttl看看跑码是啥情况

难搞, 我之ttl线找不到了, 我明天再找找看看能不能找到吧.

aiamadeus, 2026-04-27:
@TMTBO Does your board have 1.5GB of RAM?

aiamadeus, 2026-05-05:
> I tried this image. Maybe something wrong with this image

@TMTBO `md5sum 53fcc24fc2fd394a5a9ec28fbe4d2bc1`
Can you use the following command to write the image?
```
gzip -d openwrt-sunxi-cortexa53-xunlong_orangepi-zero3-squashfs-sdcard.img.gz
dd if=openwrt-sunxi-cortexa53-xunlong_orangepi-zero3-squashfs-sdcard.img of=xx
```