openwrt/openwrt #18088: CM5 Wifi doesn't work. Can't reach wifi device.

uysalerinc, 2025-02-24:
### Describe the bug

[image]

[image]
Using CM5 wıth IOBoard. Openwrt cant use or find wifi device.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

broadcom/bcm2712

### Device

Raspberry Pi CM5

### Image kind

Self-built image

### Additional info

I changed on target/linux/bcm27xx/image/Makefile Device/rpi-5 defination device_dts to broadcom/bcm2712-rpi-cm5-cm5io to work with cm5 io board.

[image]

### Diffconfig

```text
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2712=y
CONFIG_TARGET_bcm27xx_bcm2712_DEVICE_rpi-5=y
CONFIG_ARIA2_BITTORRENT=y
CONFIG_ARIA2_NOXML=y
CONFIG_ARIA2_OPENSSL=y
CONFIG_ARIA2_SFTP=y
CONFIG_ARIA2_WEBSOCKET=y
CONFIG_BRCMSMAC_USE_FW_FROM_WL=y
CONFIG_DOCKER_STO_BTRFS=y
# CONFIG_FEED_telephony is not set
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_IMAGEOPT=y
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_IO_URING is not set
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
CONFIG_KERNEL_KEXEC=y
# CONFIG_KERNEL_NET_L3_MASTER_DEV is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
[... 499 lines cut ...]
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://downloads.openwrt.org/releases/23.05.2"
CONFIG_VERSION_SUPPORT_URL=""
# CONFIG_WPA_MBO_SUPPORT is not set
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
CONFIG_PACKAGE_kmod-hid=y
CONFIG_PACKAGE_kmod-hid-generic=y
CONFIG_PACKAGE_kmod-input-evdev=y
CONFIG_PACKAGE_kmod-lib-crc-ccitt=y
# CONFIG_PACKAGE_kmod-mppe is not set
CONFIG_PACKAGE_kmod-ppp=y
# CONFIG_PACKAGE_kmod-ppp-synctty is not set
# CONFIG_PACKAGE_kmod-pppoa is not set
CONFIG_PACKAGE_kmod-pppoe=y
# CONFIG_PACKAGE_kmod-pppol2tp is not set
CONFIG_PACKAGE_kmod-pppox=y
# CONFIG_PACKAGE_kmod-pptp is not set
CONFIG_PACKAGE_kmod-slhc=y
# CONFIG_VERSION_CODE_FILENAMES is not set
# CONFIG_VERSION_FILENAMES is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-24:
Shkd be in luci/network/wireless or `iw list`

uysalerinc, 2025-02-24:
outout of `iw list`
[image]

brada4, 2025-02-24:
lsusb ? lspci? Any signs in boot logs?  And post text as formatted cofe using `<>` button

brada4, 2025-02-24:
Since you are not using OpenWrt you must check with c2tech (whatever it is) or repeat the issue with generic openwrt obtainable via firmware-selector.openwrt.org.

Kazagumo, 2025-02-24:
could you post dmesg here?

brada4, 2025-02-24:
CM5 is not yet supported by official OpenWrt.  Ask at point of sales to give you a hand

uysalerinc, 2025-02-25:
`root@OpenWrt:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.69 xhci-hcd xHCI Host Controller
Bus 002 Device 001: ID 1d6b:0003 Linux 6.6.69 xhci-hcd xHCI Host Controller
Bus 003 Device 001: ID 1d6b:0002 Linux 6.6.69 xhci-hcd xHCI Host Controller
Bus 004 Device 001: ID 1d6b:0003 Linux 6.6.69 xhci-hcd xHCI Host Controller
Bus 005 Device 001: ID 1d6b:0002 Linux 6.6.69 dwc2_hsotg DWC OTG Controller`
`root@OpenWrt:~# lspci
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 30)
01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge`
`root@OpenWrt:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.6.69 (buildbot@9af8aee8c1db) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28532-3deeb7805f) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Wed Jan  8 03:41:35 2025
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 5 Rev 1.0
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000001000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000001000000..0x0000000004ffffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd16480..0x000000003fd164fe (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xff7984c0-0xff79afff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fbfffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000] On node 0, zone DMA: 1024 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s36776 r8192 d28760 u73728
[    0.000000] pcpu-alloc: s36776 r8192 d28760 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: SSBS not fully self-synchronizing
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=2C:CF:67:B8:B1:16 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  root=PARTUUID=0370a051-02 rootfstype=squashfs,ext4 rootwait
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] mempolicy: NUMA default policy overridden to 'interleave:0'
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1031168
[    0.000000] Policy zone: DMA
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 4029480K/4190208K available (9152K kernel code, 990K rwdata, 2624K rodata, 512K init, 542K bss, 95192K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000133] Console: colour dummy device 80x25
[... 393 lines cut ...]
[    8.156192] usbcore: registered new interface driver option
[    8.161800] usbserial: USB Serial support registered for GSM modem (1-port)
[    8.169384] usbcore: registered new interface driver qcserial
[    8.175159] usbserial: USB Serial support registered for Qualcomm USB modem
[    8.181762] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.txt failed with error -2
[... cut ...]
[   13.272026] br-lan: port 1(eth0) entered forwarding state
[   15.247710] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   15.255595] NFSD: Using legacy client tracking operations.
[   15.261113] NFSD: starting 10-second grace period (net f0000000)
[   18.794760] overlay: filesystem on /opt/docker/check-overlayfs-support3964195710/upper not supported as upperdir`

i know cm5 is not supported yet. but i can compile it for cm5 and only issue is CM5's wifi doesnt work. I want to make this work to contrubute this project :(

brada4, 2025-02-25:
CM5 is NOT supported by official OpenWrt. You should ask at the place of purchase (who bought raspberry stores empty so none is available for hobbyists who could port OpenWrt to this device). No change here can fix your 3rd party firmware.

uysalerinc, 2025-02-25:
i solved my problem. I just create symlink of `/lib/firmware/cypress/cyfmac43455-sdio.bin` to /`lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,5-compute-module.bin` and symlink of `/lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt `to /`lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,5-compute-module.txt`

closed 2025-02-25 by uysalerinc