openwrt/openwrt #18095: [BUG] OpenWrt UEFI Boot Failure on Proxmox VE (QEMU) with OVMF

sydnkj, 2025-02-25:
### Describe the bug

#### Problem Description  
When deploying OpenWrt on Proxmox VE 8.3.4 (QEMU) with UEFI configuration (**Machine: `q35`**, **BIOS: OVMF**), the UEFI-compatible OpenWrt images fail to boot. Observed behavior:  
- **Both official and self-compiled UEFI images** (e.g., `openwrt-24.10.0-x86-64-generic-ext4-combined-efi.img.gz`) cannot be booted (no bootable device detected).  
- Using **SeaBIOS** allows the boot process to start, but the system freezes after printing excessive logs during initialization.  
- **Non-EFI images** (e.g., `*-combined-squashfs.img.gz`) work without issues.  

---

#### Steps to Reproduce  
1. **Proxmox VE 8.3.4 Setup**:  
   - Create a VM with:  
     - Machine Type: `q35`  
     - BIOS: `OVMF (UEFI)`  
     - Disk: Convert OpenWrt UEFI image to VM disk (e.g., `qm importdisk`).  
     - Network: VirtIO.
   - Start the VM.  
2. **Result with OVMF**:  
   - VM fails to boot, showing `No bootable device`.  
3. **Switch to SeaBIOS**:  
   - VM starts booting but freezes during kernel initialization (logs attached).  
4. **Non-EFI Image Test**:  
   - VM boots and operates normally.  

---

#### Environment Details  
- **Proxmox VE Version**: 8.3.4  
- **QEMU Version**: 9.2 
- **OpenWrt Version**: 24.10.0 (official release and self-compiled from `master` branch)  
- **Firmware Type**: `x86-64-generic` (UEFI and non-UEFI tested)  
- **OVMF Version**: `edk2-ovmf_4.2023.08-4` (default Proxmox package)  

---

#### Expected Behavior  
OpenWrt UEFI images should boot successfully with OVMF configuration.  

---

#### Actual Behavior  
- **OVMF Mode**: Boot fails immediately (`No bootable device`).  
- **SeaBIOS Mode**: Boot starts but freezes during kernel logs.  
- **Non-EFI Image**: Works as expected.  

---

#### Additional Information  
1. **Troubleshooting Attempted**:  
   - Verified image integrity via SHA256 checksum.  
   - Tested different OVMF versions (Proxmox default and manually compiled).  
   - Adjusted VM settings (enabled/disabled TPM, Secure Boot, varied disk controllers).  
2. **Logs**:  
   - SeaBIOS boot logs showing freeze point (attached).  
   - Proxmox VM configuration screenshots (attached).  
3. **Related Discussions**:  
   - [OpenWrt Forum Thread](https://forum.openwrt.org/) (similar UEFI issues).  
   - [Proxmox Bug Tracker](https://bugzilla.proxmox.com/) (QEMU/OVMF boot behavior).  

---

#### Potential Causes  
1. **Missing EFI Bootloader**: OpenWrt UEFI image might lack proper `EFI/BOOT/BOOTx64.EFI` files.  
2. **OVMF Compatibility Issues**: Conflicts with OpenWrt kernel drivers (e.g., VirtIO).  
3. **Proxmox/QEMU Configuration**: Incorrect NVRAM settings or EFI disk partitioning.  

---

#### Suggested Fixes  
1. Validate UEFI image structure (ensure EFI boot files are included).  
2. Provide Proxmox-specific UEFI deployment documentation for OpenWrt.  
3. Add QEMU/OVMF testing to OpenWrt CI pipelines.  

---

**Note**: Happy to provide further details or test patches if needed!  

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-25:
Please get a text log using virtio serial console in BIOS mode.
For EFI you have to disable SecureBoot.

sydnkj, 2025-02-25:
Sure, I disabled the SecureBoot.
This is the text log.
```
  Booting a command list

[    0.000000] Linux version 6.6.73 (builder@buildhost) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28427-6df0e3d02a) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Mon Feb  3 23:09:37 2025
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=11458228-0839-40e2-fe5c-7d7fc0445102 rootwait console=ttyS0,115200 noinitrd
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003ffd2fff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ffd3000-0x000000003fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b0000000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.16.3-0-ga6ed6b701f0a-prebuilt.qemu.org 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: using sched offset of 49420121888 cycles
[    0.000001] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000003] tsc: Detected 2599.996 MHz processor
[    0.000610] last_pfn = 0x3ffd3 max_arch_pfn = 0x400000000
[    0.000637] MTRR map: 4 entries (3 fixed + 1 variable; max 19), built from 8 variable MTRRs
[    0.000639] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.006487] found SMP MP-table at [mem 0x000f5430-0x000f543f]
[    0.006499] Using GB pages for direct mapping
[    0.006617] ACPI: Early table checksum verification disabled
[    0.006619] ACPI: RSDP 0x00000000000F5260 000014 (v00 BOCHS )
[    0.006626] ACPI: RSDT 0x000000003FFE533E 00003C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006630] ACPI: FACP 0x000000003FFE5064 0000F4 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006637] ACPI: DSDT 0x000000003FFDF040 006024 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006640] ACPI: FACS 0x000000003FFDF000 000040
[    0.006643] ACPI: APIC 0x000000003FFE5158 000080 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006646] ACPI: SSDT 0x000000003FFE51D8 0000CA (v01 BOCHS  VMGENID  00000001 BXPC 00000001)
[    0.006649] ACPI: HPET 0x000000003FFE52A2 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006652] ACPI: MCFG 0x000000003FFE52DA 00003C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006655] ACPI: WAET 0x000000003FFE5316 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.006657] ACPI: Reserving FACP table memory at [mem 0x3ffe5064-0x3ffe5157]
[    0.006658] ACPI: Reserving DSDT table memory at [mem 0x3ffdf040-0x3ffe5063]
[    0.006659] ACPI: Reserving FACS table memory at [mem 0x3ffdf000-0x3ffdf03f]
[    0.006659] ACPI: Reserving APIC table memory at [mem 0x3ffe5158-0x3ffe51d7]
[    0.006660] ACPI: Reserving SSDT table memory at [mem 0x3ffe51d8-0x3ffe52a1]
[    0.006660] ACPI: Reserving HPET table memory at [mem 0x3ffe52a2-0x3ffe52d9]
[    0.006661] ACPI: Reserving MCFG table memory at [mem 0x3ffe52da-0x3ffe5315]
[    0.006662] ACPI: Reserving WAET table memory at [mem 0x3ffe5316-0x3ffe533d]
[    0.006702] Zone ranges:
[    0.006703]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.006704]   DMA32    [mem 0x0000000001000000-0x000000003ffd2fff]
[    0.006705]   Normal   empty
[    0.006706] Movable zone start for each node
[    0.006707] Early memory node ranges
[    0.006707]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.006708]   node   0: [mem 0x0000000000100000-0x000000003ffd2fff]
[    0.006709] Initmem setup node 0 [mem 0x0000000000001000-0x000000003ffd2fff]
[    0.006721] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.006822] On node 0, zone DMA: 97 pages in unavailable ranges
[    0.011214] On node 0, zone DMA32: 45 pages in unavailable ranges
[    0.014114] ACPI: PM-Timer IO Port: 0x608
[    0.014123] ACPI: LAPIC_NMI (acpi_id[0xff] dfl dfl lint[0x1])
[    0.014160] IOAPIC[0]: apic_id 0, version 17, address 0xfec00000, GSI 0-23
[    0.014163] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.014165] ACPI: INT_SRC_OVR (bus 0 bus_irq 5 global_irq 5 high level)
[    0.014165] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
[    0.014166] ACPI: INT_SRC_OVR (bus 0 bus_irq 10 global_irq 10 high level)
[    0.014167] ACPI: INT_SRC_OVR (bus 0 bus_irq 11 global_irq 11 high level)
[... 633 lines cut ...]
[    1.349282] hv_vmbus: registering driver hv_balloon
[    1.350883] NET: Registered PF_INET6 protocol family
[    1.352312] Segment Routing with IPv6
[    1.353079] In-situ OAM (IOAM) with IPv6
[    1.353856] NET: Registered PF_PACKET protocol family
[    1.354756] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[... cut ...]
[    1.539328] ata4: SATA link down (SStatus 0 SControl 300)
[    1.540326] ata1: SATA link down (SStatus 0 SControl 300)
[    1.542027] Waiting for root device PARTUUID=11458228-0839-40e2-fe5c-7d7fc0445102...
[    1.570024] usb 3-1: new high-speed USB device number 2 using ehci-pci
[    1.773709] input: QEMU QEMU USB Tablet as /devices/pci0000:00/0000:00:1d.7/usb3/3-1/3-1:1.0/0003:0627:0001.0001/input/input4
[    1.776087] hid-generic 0003:0627:0001.0001: input,hidraw0: USB HID v0.01 Mouse [QEMU QEMU USB Tablet] on usb-0000:00:1d.7-1/input0
```

sydnkj, 2025-02-25:
> Please get a text log using virtio serial console in BIOS mode. For EFI you have to disable SecureBoot.

Sure, I disabled the SecureBoot when i using EFI.

This is the text log:

[log.txt](https://github.com/user-attachments/files/18969443/log.txt)

Canary233, 2025-02-25:
Have you tried specifying the boot device?
[image]

brada4, 2025-02-25:
Act on the last line - just remove "USB tablet device"? Not much tablet tapping/mouse clicking on OpenWrt console.
Please stop posting random pictures. They tell nothing about general non-essential hardware. And boot without PCI passthrough.

Canary233, 2025-02-25:
> Act on the last line - just remove "USB tablet device"? Not much tablet tapping/mouse clicking on OpenWrt console. Please stop posting random pictures. They tell nothing about general non-essential hardware. And boot without PCI passthrough.

Oh, I just wanted to make it clear that the VM needs to specify a boot device, such as a virtual disk, which has nothing to do with PCI pass-through

brada4, 2025-02-25:
Does vm boot without passthrough?
Does other linux boot with same passthrough?

Check if vm host and bios is in good state

[    0.532937] pci_bus 0000:09: extended config space not accessible

sydnkj, 2025-02-26:
> Have you tried specifying the boot device? [image]

yeah, the first boot order is scsi0

sydnkj, 2025-02-26:
> Act on the last line - just remove "USB tablet device"? Not much tablet tapping/mouse clicking on OpenWrt console. Please stop posting random pictures. They tell nothing about general non-essential hardware. And boot without PCI passthrough.

no pci passthrough device added. Then I also unchecked USB TABLE. Problems still exist

sydnkj, 2025-02-26:
I use openwrt-23.05 version and everything is OK. only 24.10 has this problem.

hululu1068, 2025-03-24:
Observation shows that `/boot` is mounted repeatedly, and the file structure under the `/boot` directory does not seem to be a standard `efi` boot structure.

[image]

brada4, 2025-03-24:
In that case you are using BIOS image and unjustly complaining about EFI images. There was never an issue to begin with.
STOP POSTING SCREENSHOTS OF TEXT