openwrt/openwrt #18097: How to port strongswan6.0.0 in openwrt and support PQC post-quantum algorithm by compiling

ate-wang, 2025-02-25:
How to upgrade the current openwrt version of strongswan5.9.2 to version 6.0.0, and compile it to support pqc related algorithms, how to achieve it, does anyone have relevant experience, can you share it?

brada4, 2025-02-25:
Start here
https://github.com/openwrt/packages/tree/master/net/strongswan
Change version
if something does not build add/remove patches
Add the module you need
Rinse repeat.
(this is wrong tracker to ask for advice, as no code change in this repository could ever fill your wish)
Try forum.openwrt.org, you can login with github federated login.

ate-wang, 2025-02-25:
> 从这里开始 https://github.com/openwrt/packages/tree/master/net/strongswan 如果某些东西没有构建， 请更改版本添加/删除补丁 添加您需要的模块 冲洗重复。 （这是错误的跟踪器来寻求建议，因为这个存储库中的任何代码更改都无法满足您的愿望） 尝试 forum.openwrt.org，您可以使用 github 联合登录登录。
It seems that strongswan6.0.0 cannot be used. This may be because the openwrt cross-compilation tool does not support C++20, but botan3.6.0 is developed based on C++20. Is there any big guy who can solve this problem?

brada4, 2025-02-25:
So you can work in 2 directions at once - with one hand bump strongswan version, with other hand get botan library compiled (watch for byte order issues). 99.9% you succeeed at 1st thing, if you manage to compile botan you can build swan botan plugins?

ate-wang, 2025-02-26:
> 因此，您可以同时从两个方向开展工作 - 一方面改进 strongswan 版本，另一方面编译 botan 库（注意字节顺序问题）。99.9% 您在第一件事上就成功了，如果您设法编译 botan，那么您可以构建 swan botan 插件吗？
Isn't the swan botan plugin a library developed based on botan3.6.0? However, botan3.6.0 uses C++20, but the openwrt cross-compilation tool does not support C++20, so the botan library cannot be compiled. Can swan botan be built without the botan library? I don't know how to do this specifically.

ate-wang, 2025-02-27:
> Your auto translations make no sense. Repetition of claim will not remove c++20 support from gcc, it is same gcc with same features as anywhere else.
Have you tried to use strongswan6.0.0 in openwrt? I feel that this is not easy. When I open the botan library, I will encounter the problem that botan-2 and botan-3 are not supported.