openwrt/openwrt #18098: High RX Packet Drops and reduced bandwidth on PPPoE with Nokia ONT X-S-010X-Q

aricade, 2025-02-25:
### Describe the bug

Connecting to ISP with PPPOE (Fibre) with a Nokia ONT X-S-010X-Q demarc.
The required settings were to setup want interface on Vlan Id 40 with a priority of 0

runing a `ifconfig -a` shows large amount of dropped RX packets for eth1:

```
eth1      RX packets:1610209 errors:0 dropped:37450 overruns:0 frame:0
eth1.40   RX packets:1572759 errors:0 dropped:0 overruns:0 frame:0
pppoe-wan RX packets:1572734 errors:0 dropped:32487 overruns:0 frame:0

```
**Expected Behavior**

**Observed Behavior**
Download speeds start at ~450-550 Mbps but improve to 750-800 Mbps after switching drivers (kmod-r8169 to kmod-r8125-rss).
Upload speeds are consistently ~900 Mbps.

**Drivers:**

- Initially: kmod-r8169
- Switched to: kmod-r8125-rss
RX packets droppped keep increasing, even after the driver change.

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R5C

### Image kind

Official downloaded image

### Steps to reproduce

`/etc/config/network` snippets:

```

config device
        option name 'eth1'
        option macaddr 'DE:AD:BE:EF:BA:D0'

config interface 'wan'
        option device 'eth1.40'
        option proto 'pppoe'
        option username '****username here*****'
        option password '****password here*****'
        option ipv6 'auto'
        option mtu '1452'

config device
        option type '8021q'
        option ifname 'eth1'
        option vid '40'
        option priority '0'
        option name 'eth1.40'

```

### Actual behaviour

ETH1 drops many RX packets 
Download speeds start at ~450-550 Mbps but improve to 750-800 Mbps after switching drivers (kmod-r8169 to kmod-r8125-rss). 
Upload speeds are consistently ~900 Mbps(no upload issues).

### Expected behaviour

Very low RX packets dropped on ETH1.  Ideally 0 packets...
I should achieve close to 1 Gbps download as advertised by the ISP and suspect the RX packets dropped are impacting this.

### Additional info

```
 > dmesg | grep r8125
[    7.728455] r8125 Ethernet controller driver 9.014.01-NAPI-RSS loaded
[    7.729172] r8125 0001:01:00.0: enabling device (0000 -> 0003)
[    8.088697] r8125 0001:01:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[    8.089554] r8125 0001:01:00.0 (unnamed net_device) (uninitialized): Random ether addr f6:1f:02:2c:d8:42
[    8.090937] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    8.094019] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com>
[    8.096853] r8125 Ethernet controller driver 9.014.01-NAPI-RSS loaded
[    8.097544] r8125 0002:01:00.0: enabling device (0000 -> 0003)
[    8.098621] r8125 0002:01:00.0: no MSI/MSI-X. Back to INTx.
[    8.452950] r8125 0002:01:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[    8.453806] r8125 0002:01:00.0 (unnamed net_device) (uninitialized): Random ether addr 0e:85:3b:d9:5f:5f
[    8.455155] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    8.458176] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com>
[   11.896450] r8125 0001:01:00.0 eth0: entered allmulticast mode
[   11.897173] r8125 0001:01:00.0 eth0: entered promiscuous mode
[   15.323439] r8125: eth0: Link is Up - 1Gbps/Full
[   16.147100] r8125 0002:01:00.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, (Receiver ID)
[   16.147930] r8125 0002:01:00.0:   device [10ec:8125] error status/mask=00000001/0000e000
[   16.148655] r8125 0002:01:00.0:    [ 0] RxErr
[   16.359612] r8125: eth1: Link is Up - 2.5Gbps/Full
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Noltari, 2025-02-25:
@aricade did you try with `kmod-r8125` instead of `kmod-r8125-rss`?

grobian, 2025-03-02:
FWIW, I see the same with an OpenWrt One after switching from an ER-4 on this ONT.  The amount of dropped packets is MUCH higher, but it doesn't seem to affect my connection too much (but I'm capped at ~200Mbps).

aricade, 2025-03-03:
> [@aricade](https://github.com/aricade) did you try with `kmod-r8125` instead of `kmod-r8125-rss`?

Sorry for the late reply, I have been busy but I will give your suggestion some investigation and a try.

aricade, 2025-03-03:
> FWIW, I see the same with an OpenWrt One after switching from an ER-4 on this ONT. The amount of dropped packets is MUCH higher, but it doesn't seem to affect my connection too much (but I'm capped at ~200Mbps).

No for the most part I lose 200 to 300 Mbps on Download which is not noticeable practically speaking. I was considering moving to 500Mbps service as it would be cheaper, and I theorize that it would have less issues.