openwrt/openwrt #18100: WAN port link autonegotiation problem on 24.10

wein72, 2025-02-25:
### Describe the bug

After upgrading to 24.10 I found my ISP link not working.
As a result of some diagnostics, I found that there's a problem with link speed autonegotiation and it is set to 1000Mbits, but my ISP's hardware doesn't support such link speed. At 23.05 it has been always successfully autonegotiated at 100Mbits. 

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

Beeline SmartBox TURBO+

### Image kind

Official downloaded image

### Steps to reproduce

Connect 100Mbit ISP link to WAN port (ISP hardware doesn't support 1000Mbit link)

### Actual behaviour

Link speed sets up to 1000Mbit and nothing works.
If I connect some real 1000Mb link to wan port, it works fine and a packets are going (I can get DHCP etc).

### Expected behaviour

Link speed must be autonegotiated at 100Mbit and everything must work (like it was on 23.05)

### Additional info

I found such behavior right after upgrading from 23.05, where everything worked fine.
I tried to fix this manually and even got some success, but why it doesn't work as expected out of box?
I installed ethtool and just set link mode manually. My workaround looks like this (I've just added this to my rc.local):
**ethtool -s wan autoneg off speed 100 duplex full && ifdown wan && sleep 3 && ifup wan**
 

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

wein72, 2025-02-25:
> Is it not this [#15348](https://github.com/openwrt/openwrt/issues/15348) ?

Yes, looks very similar.

vadim-brave, 2025-03-03:
I had similar issue  #17371 

My PC on Windows don't receive IP from router

As result - Unrecognised network and internet connection unavailable

Linux PC works as on 23.05.5

vadim-brave, 2025-03-03:
> > Is it not this [#15348](https://github.com/openwrt/openwrt/issues/15348) ?
> 
> Yes, looks very similar.

I had similar issue  #17371 

My PC on Windows don't receive IP from router

As result - Unrecognised network and internet connection unavailable

Linux PC works as on 23.05.5

BUG-reporter, 2025-03-10:
The same as on my AC-85P: https://github.com/openwrt/openwrt/issues/17711

nrobert13, 2026-03-29:
@Neustradamus , sorry for chiming in, I tried today 25.12.2 on EA8500 / ipq806x and the autonegotiation still not working.