openwrt/openwrt #18107: IPv6 address distribution is slow and Windows devices are not working properly

Canary233, 2025-02-25:
### Describe the bug

After a simple test, using the BCM57810 10 Gigabit Ethernet Virtual Function as the br-lan for OpenWrt will cause the LAN device to distribute IPv6 slowly (about 2-3 minutes), and the Windows device will not be able to use IPv6 (even if the global unicast address has been obtained), and the ping command will report that the target host cannot be reached or the request times out. This is not the case with a virtual NIC that bridges to the same network adapter or use physical function directly as a br-lan. I don't know if it's a driver issue or a system issue, as this issue is not reproducible in iKuaiOS.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-02-25:
virtual functions do not encapsulate vlans well often.

Canary233, 2025-02-25:
> virtual functions do not encapsulate vlans well often.

Im not using any vlan.

schuettecarsten, 2025-02-28:
Maybe...?
https://github.com/openwrt/odhcpd/issues/230
https://github.com/openwrt/odhcpd/issues/229