openwrt/openwrt #18108: Usb on FritzBox 7490 can be fixed

sateuwdie, 2025-02-25:
### Describe the bug

On stable and snapshot release for Avm FritzBox 7490 works all (except fxs ports and isdn but is not a serious problem atm). The usb3 controller is not working

```
[    0.359179] usbcore: registered new interface driver usbfs
[    0.364729] usbcore: registered new interface driver hub
[    0.370034] usbcore: registered new device driver usb
[   74.749426] xhci_hcd 0000:01:00.0: failed to load firmware renesas_usb_fw.mem: -145
```

on this forum

https://forum.openwrt.org/t/no-way-to-get-usb-working-on-my-router-using-snapshot/224529/11

A user suggest a solution which seems to be working

"Copy renesas_usb_fw.mem to target/linux/lantiq/files/firmware

Then edit kernel config file:
target/linux/lantiq/xrx200/config-6.6

change:
CONFIG_EXTRA_FIRMWARE="lantiq/xrx200_phy11g_a14.bin lantiq/xrx200_phy11g_a22.bin lantiq/xrx200_phy22f_a14.bin lantiq/xrx200_phy22f_a22.bin"

to:
CONFIG_EXTRA_FIRMWARE="renesas_usb_fw.mem lantiq/xrx200_phy11g_a14.bin lantiq/xrx200_phy11g_a22.bin lantiq/xrx200_phy22f_a14.bin lantiq/xrx200_phy22f_a22.bin"

Other user report usb is now working

lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 001: Dev 003, If 0, Class=[unknown], Driver=usb-storage, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 5000M

### OpenWrt version

r28797-302c717aa3

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

lantiq/xrx200

### Device

AVM FRITZ!Box 7490 (Micron NAND)

### Image kind

Official downloaded image

### Steps to reproduce

Simply try to use usb controller.

### Actual behaviour

No usb working, kernel dmesg give error even you put correct firmware in /lib/firmware

### Expected behaviour

Usb working.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-02-26:
The [FB7490 device page](https://openwrt.org/toh/avm/fritz.box.7490) documents that the USB file needs to be manually copied during installation (step 13), presumably because this file is not legally redistributable hence OpenWrt cannot package it for inclusion in images.  Unless a legally redistributable USB firmware file can be identified so OpenWrt can package it, this situation won't change.

sateuwdie, 2025-02-26:
Copying the file don't work

Let's see on my router

```
file /lib/firmware/renesas_usb_fw.mem 
/lib/firmware/renesas_usb_fw.mem: data
ls -lhd /lib/firmware/renesas_usb_fw.mem
-rw-r--r--    1 root     root       12.7K Feb 11 08:04 /lib/firmware/renesas_usb_fw.mem
```

results

```
lsusb 
unable to initialize libusb: -99
```

```
dmesg |grep usb
[    0.359106] usbcore: registered new interface driver usbfs
[    0.364665] usbcore: registered new interface driver hub
[    0.369968] usbcore: registered new device driver usb
[    9.073205] pci 0000:01:00.0: quirk_usb_early_handoff+0x0/0x9d4 took 5615674 usecs
[   74.779502] xhci_hcd 0000:01:00.0: failed to load firmware renesas_usb_fw.mem: -145
```

sateuwdie, 2025-02-26:
Of course after the copy I have reboot

UncleQuarl, 2026-02-15:
> fyi
> 
> ... in 25.12 snaphsot these two lines make the same:
> 
> ```
> # rmmod xhci_pci_renesas
> # modprobe xhci_pci_renesas
> ```
> 
> https://forum.openwrt.org/t/port-to-avm-fritz-box-3490/52692/102

Thanks, frank-44, for your feedback.
I installed version 24.10.03 because it's more stable than the later versions, at least on this fairly advanced device. More importantly, it doesn't lose two of the four Ethernet ports.