openwrt/openwrt #18113: Extreme 3935 Will not boot openwrt (24.10), root partition error shown.

L-Ances, 2025-02-26:
### Describe the bug

log: https://pasteimg.com/image/image-2025-02-24-214326275.nEcxj
Have attempted multiple flashes of the image, resetting env and using guides that have worked for others, listed below.
Used guides:https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=3aef61060e3f51aa43fe494d5ff173e81dd43003
https://forum.openwrt.org/t/adding-openwrt-support-for-extreme-network-ap3935i/87234/87 

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10

### OpenWrt target/subtarget

ipq806x

### Device

Extreme 3935i

### Image kind

Official downloaded image

### Steps to reproduce

set env vars to those seen in guides above, tftpboot image, run sysupgrade

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

L-Ances, 2025-02-26:
@stangri would love to know if there were any updates for you with full 24.10 release?

aimacintyre, 2025-02-27:
Please close this issue now that you have opened a new version.

closed 2025-02-27 by L-Ances