openwrt/openwrt #18123: Extreme 3935 Will not boot openwrt (24.10), root partition error shown.

L-Ances, 2025-02-26:
### Describe the bug

log: https://pasteimg.com/image/image-2025-02-24-214326275.nEcxj
Have attempted multiple flashes of the image, resetting env and using guides that have worked for others, listed below.
Used guides:https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=3aef61060e3f51aa43fe494d5ff173e81dd43003
https://forum.openwrt.org/t/adding-openwrt-support-for-extreme-network-ap3935i/87234/87

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ipq806x/generic

### Device

Extreme AP 3935i

### Image kind

Official downloaded image

### Steps to reproduce

set env vars to those seen in guides above, tftpboot image, run sysupgrade

### Actual behaviour

file system Root Partition error consistently.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

L-Ances, 2025-02-26:
@stangri apologies for the re tagging, as I messed up previous post, but please by all means chime in if there have been any updates as I know you were having issues with the 3935's aswell. Thanks!

grische, 2025-06-02:
@L-Ances I have never seen this before, but I have successfully flashed a few devices using our tool [extremeflash](https://github.com/grische/extremeflash). It doesn't use the same parameters as those specified in the commit, but alternative ones that seem to work at least on the models I saw. Maybe it helps.