openwrt/openwrt #18132: Boot failure after upgrade from 23.05.5 to 24.10.0 on NETGEAR R8000p

etulupov, 2025-02-27:
### Describe the bug

After upgrade device is not started. It's continuously rebooted.

```
[    2.240710] Internal error: synchronous external abort: 0000000096000210 [#1] SMP
[    2.248367] Modules linked in: ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug(O) usbcore nls_base usb_common crc32c_generic
[    2.263601] CPU: 0 PID: 94 Comm: kworker/u5:3 Tainted: G           O       6.6.73 #0
[    2.271574] Hardware name: Netgear R8000P (DT)
[    2.276144] Workqueue: events_unbound 0xffffffc0800e09fc
[    2.281608] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    2.288775] pc : 0xffffffc078b9aad0 [ohci_hcd@(____ptrval____)+0x7000]
[    2.295495] lr : 0xffffffc078b9ac80 [ohci_hcd@(____ptrval____)+0x7000]
[    2.302217] sp : ffffffc080d7bb60
[    2.305621] x29: ffffffc080d7bb60 x28: 0000000000000000 x27: ffffff800186dee8
[    2.312967] x26: ffffffc078b85000 x25: ffffff800186d800 x24: ffffffc078b50128
[    2.320315] x23: 0000000000000080 x22: 0000000000000014 x21: ffffffc078b5c000
[    2.327662] x20: ffffff800186d800 x19: ffffff800186da60 x18: 0000000000000000
[    2.335010] x17: 7375622064656e67 x16: 69737361202c6465 x15: ffffffffffffffff
[    2.342357] x14: ffffff8001c4b40d x13: ffffff8001c4b40b x12: 00000000ffffffea
[    2.349704] x11: 00000000ffffefff x10: 0000000000000062 x9 : 00000000ffffffd0
[    2.357051] x8 : 0000000000000073 x7 : ffffffc080d7bb50 x6 : 0000000000000075
[    2.364399] x5 : 0000000000000081 x4 : 00000000ffff8bad x3 : ffffff800186ddc0
[    2.371746] x2 : ffffff800186ddd0 x1 : 0000000000000000 x0 : ffffffc08008d404
[    2.379094] Call trace:
[    2.381602]  0xffffffc078b9aad0 [ohci_hcd@(____ptrval____)+0x7000]
[    2.387963]  0xffffffc078b9ac80 [ohci_hcd@(____ptrval____)+0x7000]
[    2.394325]  0xffffffc078b370d4 [usbcore@(____ptrval____)+0x23000]
[    2.400685]  0xffffffc078b7d398 [ohci_platform@(____ptrval____)+0x1000]
[    2.407496]  0xffffffc0804a50ac
[    2.410721]  0xffffffc0804a2660
[    2.413946]  0xffffffc0804a2840
[    2.417172]  0xffffffc0804a2930
[    2.420398]  0xffffffc0804a2c70
[    2.423623]  0xffffffc0800e0a2c
[    2.426850]  0xffffffc0800d410c
[    2.430076]  0xffffffc0800d4574
[    2.433301]  0xffffffc0800dc1f0
[    2.436526]  0xffffffc080093a50
[    2.439754] Code: 912ce000 95de9bb2 17ffffa8 91001000 (b9400000) 
[    2.446024] ---[ end trace 0000000000000000 ]---
[    2.450775] Kernel panic - not syncing: synchronous external abort: Fatal exception
[    2.458658] SMP: stopping secondary CPUs
[    2.462692] Kernel Offset: disabled
[    2.466274] CPU features: 0x0,00000000,00000000,0000400b
[    2.471739] Memory Limit: none
[    2.474876] Rebooting in 1 seconds..

```

Full boot logs:
[boot.log](https://github.com/user-attachments/files/19006111/boot.log)

### OpenWrt version

24.04.0

### OpenWrt release

24.04.0

### OpenWrt target/subtarget

bcm4908

### Device

netgear-r8000p

### Image kind

Official downloaded image

### Steps to reproduce

Perform sys-upgrade from 23.05.5 to 24.10.0

### Actual behaviour

Device is soft-bricked, not boot successfully.

### Expected behaviour

Successful upgrade and running device

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

halflife-2022, 2025-03-15:
I meet the same problem ,and the R8000P keeps restarting after I try to upgrade from 23.05.5 to 24.10.0. Anyone can give  me the CFE files of the R8000P please?

etulupov, 2025-04-21:
@halflife-2022 I used https://github.com/jclehner/nmrpflash utility for unbrick my router.

etulupov, 2025-10-24:
I prepared a custom build of firmware without any usb kernel modules and device is boot up successfully.