openwrt/openwrt #18142: tools/llvm-bpf: compilation fails

Timon48, 2025-03-01:
llvm-bpf fails to compile in snapshot 

[2250/3807] /home/debian/openwrt/staging_dir/host/bin/g++ -DGTEST_HAS_RTTI=0 -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/build/tools/clang/lib/Sema -I/home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/clang/lib/Sema -I/home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/clang/include -I/home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/build/tools/clang/include -I/home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/build/include -I/home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/llvm/include -O2 -I/home/debian/openwrt/staging_dir/host/include -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werror=date-time -fno-lifetime-dse -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-nonnull -Wno-class-memaccess -Wno-redundant-move -Wno-pessimizing-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wsuggest-override -Wno-comment -Wno-misleading-indentation -Wctad-maybe-unsupported -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -fno-strict-aliasing -DNDEBUG -std=c++17  -fno-exceptions -funwind-tables -fno-rtti -MD -MT tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaExpr.cpp.o -MF tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaExpr.cpp.o.d -o tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaExpr.cpp.o -c /home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/clang/lib/Sema/SemaExpr.cpp
In file included from /home/debian/openwrt/staging_dir/host/include/endian.h:5,
                 from /home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/llvm/include/llvm/ADT/bit.h:33,
                 from /home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/llvm/include/llvm/Support/MathExtras.h:16,
                 from /home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/llvm/include/llvm/ADT/APInt.h:19,
                 from /home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/llvm/include/llvm/ADT/APFloat.h:18,
                 from /home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/clang/include/clang/AST/APNumericStorage.h:12,
                 from /home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/clang/include/clang/AST/Decl.h:16,
                 from /home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/clang/lib/Sema/CoroutineStmtBuilder.h:16,
                 from /home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/clang/lib/Sema/TreeTransform.h:16,
                 from /home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/clang/lib/Sema/SemaExpr.cpp:13:
/home/debian/openwrt/staging_dir/host/include/byteswap.h:2:2: warning: #include_next is a GCC extension
    2 | #include_next <byteswap.h>
      |  ^~~~~~~~~~~~
/home/debian/openwrt/staging_dir/host/include/endian.h:6:2: warning: #include_next is a GCC extension
    6 | #include_next <endian.h>
      |  ^~~~~~~~~~~~
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:71: /home/debian/openwrt/build_dir/host/llvm-project-18.1.8.src/.built] Error 1
make[3]: Leaving directory '/home/debian/openwrt/tools/llvm-bpf'
time: tools/llvm-bpf/compile#83518.81#16959.82#21842.92
    ERROR: tools/llvm-bpf failed to build.
make[2]: *** [tools/Makefile:232: tools/llvm-bpf/compile] Error 1
make[2]: Leaving directory '/home/debian/openwrt'
make[1]: *** [tools/Makefile:228: /home/debian/openwrt/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyyynynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/home/debian/openwrt'
make: *** [/home/debian/openwrt/include/toplevel.mk:233: world] Error 2

brada4, 2025-03-01:
> 100x g++: fatal error: Killed signal terminated program cc1plus

This means your build environment is corrupted (the host gcc-*) or a container allocated insufficient resources.
Whos Sema? Why dont you fill issue template to get any help with it?

geldot, 2025-03-03:
> > 100x g++: fatal error: Killed signal terminated program cc1plus
> 
> This means your build environment is corrupted (the host gcc-*) or a container allocated insufficient resources. Whos Sema?

Sema is LLVM terminology for the semantic analysis layer in the compiler frontend. No idea what it means in the context of this bug report however.

brada4, 2025-03-03:
I dont see causal link between file name and debian`'s g++ feeling jealous.