openwrt/openwrt #18143: netavark package missing in 24.10

lucaluse, 2025-03-01:
### Describe the bug

`Installing netavark v1.12.2 (/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/netavark-1.12.2)
    Updating crates.io index
     Locking 319 packages to latest compatible versions
      Adding clap v4.4.18 (available: v4.5.31)
      Adding clap_builder v4.4.18 (available: v4.5.31)
      Adding clap_derive v4.4.7 (available: v4.5.28)
      Adding netlink-packet-route v0.20.1 (available: v0.21.0)
      Adding nftables v0.4.1 (available: v0.6.1)
      Adding prost v0.12.6 (available: v0.13.5)
      Adding rand v0.8.5 (available: v0.9.0)
      Adding sysctl v0.5.5 (available: v0.6.0)
      Adding tonic v0.11.0 (available: v0.12.3)
      Adding tonic-build v0.11.0 (available: v0.12.3)
      Adding tower v0.4.13 (available: v0.5.2)
      Adding zbus v3.15.2 (available: v5.5.0)
      Adding zvariant_utils v1.0.1 (available: v1.1.1)
error: failed to compile `netavark v1.12.2 (/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/netavark-1.12.2)`, intermediate artifacts can be found at `/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/netavark-1.12.2/target`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.

Caused by:
  failed to download `iptables v0.5.3`

Caused by:
  unable to get packages from source

Caused by:
  failed to parse manifest at `/builder/shared-workdir/build/sdk/dl/cargo/registry/src/index.crates.io-6f17d22bba15001f/iptables-0.5.3/Cargo.toml`

Caused by:
  feature `edition2024` is required

  The package requires the Cargo feature called `edition2024`, but that feature is not stabilized in this version of Cargo (1.84.0 (66221abde 2024-11-19)).
  Consider trying a newer version of Cargo (this may require the nightly release).
  See https://doc.rust-lang.org/nightly/cargo/reference/unstable.html#edition-2024 for more information about the status of this feature.
make[3]: *** [Makefile:49: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/netavark-1.12.2/.built] Error 101
time: package/feeds/packages/netavark/compile#1.22#0.60#3.41
`

### OpenWrt version

n/a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

n/a

### Device

n/a

### Image kind

Official downloaded image

### Steps to reproduce

Details: 
https://downloads.openwrt.org/releases/faillogs-24.10/aarch64_cortex-a53/packages/netavark/compile.txt 
https://downloads.openwrt.org/releases/faillogs-24.10/x86_64/packages/netavark/compile.txt

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-03-01:
Is your package located anywhere in this repository?

lucaluse, 2025-03-01:
opened in correct repo: https://github.com/openwrt/packages/issues/26075

closed 2025-03-01 by lucaluse