openwrt/openwrt #18144: Fritz!box (non-International) on Annex A adsl

acoul, 2025-03-01:
### Describe the bug

using openwrt 23.05 or 24.10 on an non-International Fritz!box connecting to an Annex A adsl fails

### OpenWrt version

r16847-f8282da11e

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

lantiq/xrx200

### Device

AVM FRITZ!Box 3370 

### Image kind

Official downloaded image

### Steps to reproduce

using openwrt 23.05 or 24.10 on an non-International Fritz!box connecting to an Annex A adsl fails

### Actual behaviour

using openwrt 23.05 or 24.10 on an non-International Fritz!box connecting to an Annex A adsl fails

### Expected behaviour

I am opening this ticket as a reference of my findings

using openwrt on an non-International (Annex B) Fritz!box WLAN 3370 the box is unable to connect on an Annex A adsl

using [freetz-ng](https://github.com/Freetz-NG/freetz-ng) on the same hardware, it connects fine on an Annex A adsl

searching the web about this issue I found a [discussion about soft-switching DSL Annex](https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg41145.html)

on the freetz-ng the dsl_control command is:
`dsl_control -i04_00_04_00_4C_01_04_07 -f/var/dsl/dspfw/vr9-A-dsl.bin -n/etc/dsl/notify/dsl_notify.sh`
the vr9-A-dsl.bin version is: 5.7.1.E.1.1

on the openwrt side the vdsl_cpe_control command is:
`/sbin/vdsl_cpe_control -i05_01_04_00_4C_01_04_00 -n /sbin/dsl_notify.sh -f /lib/firmware/vr9-A-dsl_7490_7.60-5.9.0.D.0.1.bin -M1 -T1:0x1:0x1_1:0x1:0x1 -a /tmp/dsl.scr -A /tmp/dsl.scr`

I have also tried the vr9-A-dsl-5_7_1_E_1_1.bin version on openwrt without a success

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-03-02:
This query would have been better raised on the OpenWrt forums as more ADSL users are likely to see it there.  Annex A ADSL apparently has been reported to work on `xrx200` devices based on forum postings over the years..

That said, at the very least you need to extract a copy of `/etc/config/network` from the device and provide it here either pasting it into a comment (with code formatting) or as an attachment, redacting any usernames/passwords.  If you create a forum thread, please also provide the contents of this file there as well.

pkgadd, 2025-03-02:
As aimacintyre suggested, this is better handled in the forum - especially as I don't think there is any bug to begin with.

> non-International Fritz!box

I read this as, you bought a Fritz!Box originally sold on the German market…
If so, the hardware is designed for Annex B, not Annex A and can't do the later. The difference is on the analogue side (bandfilters) and the SOC's bootstrapping configuration, it woun't do Annex A.

acoul, 2025-03-19:
> [@acoul](https://github.com/acoul) Please come back to forum with this. https://forum.openwrt.org/t/avm-fritzbox-lantiq-vrx200-and-adsl2/180440/2

thank you nicefile, [I did follow your suggestion](https://forum.openwrt.org/t/avm-fritzbox-lantiq-vrx200-and-adsl2/180440/8?u=acoul)

I've bought two 3370 Fritz!boxes from ebay. one of them accepts only the International freetz-ng image and the other one only the German one. so far, this is the only way for me to figure out which unit is the International & which the German one

on the International box I get the following error when I flash the `3370_06.56.ger_freetz-ng-25676M-842abee830_20250319-164312.image`
[freetz-ng-german-reboot.txt](https://github.com/user-attachments/files/19344317/freetz-ng-german-reboot.txt)

while when uploading the `3370_06.54.int_freetz-ng-25676M-842abee830_20250319-160613.image` everything goes well
[freetz-ng-int_OK.txt](https://github.com/user-attachments/files/19344335/freetz-ng-int_OK.txt)

this is a work in progress as currently I have no access to a physical adsl Annex-a to do further testing, but I will follow-up as soon as I gain such access