openwrt/openwrt #18147: Missing packages for mediatek/filogic imagebuilder Sat Mar 1 2025

nailend, 2025-03-02:
### Describe the bug

Packages are missing and therefore no image can be build.

### OpenWrt version

r28924-7560af7647

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

cudy_wr3000h-v1

### Image kind

Self-built image

### Steps to reproduce

Download imagebuilder (Sat Mar 1 21:57:25 2025): https://downloads.cdn.openwrt.org/snapshots/targets/mediatek/filogic/openwrt-imagebuilder-mediatek-filogic.Linux-x86_64.tar.zst 

make image PROFILE='cudy_wr3000h-v1' PACKAGES='collectd-mod-interface collectd-mod-iwinfo collectd-mod-load collectd-mod-memory collectd-mod-network collectd-mod-ping collectd-mod-uptime ethtool ip6tables-nft iptables-nft iperf3 iwinfo ip kmod-nft-bridge mtr nftables tcpdump-mini vnstat -ppp -ppp-mod-pppoe libiwinfo-lua luci-mod-admin-full luci-proto-ipv6 luci-theme-bootstrap px5g-mbedtls rpcd-mod-rrdns uhttpd uhttpd-mod-ubus -dnsmasq -odhcpd-ipv6only -odhcp6c -iptables -ip6tables -firewall -firewall4 irqbalance dawn luci-app-dawn umdns ip-bridge' DISABLED_SERVICES='olsrd6 vnstat'

### Actual behaviour

This is the error message:

Generate local signing keys...
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
read EC key
writing EC key
Checking 'true'... ok.
Checking 'false'... ok.
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'perl-data-dumper'... ok.
Checking 'perl-findbin'... ok.
Checking 'perl-file-copy'... ok.
Checking 'perl-file-compare'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'perl-ipc-cmd'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... updated.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... updated.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... updated.
Checking 'egrep'... updated.
Checking 'getopt'... ok.
Checking 'realpath'... ok.
Checking 'stat'... ok.
Checking 'gzip'... updated.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'install'... ok.
Checking 'perl'... ok.
Checking 'python'... updated.
Checking 'python3'... updated.
Checking 'python3-distutils'... ok.
Checking 'python3-stdlib'... ok.
Checking 'file'... ok.
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.
Building images for mediatek - Cudy WR3000H v1
Packages: collectd-mod-interface collectd-mod-iwinfo collectd-mod-load collectd-mod-memory collectd-mod-network collectd-mod-ping collectd-mod-uptime ethtool ip6tables-nft iptables-nft iperf3 iwinfo ip kmod-nft-bridge mtr nftables tcpdump-mini vnstat libiwinfo-lua luci-mod-admin-full luci-proto-ipv6 luci-theme-bootstrap px5g-mbedtls rpcd-mod-rrdns uhttpd uhttpd-mod-ubus irqbalance dawn luci-app-dawn umdns ip-bridge apk-mbedtls base-files ca-bundle dropbear fitblk fstools kernel kmod-crypto-hw-safexcel kmod-gpio-button-hotplug kmod-leds-gpio kmod-mt7915e kmod-mt7981-firmware kmod-nft-offload kmod-phy-aquantia libc libgcc libustream-mbedtls logd mt7981-wo-firmware mtd netifd nftables procd-ujail uboot-envtools uci uclient-fetch urandom-seed urngd wpad-basic-mbedtls

WARNING: opening /home/julian/bbb-configs/tmp/build/k9ap-24/packages/packages.adb: No such file or directory
OK: 0 MiB in 0 packages
Package list missing or not up-to-date, generating it.

Building package index...

Installing packages...
ERROR: unable to select packages:
  collectd-mod-interface (no such package):
    required by: world[collectd-mod-interface]
  collectd-mod-iwinfo (no such package):
    required by: world[collectd-mod-iwinfo]
  collectd-mod-load (no such package):
    required by: world[collectd-mod-load]
  collectd-mod-memory (no such package):
    required by: world[collectd-mod-memory]
  collectd-mod-network (no such package):
    required by: world[collectd-mod-network]
  collectd-mod-ping (no such package):
    required by: world[collectd-mod-ping]
  collectd-mod-uptime (no such package):
    required by: world[collectd-mod-uptime]
  ip (no such package):
    required by: world[ip]
  ip-bridge (no such package):
    required by: world[ip-bridge]
  mtr (no such package):
    required by: world[mtr]
  nftables (no such package):
    required by: world[nftables]
  libnftnl11 (no such package):
    required by: xtables-nft-1.8.10-r1[libnftnl11]
make[2]: *** [Makefile:226: package_install] Error 15
make[1]: *** [Makefile:157: _call_image] Error 2
make: *** [Makefile:332: image] Error 2

### Expected behaviour

A properly build image.

### Additional info

Maybe this is connected to this:
Packages buildbot had several workers that were out of disk space. 
https://lists.openwrt.org/pipermail/openwrt-devel/2025-February/043795.html

### Diffconfig

```text
#!/bin/sh
make ./scripts/config/conf >/dev/null || { make ./scripts/config/conf; exit 1; }
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep \^CONFIG_TARGET_DEVICE_ .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_KMODS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_NONSHARED=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_BUSYBOX_CUSTOM=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TARGET_PER_DEVICE_ROOTFS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

davidandreoletti, 2025-03-02:
FYI: Similar issue for NanoPi R4s.

robimarko, 2025-03-02:
This should be already fixed as A53 packages finally got successfully rebuilt after over a week, so a lot of packages were missing and now are present again.

I tried using your reproduction steps and it worked for me just fine.

As for Nginx, no idea why is that only failing for A53 as I fixed the patch couple of days ago and when using SDK it builds just fine.

davidandreoletti, 2025-03-02:
@robimarko 

`nginx` package builds fine now. However building an image continues to fails (which is what this Github issue is about) with up to date `SNAPSHOTS`. Is there a github issue tracking this already (except this one) ? Looks like https://github.com/Vladdrako/openwrt/commit/4bc2d5e2c21fc801b8d7c4741c9806631cff13a4 should be fixing the issue but it has not so far for me.

robimarko, 2025-03-02:
@davidandreoletti ImageBuilder is working now as packages are available, I literally tried it this morning with the exact command that this issue reported failing.

closed 2025-03-03 by nailend