openwrt/openwrt #18148: libpcap can't handle platform specific DSA tags in current version

Fei1Yang, 2025-03-02:
### Describe the bug

There is a regression in the libpcap library which makes many packet capturing software (e.g. tcpdump) no longer working.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 3G

### Image kind

Official downloaded image

### Steps to reproduce

Run the following commands:

```
opkg update
opkg install tcpdump
tcpdump
```

### Actual behaviour

`tcpdump` shows captured packets

### Expected behaviour

The following error message shows up: `unsupported DSA tag: mtk`

### Additional info

This regression is fixed by upstream at https://github.com/the-tcpdump-group/libpcap/pull/1463, may this be included as a patch before a new version is released?

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-03-02:
How it is a regression when those tags were never supported.
mtk_soc_eth is built in kernel and it is used by default by tcpdump as being oldest. You can specify any other interface just fine.

brada4, 2025-03-05:
https://github.com/openwrt/openwrt/commit/fad94e8cda573389164827dc74e583d10eeae2e5

robimarko, 2025-03-07:
Should be fixed with: https://github.com/openwrt/openwrt/commit/fad94e8cda573389164827dc74e583d10eeae2e5

closed 2025-03-07 by robimarko