openwrt/openwrt #18157: wireguard: Initialization of second tunnel often fails after flash/reboot

schuettecarsten, 2025-03-03:
I have two wireguard tunnels, vpn_wg0 and vpn_wg1. Sometimes, mostly after flashing a new firmware but also after rebooting the router, vpn_wg0 initializes successfully, but vpn_wg1 gets stuck in a state where no data is transferred. Transfer counters stay at 0. The only thing that helps is to reboot the device again.

This _might_ be caused by concurrent initialization of both wireguard tunnels when internet connection is established.

Using latest SNAPSHOT on Raspberry PI 4 (bcm27xx).

egc112, 2025-03-11:
Maybe not related but your List allowed IPs overlap with your other wg interface and maybe with your lan: `list allowed_ips '192.168.0.0/16'`

You use `option nohostroute '1'` for both interface although it looks like your default route is via the WAN so you can do without a host route it does not hurt to have a host route so maybe remove this?

For vpn_wg1 you have `option defaultroute '0'`, I understand you do not want a default route but option ip4table/option ip6table already takes care of removing the default route so you can also remove this.

I see you are using table name `pbr_vpn_wg1` does that mean you are also using the [PBR app](https://openwrt.org/docs/guide-user/network/routing/pbr_app) ?

It could well be that all these things are not related to your problem and that it is indeed a netifd concurrency/bug. 
You can search the log for netifd problems with `logread | grep -E -i 'netifd|wireguard'`

schuettecarsten, 2025-10-06:
I've created a new bug in netifd project, closing this.

closed 2025-10-06 by schuettecarsten