openwrt/openwrt #18159: kmod-usb-net-cdc-ether

Maurizio-DM, 2025-03-03:
### Describe the bug

Upgrading from snapshot version, having 6.6.56-r1 working as a charm, 24.10 stopped working.

### OpenWrt version

24.10

### OpenWrt release

24.10

### OpenWrt target/subtarget

ramips/mt7621

### Device

Zyxel LTE5398-M904

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

mrkiko, 2025-03-03:
May be @Leo-PL can help ?

Leo-PL, 2025-03-03:
If only there were some logs posted.
I recall that interface naming changed recently back again from `usb<N>` to `eth<N`>` for some devices, but with no details and no device at hand, I can't do much.

Maurizio-DM, 2025-03-03:
Hello @Leo-PL, first of all thank you.
My device is placed in another location and I reinstalled old working version.
But in a couple of days I can reinstall 24.10 and extract the infos.
Consider me a super-rookie  :P.  Where can I find the logs to post ?

Leo-PL, 2025-03-03:
Log in over SSH and print the output of `logread` after boot. Would be good to compare working vs non-working case.

robimarko, 2025-03-07:
This should be fixed with:
https://github.com/openwrt/openwrt/commit/ecd609f509f29ed1f75db5c7a623f359c64efb72

closed 2025-03-07 by robimarko