openwrt/openwrt #18160: Openwrt 24.10 resolv.conf behavior change

asdfjkluiop, 2025-03-04:
### Describe the bug

I have noresolv=1 on my dnsmasq config so that /etc/resolv.conf is not set to loopback to dnsmasq, in 24.10 regardless of the noresolv setting /etc/resolv.conf is setup to use loopback for DNS.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

x86/64

### Device

amd64

### Image kind

Official downloaded image

### Steps to reproduce

Go to Network -> DHCP and DNS -> Resolv & hosts files
Then check ignore resolv file.

### Actual behaviour

/etc/resolv.conf is left with the original contents
```
search lan
nameserver 127.0.0.1
nameserver ::1
```

### Expected behaviour

/etc/resolv.conf should not be set to use dnsmasq but rather the per-interface DNS servers as in previous versions of OpenWRT

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2025-03-04:
How have you set the `localuse` option?

asdfjkluiop, 2025-03-05:
@dave14305 is that a new option? Setting it to 0 fixes the behavior but I never had to do that before, thanks for the info though. Is there a LuCI option for it? Just curious, I personally don't mind doing it from the CLI with UCI

closed 2025-03-05 by asdfjkluiop

dave14305, 2025-03-05:
Looks like the logic was changed for 24.10: https://github.com/openwrt/openwrt/commit/8815a3114ffb788ca037e634042811a9fed1562d