openwrt/openwrt #18162: Mi AX3000T güncelleme sonrası TypeError "protocol.pppoe "protocol.ppp" "protocol.dhcpv6" hatası

taroxzen, 2025-03-04:
### Describe the bug

[image]

After the recent updates, error codes started appearing on my device as seen in the image. I reinstalled the SYSUPGRADE file to solve the issue, but the errors in the image started to appear.

MI AX3000T RD23 - ESMT
```
`TypeError: "protocol.pppoe" factory yields invalid constructor at http://192.168.11.1/luci-static/resources/luci.js?v=25.059.57589~8d7a815:174:16 Dismiss 

TypeError: "protocol.ppp" factory yields invalid constructor at http://192.168.11.1/luci-static/resources/luci.js?v=25.059.57589~8d7a815:174:16 Dismiss 

TypeError: "protocol.dhcpv6" factory yields invalid constructor at http://192.168.11.1/luci-static/resources/luci.js?v=25.059.57589~8d7a815:174:16 Dismiss
```

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-03-04:
Are you sure 120MB RAM filled without installing/upgrading over sysupgrade?

taroxzen, 2025-03-04:
It was the same before. After the update last week, error codes appeared.

brada4, 2025-03-04:
You should not update separate packages in general, use auc / owut / imagebuilder / attendedsysupgrade / firmware-selector to make complete software set into image.

taroxzen, 2025-03-05:
The problem was solved by itself.

closed 2025-03-05 by taroxzen