openwrt/openwrt #18169: Problem with OpenWRT on Ubiquiti NanoStation AC Loco after compilatio

JohnyByk, 2025-03-05:
Installed OpenWRT on a Ubiquiti NanoStation AC Loco. I followed this guide: https://openwrt.org/toh/ubiquiti/common, but I compiled my own version with LuCI and Argon.

Procedure:
```
git clone https://github.com/openwrt/openwrt.git
cd openwrt

./scripts/feeds update -a
./scripts/feeds install -a

git clone https://github.com/jerrykuku/luci-theme-argon package/luci-theme-argon
git clone https://github.com/jerrykuku/luci-app-argon-config package/luci-app-argon-config

make download
make
```
The compilation completed without errors. I flashed the firmware onto the device, and at first glance, it seemed to work, but as it turned out, not entirely.

Saving any settings does not work. After a hard reset, all settings revert to default.

```
When I connect to WiFi (Scan > Select AP, Apply & Save), the device becomes unresponsive (GUI, SSH, and ping are still accessible), and the log is flooded with errors like:
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.190177] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 00786500
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.198825] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.224935] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.233526] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 00786500
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.241989] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.249961] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.266586] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.281290] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 00786500
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.289283] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 00786500
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.311251] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.319791] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8

```
I tried compiling versions 23.05.3 and 24.10.0 – both exhibit the same issue.

What could I have done wrong? Is it an incorrect compilation configuration, or something else?

My .config:
```
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
CONFIG_HOST_OS_LINUX=y
CONFIG_DEFAULT_TARGET_ath79_generic=y
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_MULTI_PROFILE=yCONFIG_TARGET_PER_DEVICE_ROOTFS=y
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_ubnt_nanostation-ac-loco=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_ubnt_nanostation-ac-loco=""CONFIG_HAS_SUBTARGETS=y
CONFIG_HAS_DEVICES=y
CONFIG_TARGET_BOARD="ath79"
CONFIG_TARGET_SUBTARGET="generic"
CONFIG_TARGET_ARCH_PACKAGES="mips_24kc"
CONFIG_DEFAULT_TARGET_OPTIMIZATION="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc"
CONFIG_CPU_TYPE="24kc"
CONFIG_LINUX_5_15=y
CONFIG_MODULE_DEFAULT_ath10k-firmware-qca988x-ct=y
CONFIG_DEFAULT_base-files=y
CONFIG_MODULE_DEFAULT_base-files=y
CONFIG_DEFAULT_busybox=y
CONFIG_MODULE_DEFAULT_busybox=y
CONFIG_DEFAULT_ca-bundle=y
CONFIG_MODULE_DEFAULT_ca-bundle=y
CONFIG_DEFAULT_dnsmasq=y
CONFIG_MODULE_DEFAULT_dnsmasq=y
CONFIG_DEFAULT_dropbear=y
CONFIG_MODULE_DEFAULT_dropbear=y
CONFIG_DEFAULT_firewall4=y
CONFIG_MODULE_DEFAULT_firewall4=y
[... 668 lines cut ...]
CONFIG_PACKAGE_luci-proto-ppp=yCONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_nftables-json=yCONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_hostapd-common=y
CONFIG_WPA_MSG_MIN_PRIORITY=3
CONFIG_DRIVER_11AC_SUPPORT=y
CONFIG_PACKAGE_wpad-basic-mbedtls=y
CONFIG_PACKAGE_iw=y
CONFIG_PACKAGE_odhcp6c=y
CONFIG_PACKAGE_odhcp6c_ext_cer_id=0
CONFIG_PACKAGE_odhcpd-ipv6only=yCONFIG_PACKAGE_odhcpd_ipv6only_ext_cer_id=0CONFIG_PACKAGE_ppp=y
CONFIG_PACKAGE_ppp-mod-pppoe=y
CONFIG_PACKAGE_uclient-fetch=yCONFIG_PACKAGE_uboot-envtools=yCONFIG_PACKAGE_iwinfo=y
CONFIG_PACKAGE_jshn=y
CONFIG_PACKAGE_libjson-script=y
CONFIG_STRACE_NONE=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
```
Please help me :)

brada4, 2025-03-05:
Could you retry with standard image and edit your post to contain infos required in issue template?

JohnyByk, 2025-03-21:
I don't know if it's the right approach, but I created a new post with the updated information I gathered in the meantime. I believe this is a bug, not a compilation issue.

brada4, 2025-03-21:
clise with link in comment

brada4, 2025-08-14:
author can do.

closed 2025-08-14 by JohnyByk