openwrt/openwrt #18177: When making v24.10.0 "feeds update -a" didn't report a fatal error (x86/64 generic)

hanepudding, 2025-03-06:
### Describe the bug

The very first time learning to make a image in my whole life, pardon me if I am not giving a good one.
On the branch v24.10.0, I follow the official instruction:
```bash
./scripts/feeds update -a
```

The script didn't tell a fatal error, until I check by myself:
```bash
hanep@debian:~/openwrt/x86-64$ ./scripts/feeds update -a
Updating feed 'packages' from 'https://git.openwrt.org/feed/packages.git^201fd099b80a2931b7326ce20b0cbb824296c99f' ...
Cloning into './feeds/packages'...
remote: Enumerating objects: 215663, done.
remote: Counting objects: 100% (215663/215663), done.
remote: Compressing objects: 100% (88439/88439), done.
remote: Total 215663 (delta 126640), reused 201601 (delta 116003), pack-reused 0
Receiving objects: 100% (215663/215663), 55.60 MiB | 8.48 MiB/s, done.
Resolving deltas: 100% (126640/126640), done.
Switched to a new branch '201fd099b80a2931b7326ce20b0cbb824296c99f'
/home/hanep/openwrt/x86-64
Updating feed 'luci' from 'https://git.openwrt.org/project/luci.git^7b0663a5557118499dc3b3d44550efc1b6fa3feb' ...
Cloning into './feeds/luci'...
remote: Enumerating objects: 6811, done.
remote: Counting objects: 100% (6811/6811), done.
remote: Compressing objects: 100% (3017/3017), done.
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fetch-pack: unexpected disconnect while reading sideband packet
# fatal here
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
failed.
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git^e87b55c6a642947ad7e24cd5054a637df63d5dbe' ...
Cloning into './feeds/routing'...
remote: Enumerating objects: 11621, done.
remote: Counting objects: 100% (11621/11621), done.
remote: Compressing objects: 100% (7119/7119), done.
remote: Total 11621 (delta 5346), reused 9085 (delta 3826), pack-reused 0
Receiving objects: 100% (11621/11621), 2.67 MiB | 2.34 MiB/s, done.
Resolving deltas: 100% (5346/5346), done.
Switched to a new branch 'e87b55c6a642947ad7e24cd5054a637df63d5dbe'
/home/hanep/openwrt/x86-64
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git^fd605af7143165a2490681ec1752f259873b9147' ...
Cloning into './feeds/telephony'...
remote: Enumerating objects: 12025, done.
remote: Counting objects: 100% (12025/12025), done.
remote: Compressing objects: 100% (7389/7389), done.
remote: Total 12025 (delta 6757), reused 8137 (delta 4394), pack-reused 0
Receiving objects: 100% (12025/12025), 2.44 MiB | 2.31 MiB/s, done.
Resolving deltas: 100% (6757/6757), done.
Switched to a new branch 'fd605af7143165a2490681ec1752f259873b9147'
/home/hanep/openwrt/x86-64
Create index file './feeds/packages.index' 
Checking 'true'... ok.
Checking 'false'... ok.
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... updated.
Checking 'working-gcc'... ok.
Checking 'g++'... updated.
Checking 'working-g++'... ok.
Checking 'ncurses.h'... ok.
Checking 'git'... updated.
Checking 'rsync'... updated.
Checking 'perl-data-dumper'... ok.
Checking 'perl-findbin'... ok.
Checking 'perl-file-copy'... ok.
Checking 'perl-file-compare'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'perl-ipc-cmd'... ok.
Checking 'tar'... updated.
Checking 'find'... updated.
Checking 'bash'... updated.
Checking 'xargs'... updated.
Checking 'patch'... updated.
Checking 'diff'... updated.
Checking 'cp'... updated.
Checking 'seq'... updated.
Checking 'awk'... updated.
Checking 'grep'... updated.
Checking 'egrep'... updated.
Checking 'getopt'... updated.
Checking 'realpath'... updated.
Checking 'stat'... updated.
Checking 'gzip'... updated.
Checking 'unzip'... updated.
Checking 'bzip2'... updated.
Checking 'wget'... updated.
Checking 'install'... updated.
Checking 'perl'... updated.
Checking 'python'... updated.
Checking 'python3'... updated.
Checking 'python3-distutils'... ok.
Checking 'python3-stdlib'... ok.
Checking 'file'... updated.
Checking 'which'... updated.
Checking 'argp.h'... ok.
Checking 'fts.h'... ok.
Checking 'obstack.h'... ok.
Checking 'libintl.h'... ok.
Checking 'ldconfig-stub'... ok.
Collecting package info: done
Collecting target info: done
Create index file './feeds/luci.index' 
find: ‘feeds/luci’: No such file or directory
Collecting package info: done
find: ‘feeds/luci’: No such file or directory
Collecting target info: done
Create index file './feeds/routing.index' 
Collecting package info: done
Collecting target info: done
Create index file './feeds/telephony.index' 
Collecting package info: done
Collecting target info: done
```
I spot the error by incident, so I redo that update prompt.

That arise me about another problematic image I have built yesterday, I thought that was bacause of my misoperation then, but now I may have the aliby :) Is it an intended behavior?

### OpenWrt version

24.10.0

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

x86/64 generic

### Device

x86/64 generic

### Image kind

Self-built image

### Steps to reproduce

git checkout v24.10.0
make distclean
./scripts/feeds update -a

### Actual behaviour

Didn't report fatal when fatal.

### Expected behaviour

Prompt error at the end of the script when fatal

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2025-03-06:
> error: RPC failed; curl 18 transfer closed with outstanding read data remaining
> fetch-pack: unexpected disconnect while reading sideband packet

You just need to retry that step, it might be something on the network or on the server, who knows, we just know, that this `sideband packet` issues started appearing sometime ago more frequently

Other option is to try switching to github.com repositories using following command:

```shell
sed -i -E 's;git.openwrt.org/(feed|project);github.com/openwrt;' feeds.conf.default
```

ynezz, 2025-03-06:
There is second option, cherry-pick new feed fetching improvements:

```shell
git cherry-pick 32d0a57dc10 9ec32cfb2733
```

They're going to be part of the next OpenWrt 24.10.1 release.

closed 2025-03-06 by ynezz

hanepudding, 2025-03-07:
Many thanks, ynezz. I figured it out and retried that step, and I succeed!
Just a kind reminder from a junior that, a prompt like "ERROR: Failed to download from source example.com" **at the end** will be good maybe?