openwrt/openwrt #18183: Openwrt 24.10 MPTCP not working

alfindk, 2025-03-07:
### Describe the bug

root@wg-server:~# multipath -c
-ash: multipath: not found

### OpenWrt version

24.10

### OpenWrt release

stable

### OpenWrt target/subtarget

ramips/mt7621

### Device

xiaomi router 3 pro 

### Image kind

Official downloaded image

### Expected behaviour

I need to start I need to start mptcp on the router but this error comes out 

root@wg-server:~# multipath -c
-ash: multipath: not found

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-03-07:
Try `ip mptcp` instead, and complain about random internet page suggesting you use legacy command utility

alfindk, 2025-03-10:
root@wg-server:~# opkg install ip-full
Installing ip-full (6.11.0-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/base/ip-full_6.11.0-r1_x86_64.ipk
Installing libelf1 (0.192-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/base/libelf1_0.192-r1_x86_64.ipk
Installing libbpf1 (1.5.0-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/base/libbpf1_1.5.0-r1_x86_64.ipk
Configuring libelf1.
Configuring libbpf1.
Configuring ip-full.

root@wg-server:~# ip mptcp
Usage:  ip mptcp endpoint add ADDRESS [ dev NAME ] [ id ID ]
                                      [ port NR ] [ FLAG-LIST ]
        ip mptcp endpoint delete id ID [ ADDRESS ]
        ip mptcp endpoint change [ id ID ] [ ADDRESS ] [ port NR ] CHANGE-OPT
        ip mptcp endpoint show [ id ID ]
        ip mptcp endpoint flush
        ip mptcp limits set [ subflows NR ] [ add_addr_accepted NR ]
        ip mptcp limits show
        ip mptcp monitor
FLAG-LIST := [ FLAG-LIST ] FLAG
FLAG  := [ signal | subflow | backup | fullmesh ]
CHANGE-OPT := [ backup | nobackup | fullmesh | nofullmesh ]

CoelacanthusHex, 2025-05-18:
> Try `ip mptcp` instead, and complain about random internet page suggesting you use legacy command utility

1. It's not a random Internet page, it's from the OpenWRT wiki: https://openwrt.org/docs/guide-user/network/mptcp#test
2. It's even not "legacy command utility", it's a command utility for the totally different thing: multipath block device.