openwrt/openwrt #18187: LAN and WDS totally unstable since kernel 6.6.73

grosjo, 2025-03-07:
### Describe the bug

Hi,

Recent git let the device reboot every 2 min.

Getting back to the git commit of 6.6.73 allow recovery of stability both on WDS and LAN ( git commit 761b244964f8d5da9fbd27a618565b4c38ae0cbc )

Befoe 6.6.80, LAN was having lags (up to 1000ms from the router to my laptop)

6.6.80 doe snot let the device survive for more than a few minutes

THank you for your help

### OpenWrt version

r28523-9836836302

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

YunCore/KuWfi CPE830(D)

### Image kind

Self-built image

### Steps to reproduce

Just upgrade to latest git

### Actual behaviour

Device reboot after 2 min

### Expected behaviour

Stable LAN and WDS and device

### Diffconfig

```text
# ./scripts/diffconfig.sh
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_kuwfi_cpe830=y
# CONFIG_ATH9K_HWRNG is not set
CONFIG_ATH9K_SUPPORT_PCOEM=y
# CONFIG_ATH9K_UBNTHSR is not set
# CONFIG_ATH_USER_REGD is not set
CONFIG_CLEAN_IPKG=y
# CONFIG_DOWNLOAD_CHECK_CERTIFICATE is not set
# CONFIG_DROPBEAR_AGENTFORWARD is not set
# CONFIG_DROPBEAR_CHACHA20POLY1305 is not set
# CONFIG_DROPBEAR_CURVE25519 is not set
# CONFIG_DROPBEAR_DBCLIENT_AGENTFORWARD is not set
# CONFIG_DROPBEAR_ED25519 is not set
# CONFIG_DROPBEAR_U2F is not set
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CGROUPS is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_DEBUG_FS is not set
# CONFIG_KERNEL_DEBUG_INFO is not set
CONFIG_KERNEL_DEVKMEM=y
CONFIG_KERNEL_DEVMEM=y
# CONFIG_KERNEL_DEVPTS_MULTIPLE_INSTANCES is not set
CONFIG_KERNEL_DEVTMPFS=y
CONFIG_KERNEL_DEVTMPFS_MOUNT=y
# CONFIG_KERNEL_ELF_CORE is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_IPV6_MROUTE is not set
# CONFIG_KERNEL_IPV6_SEG6_LWTUNNEL is not set
# CONFIG_KERNEL_IP_MROUTE is not set
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
# CONFIG_KERNEL_KALLSYMS is not set
# CONFIG_KERNEL_KEYS is not set
# CONFIG_KERNEL_MAGIC_SYSRQ is not set
# CONFIG_KERNEL_MPTCP is not set
# CONFIG_KERNEL_NAMESPACES is not set
# CONFIG_KERNEL_NET_L3_MASTER_DEV is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_NF_CONNTRACK_TIMEOUT is not set
# CONFIG_KERNEL_POSIX_MQUEUE is not set
CONFIG_KERNEL_PROC_PAGE_MONITOR=y
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_SECCOMP is not set
# CONFIG_KERNEL_SWAP is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
# CONFIG_KERNEL_UBIFS_FS_SECURITY is not set
# CONFIG_KERNEL_WERROR is not set
# CONFIG_KERNEL_XFS_POSIX_ACL is not set
CONFIG_LUCI_SRCDIET=y
# CONFIG_PACKAGE_ATH_DFS is not set
CONFIG_PACKAGE_ATH_DYNACK=y
# CONFIG_PACKAGE_MAC80211_DEBUGFS is not set
# CONFIG_PACKAGE_MAC80211_MESH is not set
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_getrandom is not set
CONFIG_PACKAGE_iptables-mod-condition=y
CONFIG_PACKAGE_iptables-mod-ipv4options=y
CONFIG_PACKAGE_iptables-mod-nfqueue=y
CONFIG_PACKAGE_iptables-mod-physdev=y
CONFIG_PACKAGE_iptables-mod-proto=y
CONFIG_PACKAGE_iptables-nft=y
CONFIG_PACKAGE_kmod-br-netfilter=y
# CONFIG_PACKAGE_kmod-gpio-button-hotplug is not set
CONFIG_PACKAGE_kmod-ipt-condition=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-ipv4options=y
CONFIG_PACKAGE_kmod-ipt-nfqueue=y
CONFIG_PACKAGE_kmod-ipt-physdev=y
CONFIG_PACKAGE_kmod-ipt-proto=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nfnetlink-queue=y
CONFIG_PACKAGE_kmod-nft-compat=y
# CONFIG_PACKAGE_kmod-ppp is not set
# CONFIG_PACKAGE_knot-resolver_dnstap is not set
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext-nft=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
# CONFIG_PACKAGE_libpthread is not set
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-filebrowser=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_openwrt-keyring is not set
# CONFIG_PACKAGE_opkg is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-digest=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-lua=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_urandom-seed is not set
# CONFIG_PACKAGE_urngd is not set
# CONFIG_PACKAGE_usign is not set
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_PACKAGE_zoneinfo-asia=y
CONFIG_PACKAGE_zoneinfo-core=y
CONFIG_PROCD_SHOW_BOOT=y
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_MKLIBS=y
# CONFIG_USE_SECCOMP is not set
# CONFIG_WPA_MBO_SUPPORT is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-03-07:
Does default image brwak in any way? In particular DEVMEM looks gross

grosjo, 2025-03-07:
What is devmem ?

brada4, 2025-03-07:
One of parameters you changed. Ask yourself or google?

grosjo, 2025-03-07:
What the link with the network issue ?

My config is the same in both case (kernel < 6.6.73 or > 6.6.73)

grosjo, 2025-03-08:
> Does default image brwak in any way? In particular DEVMEM looks gross

default image does not provide Luci, really hard to monitor the connection

grosjo, 2025-03-08:
I remove all the weird additions

```
$ ./scripts/diffconfig.sh 
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_kuwfi_cpe830=y
# CONFIG_ATH9K_HWRNG is not set
# CONFIG_ATH9K_UBNTHSR is not set
# CONFIG_ATH_USER_REGD is not set
CONFIG_CLEAN_IPKG=y
# CONFIG_DOWNLOAD_CHECK_CERTIFICATE is not set
# CONFIG_DROPBEAR_AGENTFORWARD is not set
# CONFIG_DROPBEAR_CHACHA20POLY1305 is not set
# CONFIG_DROPBEAR_CURVE25519 is not set
# CONFIG_DROPBEAR_DBCLIENT_AGENTFORWARD is not set
# CONFIG_DROPBEAR_ED25519 is not set
# CONFIG_DROPBEAR_U2F is not set
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CGROUPS is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_DEBUG_FS is not set
# CONFIG_KERNEL_DEBUG_INFO is not set
# CONFIG_KERNEL_DEVPTS_MULTIPLE_INSTANCES is not set
CONFIG_KERNEL_DEVTMPFS=y
CONFIG_KERNEL_DEVTMPFS_MOUNT=y
# CONFIG_KERNEL_ELF_CORE is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_FANOTIFY is not set
# CONFIG_KERNEL_FHANDLE is not set
# CONFIG_KERNEL_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_IPV6_MROUTE is not set
# CONFIG_KERNEL_IPV6_SEG6_LWTUNNEL is not set
# CONFIG_KERNEL_IP_MROUTE is not set
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
# CONFIG_KERNEL_KALLSYMS is not set
# CONFIG_KERNEL_KEYS is not set
# CONFIG_KERNEL_MAGIC_SYSRQ is not set
# CONFIG_KERNEL_MPTCP is not set
# CONFIG_KERNEL_NAMESPACES is not set
# CONFIG_KERNEL_NET_L3_MASTER_DEV is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_NF_CONNTRACK_TIMEOUT is not set
# CONFIG_KERNEL_POSIX_MQUEUE is not set
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_SECCOMP is not set
# CONFIG_KERNEL_SWAP is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
# CONFIG_KERNEL_UBIFS_FS_SECURITY is not set
# CONFIG_KERNEL_WERROR is not set
# CONFIG_KERNEL_XFS_POSIX_ACL is not set
CONFIG_LUCI_SRCDIET=y
# CONFIG_PACKAGE_ATH_DFS is not set
# CONFIG_PACKAGE_MAC80211_DEBUGFS is not set
# CONFIG_PACKAGE_MAC80211_MESH is not set
# CONFIG_PACKAGE_apk-mbedtls is not set
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_getrandom is not set
CONFIG_PACKAGE_iptables-mod-condition=y
CONFIG_PACKAGE_iptables-mod-ipv4options=y
CONFIG_PACKAGE_iptables-mod-nfqueue=y
CONFIG_PACKAGE_iptables-mod-physdev=y
CONFIG_PACKAGE_iptables-mod-proto=y
CONFIG_PACKAGE_iptables-nft=y
CONFIG_PACKAGE_kmod-br-netfilter=y
# CONFIG_PACKAGE_kmod-gpio-button-hotplug is not set
CONFIG_PACKAGE_kmod-ipt-condition=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-ipv4options=y
CONFIG_PACKAGE_kmod-ipt-nfqueue=y
CONFIG_PACKAGE_kmod-ipt-physdev=y
CONFIG_PACKAGE_kmod-ipt-proto=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nfnetlink-queue=y
CONFIG_PACKAGE_kmod-nft-compat=y
# CONFIG_PACKAGE_kmod-ppp is not set
# CONFIG_PACKAGE_knot-resolver_dnstap is not set
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext-nft=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-filebrowser=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_openwrt-keyring is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
# CONFIG_PACKAGE_swconfig is not set
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_urandom-seed is not set
# CONFIG_PACKAGE_urngd is not set
# CONFIG_PACKAGE_usign is not set
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zoneinfo-asia=y
CONFIG_PACKAGE_zoneinfo-core=y
CONFIG_PROCD_SHOW_BOOT=y
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_USE_APK is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_MKLIBS=y
# CONFIG_USE_SECCOMP is not set
# CONFIG_WPA_MBO_SUPPORT is not set
# CONFIG_PACKAGE_libncurses-dev is not set
# CONFIG_PACKAGE_libpthread is not set
# CONFIG_PACKAGE_opkg is not set
CONFIG_PACKAGE_zlib=y
# CONFIG_PACKAGE_zlib-dev is not set
# CONFIG_ZLIB_OPTIMIZE_SPEED is not set

```

grosjo, 2025-03-08:
> Does default image brwak in any way? In particular DEVMEM looks gross

default image also leads to device rebooting every 2 min

brada4, 2025-03-08:
Please run scripts/diffconfig.sh and edit your posts accordingly.

Try to enter failsafe mode
https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset#entering_failsafe_mode
Does it crash? dmesg ?

grosjo, 2025-03-10:
No safe mode here. The device just reboot after about 2 min without any message

Getting back to 6.6.73 solves the issue

grosjo, 2025-03-10:
Same results if factory reset (loading the image from TFTP, therefore erasing all configs)

grosjo, 2025-03-12:
For devices not using the network, the WDS (Wifi) disconnects after 12 to 48h. 
Again, getting back to older version resolves the problem

grosjo, 2025-03-14:
allo ?

grosjo, 2025-03-21:
who cares that recent openwrt is bricked ?

brada4, 2025-03-21:
Mixing xtables and nftables rules in kernel is documented and guaranteed to malfunction. Kind of keep on bricking to your liking, cand brivk failsafe so far.

grosjo, 2025-03-22:
Don't see the link with that issue, but curious to know how to get iptables working otherwise ?

brada4, 2025-03-22:
What issue? From official nftables wiki:
> Beware of using both the nft and the legacy tools at the same time. That means using both x_tables and nf_tables kernel subsystems at the same time, and could lead to unexpected results.

grosjo, 2025-03-22:
the issue here is that both LAN and WDS collapse (LAN after 2 min, WDS after abotu 24h)

grosjo, 2025-03-22:
Removing iptables does not solve that issue (as the same issue appears with the precompiled version anyway)

Below the new config I am using

```
# ./scripts/diffconfig.sh 
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_kuwfi_cpe830=y
# CONFIG_ATH9K_HWRNG is not set
# CONFIG_ATH9K_UBNTHSR is not set
# CONFIG_ATH_USER_REGD is not set
CONFIG_CLEAN_IPKG=y
# CONFIG_DOWNLOAD_CHECK_CERTIFICATE is not set
# CONFIG_DROPBEAR_AGENTFORWARD is not set
# CONFIG_DROPBEAR_CHACHA20POLY1305 is not set
# CONFIG_DROPBEAR_CURVE25519 is not set
# CONFIG_DROPBEAR_DBCLIENT_AGENTFORWARD is not set
# CONFIG_DROPBEAR_ED25519 is not set
# CONFIG_DROPBEAR_U2F is not set
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CGROUPS is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_DEBUG_FS is not set
# CONFIG_KERNEL_DEBUG_INFO is not set
# CONFIG_KERNEL_DEVPTS_MULTIPLE_INSTANCES is not set
CONFIG_KERNEL_DEVTMPFS=y
CONFIG_KERNEL_DEVTMPFS_MOUNT=y
# CONFIG_KERNEL_ELF_CORE is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_FANOTIFY is not set
# CONFIG_KERNEL_FHANDLE is not set
# CONFIG_KERNEL_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_IPV6_MROUTE is not set
# CONFIG_KERNEL_IPV6_SEG6_LWTUNNEL is not set
# CONFIG_KERNEL_IP_MROUTE is not set
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
# CONFIG_KERNEL_KALLSYMS is not set
# CONFIG_KERNEL_KEYS is not set
# CONFIG_KERNEL_MAGIC_SYSRQ is not set
# CONFIG_KERNEL_MPTCP is not set
# CONFIG_KERNEL_NAMESPACES is not set
# CONFIG_KERNEL_NET_L3_MASTER_DEV is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_NF_CONNTRACK_TIMEOUT is not set
# CONFIG_KERNEL_POSIX_MQUEUE is not set
CONFIG_KERNEL_PROC_STRIPPED=y
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_SECCOMP is not set
# CONFIG_KERNEL_SWAP is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
# CONFIG_KERNEL_UBIFS_FS_SECURITY is not set
# CONFIG_KERNEL_WERROR is not set
# CONFIG_KERNEL_XFS_POSIX_ACL is not set
CONFIG_LUCI_SRCDIET=y
# CONFIG_PACKAGE_ATH_DFS is not set
# CONFIG_PACKAGE_MAC80211_DEBUGFS is not set
# CONFIG_PACKAGE_MAC80211_MESH is not set
# CONFIG_PACKAGE_ca-bundle is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_getrandom is not set
# CONFIG_PACKAGE_kmod-gpio-button-hotplug is not set
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-nft-bridge=y
# CONFIG_PACKAGE_kmod-ppp is not set
# CONFIG_PACKAGE_knot-resolver_dnstap is not set
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext-nft=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-filebrowser=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
# CONFIG_PACKAGE_openwrt-keyring is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_urandom-seed is not set
# CONFIG_PACKAGE_urngd is not set
# CONFIG_PACKAGE_usign is not set
CONFIG_PACKAGE_zoneinfo-asia=y
CONFIG_PACKAGE_zoneinfo-core=y
CONFIG_PROCD_SHOW_BOOT=y
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_USE_APK is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_MKLIBS=y
# CONFIG_USE_SECCOMP is not set
# CONFIG_WPA_MBO_SUPPORT is not set
CONFIG_PACKAGE_apk-mbedtls=y
# CONFIG_PACKAGE_libncurses-dev is not set
# CONFIG_PACKAGE_libpthread is not set
# CONFIG_PACKAGE_opkg is not set
CONFIG_PACKAGE_zlib=y
# CONFIG_PACKAGE_zlib-dev is not set
# CONFIG_ZLIB_OPTIMIZE_SPEED is not set

```

brada4, 2025-03-22:
You can use standard install with 16mb flash and change few packages you need. I see apk forced, while opkg is default in 24.10. Id suggest starting fresh with binry images.

grosjo, 2025-03-27:
pre-build image do the same , as already stated before

grosjo, 2025-05-15:
When this can be addressed? 

Broken wifi seems quite a serious issue

brada4, 2025-05-15:
If you could bisect between .73 and you reporting the crash.