openwrt/openwrt #18190: 5 GHz PHY LED on RT-AC1200 v2 not working

LegoLivesMatter, 2025-03-08:
### Describe the bug

The 5 GHz LED on the router does not light up at all. The 2.4 GHz LED works normally.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt76x8

### Device

ASUS RT-AC1200 V2

### Image kind

Official downloaded image

### Steps to reproduce

Create an AP on the 5 GHz PHY and enable it.

### Actual behaviour

The 5 GHz LED does not light up at all.

### Expected behaviour

The 5 GHz LED turns on and flashes on activity.

### Additional info

There was a PR to fix this, but it wasn't merged: https://github.com/openwrt/mt76/pull/651

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

vok4us, 2025-07-07:
Confirm. I have OpenWRT 24.10.2 Ramips mt76x8 for Asus RT-AC1200 v2 - the 5 GHz LED is not working. OpenWRT developers add a patch https://github.com/openwrt/mt76/pull/651 in the next update

vok4us, 2026-09-17:
Hello. I made changes to the init.c, pci_init.c, and mac80211.c files, as well as to the DTS file of the mt76 driver — the LED turned on, but it was blinking too frequently. What I managed to find out: the mt76 driver does not control the 5 GHz LED. This LED is controlled by the MT7663 chip firmware — the mt7663pr2h_rebb.bin file. To make the LED work properly, you need to find MCU commands specific to MT7663.

LegoLivesMatter, 2026-09-21:
> This LED is controlled by the MT7663 chip firmware — the mt7663pr2h_rebb.bin file. To make the LED work properly, you need to find MCU commands specific to MT7663.

I'm curious if LLMs could be used to help figure this out? There definitely exists a precedent; someone used one to [fix a capture card firmware bug](https://www.downtowndougbrown.com/2026/09/fixing-an-nzxt-signal-4k30-part-2-the-green-pink-video-bug/).