openwrt/openwrt #18198: add support ws1610 mt7981b

nerdjb, 2025-03-08:
how to get nvmem-layout  bootlog 
 i want create dts 
`/dts-v1/;

/ {
	compatible = "M01K45";
	interrupt-parent = <0x01>;
	#address-cells = <0x02>;
	#size-cells = <0x02>;
	model = "M01K45";

	cpus {
		#address-cells = <0x01>;
		#size-cells = <0x00>;

		cpu@0 {
			device_type = "cpu";
			compatible = "arm,cortex-a53";
			enable-method = "psci";
			reg = <0x00>;
		};

		cpu@1 {
			device_type = "cpu";
			compatible = "arm,cortex-a53";
			enable-method = "psci";
			reg = <0x01>;
		};
	};

	pwm@10048000 {
		compatible = "mediatek,mt7981-pwm";
		reg = <0x00 0x10048000 0x00 0x1000>;
		#pwm-cells = <0x02>;
		clocks = <0x02 0x0d 0x02 0x0c 0x02 0x0e 0x02 0x0f 0x02 0x10>;
		clock-names = "top\0main\0pwm1\0pwm2\0pwm3";
	};

	thermal-zones {

		cpu-thermal {
			polling-delay-passive = <0x3e8>;
			polling-delay = <0x3e8>;
			thermal-sensors = <0x03 0x00>;
		};
	};

	thermal@1100c800 {
		#thermal-sensor-cells = <0x01>;
		compatible = "mediatek,mt7981-thermal";
		reg = <0x00 0x1100c800 0x00 0x800>;
		interrupts = <0x00 0x8a 0x04>;
		clocks = <0x02 0x1c 0x02 0x2f 0x02 0x30>;
		clock-names = "therm\0auxadc\0adc_32k";
		mediatek,auxadc = <0x04>;
		mediatek,apmixedsys = <0x05>;
		nvmem-cells = <0x06>;
		nvmem-cell-names = "calibration-data";
		phandle = <0x03>;
	};

	adc@1100d000 {
		compatible = "mediatek,mt7981-auxadc\0mediatek,mt7622-auxadc";
		reg = <0x00 0x1100d000 0x00 0x1000>;
		clocks = <0x02 0x2f 0x02 0x30>;
		clock-names = "main\032k";
		#io-channel-cells = <0x01>;
		phandle = <0x04>;
	};

[... 748 lines cut ...]

			partition@620000 {
				label = "ubi";
				reg = <0x620000 0x7000000>;
			};

			partition@7620000 {
				label = "ubi2";
				reg = <0x7620000 0x7000000>;
			};

			partition@E620000 {
				label = "wtinfo";
				reg = <0xe620000 0x60000>;
			};

			partition@E680000 {
				label = "nvram";
				reg = <0xe680000 0x60000>;
			};
		};
	};
};`

this is dtb from orignial fw i want create dts for device

nerdjb, 2025-03-08:
device specs ws1610 
WS1610
mb MT7981B
MT7976CN WIFI6 chip
mt7531ae
RealTek RTL8221B-VB-CG at MDIO address 6.

```
gpiochip0: GPIOs 455-511, parent: platform/11d00000.pinctrl, pinctrl_moore:
 gpio-455 (                    |wps                 ) in  lo ACTIVE LOW
 gpio-456 (                    |reset               ) in  hi ACTIVE LOW
 gpio-459 (                    |5greset             ) out lo ACTIVE LOW
 gpio-460 (                    |5gpower             ) out hi ACTIVE LOW
 gpio-461 (                    |lede:blue:4g        ) out lo ACTIVE LOW
 gpio-462 (                    |lede:blue:5g        ) out hi ACTIVE LOW
 gpio-463 (                    |lede:blue:lan       ) out hi ACTIVE LOW
 gpio-464 (                    |lede:blue:sim2      ) out lo ACTIVE LOW
 gpio-465 (                    |lede:blue:wifi      ) out lo ACTIVE LOW
 gpio-466 (                    |lede:blue:sim1      ) out hi ACTIVE LOW
 gpio-485 (                    |ws222-scl           ) out lo
 gpio-486 (                    |ws222-sda           ) out hi

```

nerdjb, 2025-03-08:
256mb rom 
1gb ram ddr4

nerdjb, 2025-03-08:
https://github.com/nerdjb/ws1610 
this all files from stock rom

nerdjb, 2025-03-08:
bootlog stock fw 
`7981@C23L3,perform_channel_change() 2848: (Channel=48)
[   43.914923] 7981@C23L3,MakeChSwitchAnnounceIEandExtend() 667: Channel=48, bw=3
[   43.914928] 7981@C23L3,DfsBuildChannelList() 4114: Done
[   43.927381] 7981@C23L3,zero_wait_dfs_switch_ch() 1215: outband ch 0, ch_stat 0
[   43.934598] 7981@C23L3,zero_wait_dfs_switch_ch() 1222: bDedicatedZeroWaitDefault == 0
[   43.942430] 7981@C08L1,UpdateBeaconHandler() 2057: wdev(2) CSA counting, ignore!! (caller:AutoChSelScanNextChannel+0x378/0xce0)!!
[   43.954072] 7981@C08L1,UpdateBeaconHandler() 2057: wdev(3) CSA counting, ignore!! (caller:AutoChSelScanNextChannel+0x378/0xce0)!!
[   44.279923] 7981@C08L1,UpdateBeaconHandler() 2057: wdev(2) CSA counting, ignore!! (caller:update_ap_qload_to_bcn+0xa0/0xc8)!!
[   44.291232] 7981@C08L1,UpdateBeaconHandler() 2057: wdev(3) CSA counting, ignore!! (caller:update_ap_qload_to_bcn+0xa0/0xc8)!!
[   44.956699] 7981@C14L3,ExtEventCswNotifyHandler() 2367: wdev:rax1, wdev->csa_count:0
[   44.964450] 7981@C14L3,ExtEventCswNotifyHandler() 2367: wdev:rax0, wdev->csa_count:0
[   44.972192] 7981@C23L3,ChannelSwitchingCountDownProc() 500:  CSAEventTimer normal end!!
[   44.980218] 7981@C23L3,Dot11HCntDownTimeoutAction() 607: [RDM] Change to RD_SILENCE_MODE
[   44.989173] 7981@C23L3,Dot11HCntDownTimeoutAction() 621: wdev name(rax0), quick 1 ,ch 48
[   44.997259] 7981@C23L3,Dot11HCntDownTimeoutAction() 624: cacreq 0
[   45.003363] 7981@C13L3,Dot11HCntDownTimeoutAction() 643: CSA done and Disable Beacon: rax0
[   45.011630] 7981@C13L3,Dot11HCntDownTimeoutAction() 643: CSA done and Disable Beacon: rax1
[   45.011841] 7981@C03L2,ExtEventBeaconLostHandler() 170: FW EVENT (02:86:88:20:00:00), Reason 0x10
[   45.019905] 7981@C23L3,ap_phy_rrm_init_byRf() 1900: (caller:Dot11HCntDownTimeoutAction+0x204/0x800), wdev_idx:2.
[   45.028742] 7981@C03L1,ExtEventBeaconLostHandler() 176:   AP Beacon OFF!!!
[   45.038909] 7981@C08L3,MacTableResetNonMapWdev() 2097: (caller:ap_phy_rrm_init_byRf+0x3b4/0x608), wdev_idx:2.
[   45.045961] 7981@C03L2,ExtEventBeaconLostHandler() 170: FW EVENT (02:86:88:30:00:00), Reason 0x10
[   45.055743] 7981@C08L3,MacTableResetNonMapWdev() 2097: (caller:ap_phy_rrm_init_byRf+0x3b4/0x608), wdev_idx:3.
[   45.064517] 7981@C03L1,ExtEventBeaconLostHandler() 176:   AP Beacon OFF!!!
[   45.091357] 7981@C13L3,ap_phy_rrm_init_byRf() 2017: Wlan_operate_init and switch channel for : rax0
[   45.100414] 7981@C23L3,DfsGetSysParameters() 1682: Get before pDfsParam->PrimCh(0) band(0) band_bw(0), outbw(0)
[   45.110496] 7981@C23L3,DfsGetSysParameters() 1728: Get after pDfsParam->PrimCh(48) band(1) band_bw(3), outbw(0)
[   45.120578] 7981@C23L3,RadarStateCheck() 291:  RD_SILENCE_MODE  
[   45.127451] 7981@C23L3,operate_loader_phy() 360: oper_cfg: prim_ch(48), ht_bw(1), extcha(1), vht_bw(2), cen_ch_2(0), PhyMode=177!
[   45.139094] 7981@C23L3,operate_loader_phy() 374: oper_dev after adjust: bw(3), prim_ch(48), cen_ch_1(50), cen_ch_2(0),ext_cha(3)!
[   45.150735] 7981@C23L3,operate_loader_phy() 393: oper_dev after decision: bw(3), prim_ch(48), cen_ch_1(50), cen_ch_2(0)!
[   45.161596] 7981@C23L3,DfsGetSysParameters() 1682: Get before pDfsParam->PrimCh(48) band(1) band_bw(3), outbw(0)
[   45.171761] 7981@C23L3,DfsGetSysParameters() 1728: Get after pDfsParam->PrimCh(48) band(1) band_bw(3), outbw(0)
[   45.181840] 7981@C23L3,DfsCacNormalStart() 3980: [RDM] CAC 65 seconds start . Disable MAC TX
[   45.190285] 7981@C08L1,AsicSwitchChannel() 334: AsicSwitchChannel::bandIdx:1 ch_band:1 htbw:1 vht_bw:2 bw:3 extcha:3
[   45.200820] 7981@C08L1,AsicSwitchChannel() 339: AsicSwitchChannel::primch:48 cench1:50 cench2:0 rxstream:2 ap_bw:0 apCenCh:0 Scan:0
[   45.212636] 7981@C23L3,DfsSwitchCheck() 4270: DFS ByPass TX calibration.
[   45.219380] 7981@C00L1,mt7981_apply_dpd_flatness_data() 624: eeprom 0x19a bit 0 is 0, do runtime cal
[   45.228517] 7981@C03L3,MtCmdChannelSwitch() 2650: ctrl_chl=48, ctrl_ch2=0, cent_ch=50 DBDCIdx=1, ChBand=1, BW=3, TXStream=3, RXStream=3, scan(0)
[   45.290065] 7981@C03L3,MtCmdSetTxRxPath() 2910: ctrl_chl=48, ctrl_ch2=0, cent_ch=50, RxPath=7, BandIdx=1, ChBand=1, BW=3,TXStream=3, RXStream=7, scan(0)
[   45.308790] 7981@C12L1,EDCCAInit() 21238: EDCCA compensation: uni_compensation=0, bw_compensation=0, final compensation=0
[   45.319762] 7981@C23L3,DfsGetSysParameters() 1682: Get before pDfsParam->PrimCh(48) band(1) band_bw(3), outbw(0)
[   45.329938] 7981@C23L3,DfsGetSysParameters() 1728: Get after pDfsParam->PrimCh(48) band(1) band_bw(3), outbw(0)
[   45.340033] WiFi@C00L2,dump_ht_cap() 29: Caller: wlan_operate_init+0xa4/0x148
[   45.348130] 7981@C23L3,ap_phy_rrm_init_byRf() 2031: Updated RaBf info bw by wdev ~~~
[   45.355922] 7981@C08L3,UpdateBeaconHandler() 2087: BCN_UPDATE_ENABLE_TX, OmacIdx = 11 (rax0)
[   45.364358] 7981@C08L3,UpdateBeaconHandler() 2092:   Band1 BcnInitedRnd = 209
[   45.371408] 7981@C13L3,ap_phy_rrm_init_byRf() 2017: Wlan_operate_init and switch channel for : rax1
[   45.380466] 7981@C23L3,DfsGetSysParameters() 1682: Get before pDfsParam->PrimCh(48) band(1) band_bw(3), outbw(0)
[   45.390635] 7981@C23L3,DfsGetSysParameters() 1728: Get after pDfsParam->PrimCh(48) band(1) band_bw(3), outbw(0)
[   45.400719] 7981@C23L3,RadarStateCheck() 291:  RD_SILENCE_MODE  
[   45.407592] 7981@C23L3,operate_loader_phy() 360: oper_cfg: prim_ch(48), ht_bw(1), extcha(1), vht_bw(2), cen_ch_2(0), PhyMode=177!
[... cut ...]
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 21.02-SNAPSHOT, 4.3.2
 -----------------------------------------------------
root@OpenWrt:/# [   48.101831] 7981@C12L1,rt28xx_ap_ioctl() 833: interface is down, cmd [8be1] return!!!
[   48.111518] 7981@C12L1,rt28xx_ap_ioctl() 833: interface is down, cmd [8be1] return!!!

`

brada4, 2025-03-09:
Best to try forum. Normalizing oem dts is quite a feat - identufy and inlude soc-shared parts to start with

nerdjb, 2025-03-09:
no one help me there can you help me

nerdjb, 2025-03-09:
im able to make device running and work 
this bootlog 

```

  *** U-Boot Boot Menu ***

      1. Startup system (Default)
      2. Upgrade firmware
      3. Upgrade ATF BL2
      4. Upgrade ATF FIP
      5. Upgrade single image
      6. Load image
      7. Start Web failsafe
      0. U-Boot console

  Press UP/DOWN to move, ENTER to select, ESC to quit

ubi0: attaching mtd6
UBI: EOF marker found, PEBs from 77 will be erased
ubi0: scanning is finished
ubi0: volume 2 ("rootfs_data") re-sized from 9 to 1869 LEBs
ubi0: attached mtd6 (name "ubi", size 248 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 1988, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 13
ubi0: available PEBs: 0, total reserved PEBs: 1988, PEBs reserved for bad PEB h0
No size specified -> Using max size (4444160)
Read 4444160 bytes from volume kernel to 0000000046000000
## Loading kernel from FIT Image at 46000000 ...
   Using 'config-1' configuration
   Trying 'kernel-1' kernel subimage
     Description:  ARM64 OpenWrt Linux-6.6.79
     Type:         Kernel Image
     Compression:  lzma compressed
     Data Start:   0x460000e8
     Data Size:    4298345 Bytes = 4.1 MiB
     Architecture: AArch64
     OS:           Linux                                                        
     Load Address: 0x48000000                                                   
     Entry Point:  0x48000000                                                   
     Hash algo:    crc32                                                        
     Hash value:   f2b3ab4c                                                     
     Hash algo:    sha1                                                         
     Hash value:   9ea23bc85a654c286e9a6cf290c1282bf87f58b2                     
   Verifying Hash Integrity ... crc32+ sha1+ OK                                 
## Loading fdt from FIT Image at 46000000 ...                                   
   Using 'config-1' configuration                                               
   Trying 'fdt-1' fdt subimage                                                  
     Description:  ARM64 OpenWrt huasifei_ws1610 device tree blob               
     Type:         Flat Device Tree                                             
     Compression:  uncompressed                                                 
     Data Start:   0x46419894                                                   
     Data Size:    23025 Bytes = 22.5 KiB                                       
     Architecture: AArch64                                                      
     Hash algo:    crc32                                                        
     Hash value:   35db3517                                                     
     Hash algo:    sha1                                                         
     Hash value:   8387c1b9c06a49cb0c71de45b588650eb798e460                     
   Verifying Hash Integrity ... crc32+ sha1+ OK                                 
   Booting using the fdt blob at 0x46419894                                     
Working FDT set to 46419894                                                     
   Uncompressing Kernel Image                                                   
   Loading Device Tree to 000000007f7f2000, end 000000007f7fa9f0 ... OK         
Working FDT set to 7f7f2000                                                     
                                                                                
Starting kernel ...                                                             
[... 266 lines cut ...]
 -----------------------------------------------------                          
=== WARNING! =====================================                              
There is no root password defined on this device!                               
Use the "passwd" command to set up a new password                               
in order to prevent unauthorized SSH logins.                                    
--------------------------------------------------                              
                                                                                
 OpenWrt recently switched to the "apk" package manager!                        
                                                                                
 OPKG Command           APK Equivalent      Description                         
 ------------------------------------------------------------------             
 opkg install <pkg>     apk add <pkg>       Install a package                   
 opkg remove <pkg>      apk del <pkg>       Remove a package                    
 opkg upgrade           apk upgrade         Upgrade all packages                
 opkg files <pkg>       apk info -L <pkg>   List package contents               
 opkg list-installed    apk info            List installed packages             
 opkg update            apk update          Update package lists                
 opkg search <pkg>      apk search <pkg>    Search for packages                 
 ------------------------------------------------------------------             
                                                                                
For more https://openwrt.org/docs/guide-user/additional-software/opkg-to-apk-cht
                                                                                
root@OpenWrt:~# 

```

brada4, 2025-03-10:
Id gladly say angry word if you link that thread.

Few defects in your dmesg due to devices not identical:
- MAC address not read from oem partitions
might be corrupt irrecoverably due to cros-flashing, alongside wifi calibration params
- usb regulator?
- do leds work?
Bug tracker is not chatserver, no bug in code in this repo shown.

nerdjb, 2025-03-10:
yes working leds mac addres not working beacue  i erased factory mtd and wifi

nerdjb, 2025-03-10:
there no usb port i forget to remove it from dts