openwrt/openwrt #18207: LAN Port Link Speed Regression after upgrade

hamc17, 2025-03-09:
### Describe the bug

After upgrading from 23.05.5 to 24.10.10, LAN ports are stuck on 100MBps maximum.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

 24.10.0

### OpenWrt target/subtarget

ipq806x/generic

### Device

Netgear Nighthawk X4S R7800

### Image kind

Official downloaded image

### Steps to reproduce

Upgrade to 24.10.10 and check the max speed via ethtool.

### Actual behaviour

Install and run ethtool to verify the speed, example output below:

```
root@OpenWrt:~# ethtool lan4
Settings for lan4:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 100Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: Unknown
        Supports Wake-on: d
        Wake-on: d
        Link detected: yes
```

### Expected behaviour

Client has a maximum speed of 2.5GBps, whereas the router itself should have a link speed of 1000MBps.

Speed tests show a maximum of 98-99MBps.

After reverting to 23.05.5, speeds return to the max available, approx. 450MBps for my setup with my ISP.

Output from ethtool after downgrading:
```
root@OpenWrt:~# ethtool eth0.2
Settings for eth0.2:
        Supported ports: [ MII ]
        Supported link modes:   1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: No
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Link detected: yes
```

### Additional info

Not a cable issue as speeds return after downgrading.
IPv6 disabled, EEE disabled on both the client and router, all power saving methods disabled on the client machine, Windows 11, both devices rebooted

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

LGA1150, 2025-03-10:
```
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
```
lan4's link partner is not advertising 1000Mbps speed. Is it due to speed downgrade? Please attach the kernel log.

You may also try forcing lan4 to run at 1000Mbps:
`ethtool -s lan4 speed 1000 duplex full autoneg on`

aimacintyre, 2025-03-10:
> ```
> root@OpenWrt:~# ethtool lan4
> Settings for lan4:
> ```

> ```
> root@OpenWrt:~# ethtool eth0.2
> Settings for eth0.2:
> ```

You haven't reported the lan4 ethtool output for 23.05.5 as you may have intended.

hamc17, 2025-03-10:
> ```
>         Link partner advertised link modes:  10baseT/Half 10baseT/Full
>                                              100baseT/Half 100baseT/Full
>         Link partner advertised pause frame use: [...]

The speed seems to be stuck at a max of 100Mbps on 24.10.10, after reverting to 23.05.5 it returns to 1000Mbps.

> You may also try forcing lan4 to run at 1000Mbps: `ethtool -s lan4 speed 1000 duplex full autoneg on`|

Just tested this, unfortunately this stops the client from connecting on lan4 entirely.
Kernel logs after running this command:
```
[  439.457573] br-lan: port 4(lan4) entered disabled state
[  439.458197] qca8k 37000000.mdio-mii:10 lan4: Link is Down
[  555.133752] qca8k 37000000.mdio-mii:10 lan3: Link is Up - 100Mbps/Full - flow control rx/tx
[  555.133826] br-lan: port 3(lan3) entered blocking state
[  555.140995] br-lan: port 3(lan3) entered forwarding state
```

Complete kernel log:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.6.73 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 13.3.0 r28427-6df0e3d02a) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Mon Feb  3 23:09:37 2025
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Netgear Nighthawk X4S R7800
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] OF: reserved mem: 0x40000000..0x40ffffff (16384 KiB) nomap non-reusable nss@40000000
[    0.000000] OF: reserved mem: 0x41000000..0x411fffff (2048 KiB) nomap non-reusable smem@41000000
[    0.000000] OF: reserved mem: 0x41200000..0x414fffff (3072 KiB) nomap non-reusable rsvd@41200000
[    0.000000] OF: reserved mem: 0x42100000..0x4213ffff (256 KiB) map non-reusable ramoops@42100000
[    0.000000] OF: reserved mem: 0x5fe00000..0x5fffffff (2048 KiB) map reusable rsvd@5fe00000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000] percpu: Embedded 13 pages/cpu s21716 r8192 d23340 u53248
[    0.000000] pcpu-alloc: s21716 r8192 d23340 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Kernel command line: 
[    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121800
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 473084K/491520K available (7726K kernel code, 648K rwdata, 1208K rodata, 1024K init, 250K bss, 18436K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000000] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000020] Switching to timer-based delay loop, resolution 160ns
[    0.000326] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000352] CPU: Testing write buffer coherency: ok
[    0.000407] pid_max: default: 32768 minimum: 301
[    0.008690] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.008712] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.016521] qcom_scm: convention: smc legacy
[    0.018051] RCU Tasks Trace: Setting shift to 1 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=2.
[    0.018281] Setting up static identity map for 0x42300000 - 0x42300060
[    0.018501] rcu: Hierarchical SRCU implementation.
[    0.018511] rcu: 	Max phase no-delay instances is 1000.
[    0.019151] smp: Bringing up secondary CPUs ...
[... 428 lines cut ...]
[   46.657558] br-lan: port 4(lan4) entered blocking state
[   46.664733] br-lan: port 4(lan4) entered forwarding state
[  439.457573] br-lan: port 4(lan4) entered disabled state
[  439.458197] qca8k 37000000.mdio-mii:10 lan4: Link is Down
[  555.133752] qca8k 37000000.mdio-mii:10 lan3: Link is Up - 100Mbps/Full - flow control rx/tx
[  555.133826] br-lan: port 3(lan3) entered blocking state
[  555.140995] br-lan: port 3(lan3) entered forwarding state
```

> > ```
> > root@OpenWrt:~# ethtool lan4
> > Settings for lan4:
> > ```
> 
> > ```
> > root@OpenWrt:~# ethtool eth0.2
> > Settings for eth0.2:
> > ```
> 
> You haven't reported the lan4 ethtool output for 23.05.5 as you may have intended.

This is intentional, the ethernet ports are named differently on 23.05.5 on my side (eth0, eth0.2, eth1, eth1.2 vs lan1-4 on 24.10.10).

Not entirely sure what has changed, as both the client and router LAN ports advertise 1000Mbps on 23.05.5 whereas on 24.10.10 the router only sees the client as advertising 10/100Mbps.
As visible in the logs, the WAN port is unaffected.

LGA1150, 2025-03-10:
On 23.05 you cannot get switch port status via `ethtool`, you should use `swconfig` instead.

LGA1150, 2025-03-10:
What's the autonegotiation configuration on the client side?

hamc17, 2025-03-10:
> On 23.05 you cannot get switch port status via `ethtool`, you should use `swconfig` instead.

Here's the output:
```
root@OpenWrt:~# swconfig dev switch0 show
Global attributes:
        enable_vlan: 1
        ar8xxx_mib_poll_interval: 0
        ar8xxx_mib_type: 0
        enable_mirror_rx: 0
        enable_mirror_tx: 0
        mirror_monitor_port: 0
        mirror_source_port: 0
        arl_age_time: 300
        arl_table: address resolution table
Port 0: MAC 78:d2:94:51:83:98
Port 2: MAC 10:ff:e0:84:cb:d3
Port 5: MAC 00:01:5c:69:d6:46
Port 6: MAC 64:6d:2f:ab:f6:13
Port 6: MAC 78:d2:94:51:83:97

        igmp_snooping: 0
        igmp_v3: 0
Port 0:
        mib: ???
        enable_eee: ???
        igmp_snooping: 0
        vlan_prio: 0
        pvid: 0
        link: port:0 link:up speed:1000baseT full-duplex
Port 1:
        mib: ???
        enable_eee: 0
        igmp_snooping: 0
        vlan_prio: 0
        pvid: 1
        link: port:1 link:down
Port 2:
        mib: ???
        enable_eee: 0
        igmp_snooping: 0
        vlan_prio: 0
        pvid: 1
        link: port:2 link:up speed:1000baseT full-duplex txflow rxflow auto
Port 3:
        mib: ???
        enable_eee: 0
        igmp_snooping: 0
        vlan_prio: 0
        pvid: 1
        link: port:3 link:down
Port 4:
        mib: ???
        enable_eee: 0
        igmp_snooping: 0
        vlan_prio: 0
        pvid: 1
        link: port:4 link:up speed:100baseT full-duplex auto
Port 5:
        mib: ???
        enable_eee: 0
        igmp_snooping: 0
        vlan_prio: 0
        pvid: 2
        link: port:5 link:up speed:1000baseT full-duplex txflow rxflow auto
Port 6:
        mib: ???
        enable_eee: ???
        igmp_snooping: 0
        vlan_prio: 0
        pvid: 0
        link: port:6 link:up speed:1000baseT full-duplex
VLAN 1:
        vid: 1
        ports: 1 2 3 4 6t
VLAN 2:
        vid: 2
        ports: 0t 5
```
Swapped LAN ports due to the issue running `ethtool -s lan4 speed 1000 duplex full autoneg on` on 24.10.10, the client should be on port 2 now based on the MAC address.

> What's the autonegotiation configuration on the client side?

Speed & Duplex set to Auto Negotiation in Windows 11 for the Ethernet device.

I've reverted to 23.05.5 for now as I need to use the router during the day, can upgrade again and do some more testing later.

hamc17, 2025-06-28:
I can confirm this is still an issue.
On the latest release, 24.10.2, a LAN port incorrectly negotiates to 100M whereas on 23.05.5, it correctly negotiates to 1000M.
There is no change on the end device, only an upgrade to 24.10.02 on the R7800 modem itself.

The only real difference I can see that could be a factor is the upgrade from swconfig to DSA, but I am not doing anything fancy with my network config, just leaving it as default.

Specifically LAN port 4 in the images shared.
It is capable of 2.5Gbps while the router itself is capable of 1000Mbps and correctly negotiates to that value in 23.05.5.

On 23.05.5:
```
root@OpenWrt:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdbe:a02f:99f1::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1.1'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option device 'eth0.2'
        option proto 'dhcp'

config interface 'wan6'
        option device 'eth0.2'
        option proto 'dhcpv6'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '1 2 3 4 6t'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '5 0t'
```

[image]

On 24.10.2:

```
root@OpenWrt:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd52:a0a9:5126::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option device 'wan'
        option proto 'dhcp'

config interface 'wan6'
        option device 'wan'
        option proto 'dhcpv6'
```

[image]

Plugging and unplugging the cables, and rebooting the devices doesn't change anything.
Kind of at my wit's end with this, there's nothing I can see in any setting which could be causing the drop in speed negotiation.

hamc17, 2025-11-03:
Same issue on 24.10.04.

[image]

hamc17, 2026-03-14:
Closing, may have been an issue with the Ethernet port and something changing under the hood that made versions later than 23.05.x more sensitive during autonegotiation.
Switching to another device gave the full 1Gbps with the same cable, router port and firmware version.

closed 2026-03-14 by hamc17