openwrt/openwrt #18214: A hardware acceleration issue potentially related to UDP multicast communication has been identified in IPTV multicast communication

ewangsoft, 2025-03-10:
### Describe the bug

This issue was observed on the GL.iNet GL-MT6000 (device details:  https://github.com/openwrt/openwrt/commit/fe10f9743935d6986e80e7cb082469e6bc5a03f0 ).

Under the OpenWrt 23.05 system, multicast proxy forwarding of China Unicom IPTV multicast video using omcproxy or igmpproxy worked normally. However, after upgrading to OpenWrt 24.10.0 and Snapshot versions, playback of China Unicom IPTV multicast video became choppy. The issue seemed to be resolved when the WAN interface speed was set to 100 Mbps.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

**IGMP Proxy Configuration Parameters:**
config igmpproxy
	option quickleave 1
#	option verbose [0-3](none, minimal[default], more, maximum)

config phyint
	option network wan
	option zone wan
	option direction upstream
	list altnet 0.0.0.0/0

config phyint
	option network lan
	option zone lan
	option direction downstream

**/etc/config/network:**
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
        list ports 'lan4'
	list ports 'lan5'
	option igmp_snooping '1'
config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.2.1'
	option netmask '255.255.255.0'
config device
	option type 'bridge'
	option name 'br-wan'
	list ports 'eth1'
	option igmp_snooping '1'
config interface 'wan'
	option device 'br-wan'
	option proto 'static'
	option ipaddr '192.168.1.6'
	option netmask '255.255.255.0'
	option gateway '192.168.1.1'

Then, open the IPTV multicast source link `udp://225.1.4.73:1102` using a media player on the terminal.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

cocool97, 2025-03-10:
I have the same issue on a Xiaomi AX3000T, needed to downgrade to make it work

ewangsoft, 2025-03-10:
the same issue on a H3C Magic NX30 Pro: (device details: https://github.com/openwrt/openwrt/commit/e78d1a06c8a47be9ea5a306bfb43f3d7ceb89646)

possibly2, 2025-03-13:
Same issue with:
- Linksys WRT1900ACS
- BananaPI R3
- BananaPI R4

Reverted bsck to 23.05.05 for now.

danpawlik, 2025-03-13:
I have same issue but with wifi calling (vowifi). Instead of reverting version, I switch to use software offload. Also it was acceptable, if hardware offload was enabled + qosify. With that combo, establishing connection takes longer comparing to sw offload, but was working.

ewangsoft, 2025-03-18:
> I have same issue but with wifi calling (vowifi). Instead of reverting version, I switch to use software offload. Also it was acceptable, if hardware offload was enabled + qosify. With that combo, establishing connection takes longer comparing to sw offload, but was working.
> 
> Some useful links
> 
> * https://forum.openwrt.org/t/a-new-dual-10g-router-based-on-filogic-880-banana-pi-bpi-r4/163861/709
> * https://forum.openwrt.org/t/a-new-dual-10g-router-based-on-filogic-880-banana-pi-bpi-r4/163861/882
> * https://forum.openwrt.org/t/mt6000-custom-build-with-luci-and-some-optimization-kernel-6-6-x/185241/1715

I referred to the content in the third link to set it up, but the problem still persists.

danpawlik, 2025-03-18:
Wondering, if by removing the 'udp' from https://github.com/openwrt/firewall4/blob/master/root/usr/share/firewall4/templates/ruleset.uc#L138 might help.

ewangsoft, 2025-03-21:
Packet capture reveals that there are packet reordering and packet loss.

ewangsoft, 2025-03-23:
The latest situation is that when the IPTV multicast video bitrate is 5Mbps, the video does not pixelate. However, when the IPTV multicast video bitrate is 10Mbps, the video does pixelate.

mirtouf, 2025-04-15:
Release 24.10.1 has been made available and I am not sure the issue is fixed.
Did someone try with the new release?

cocool97, 2025-04-15:
I took a look at the changelog and did not see anything related to our issue.. I'll try to update later on if nobody already did it

ewangsoft, 2025-04-17:
> Release 24.10.1 has been made available and I am not sure the issue is fixed. Did someone reproduce the issue with the new release?

I have already tested it. The issue still persists under version 24.10.1.

glassd00r, 2025-04-17:
UDP packet reordering when hw flow offload is enabled has been there since i started tinking with bpir4 (mt7988) from the onset.
What I do not know is if the reordering is due to the the hardware flow offloading process or that mtk ppe is reordering the packets.
I can confirm that UDP packets are not reordered when flow offload is disabled or when software flow offload is enable. Only when hardware flow offload is enabled does UDP packet reordering happens.
Thanks to @brada4 and this patch to fw4, I have been able to disable just UDP hw flow offloading.
With his patch [https://github.com/openwrt/firewall4/pull/22](url)
create /etc/nftables.d/20-custom-filter-chains.nft
`chain handle_offload {
  # ct bytes < 1000000 counter accept
  # ct packets < 30 counter accept
  meta l4proto udp counter accept
  # meta length < 100 counter accept
  # ct direction original counter accept
  # the fw4 offload table follows
  # flow add @ft accept
  # accept
}
`
@dangowrt @nbd168 @hauke

brada4, 2025-04-17:
There was similar concern rised w r t gaming and mtk udp offload.

ewangsoft, 2025-04-18:
> UDP packet reordering when hw flow offload is enabled has been there since i started tinking with bpir4 (mt7988) from the onset.
>
> What I do not know is if the reordering is due to the the hardware [...]

> UDP packet reordering when hw flow offload is enabled has been there since i started tinking with bpir4 (mt7988) from the onset.
>
> What I do not know is if the reordering is due to the the hardware [...]

Are you referring to the "hw flow offload" setting in the firewall configuration or the offload at the hardware driver level? If you're talking about the firewall setting, my router doesn't have it enabled.

brada4, 2025-04-18:
Then what acceleration your title refers to?

ewangsoft, 2025-04-19:
> Then what acceleration your title refers to?

I didn't use the offload in the firewall settings. Before raising this issue, I consulted a friend who works on customizing OpenWRT systems. I asked him how I should file an ISSUE, and he suggested that I do it this way. Here, it is suspected that...

brada4, 2025-04-19:
You can always bridge tv or voice vlans to their voip and stb cpe-s without caring for protocols inside.

brada4, 2025-05-24:
There is a known issue that hardware offload sometimes or always re-orders some datagrams around adding offload state.

sashorion, 2025-06-29:
> Try leaving gro but disabling rx-gro-list only?

ethtool -K wan gro on => The problem is showing up.
ethtool -K wan rx-gro-only off => Everything is fine, the problem does not appear.

brada4, 2025-06-29:
For workaround - gro list disablement has minimal performance implications. 

Likely related:
https://github.com/openwrt/openwrt/issues/18608
@nbd168 may have some idea.

brada4, 2025-07-05:
@cocool97 the name of the option is `rx-gro-list`

map-b, 2025-07-05:
Potential fix in snapshot has just been committed 604355e8c4d18dda22aeb618c1489c5369860418

brada4, 2025-07-05:
@DragonBluep could you also cherry pick that for v24.10?

sashorion, 2025-07-08:
> Does anyone know how to have this setting persistent through reboots ? It seems that it is lost when I reboot my device
Just add to /etc/rc.loca(Local Startup in Luci)

sleep 5
ethtool -K wan rx-gro-list off

glassd00r, 2025-11-30:
Resolved this some months back.
https://forum.openwrt.org/t/lets-talk-about-firewall4-default-nftables-firewall/231246/66

ewangsoft, 2025-12-04:
> To summarise the discussion so far:
> 
> This tracker is mainly about multiple users observed issues that turned out to be caused by "rx-gro-list" being "on" by default. "**rx-gro-list**" is explicitly turned on by this OpenWrt patch "**600-net-enable-fraglist-GRO-by-default.patch**". Seems the patch should be removed. Users compiling their own OpenWrt can remove it from your local repo.
> 
> So it's not related to HW offload as the title may mislead people.
> 
> Except for [@glassd00r](https://github.com/glassd00r) who seems to see "out of order" UDP packets when the flow is offloaded to HW on MT7988A. But I think if "**receive side scaling (RSS)**" is working on his build, packets will always be processed in received order. Hence, no shuffling by the HW engine.

When I initially posted this ISSUE, I didn't realize the problem was related to GRO. Could you please suggest a more appropriate description for the ISSUE title? I'll update it accordingly.

> To summarise the discussion so far:
> 
> This tracker is mainly about multiple users observed issues that turned out to be caused by "rx-gro-list" being "on" by default. "**rx-gro-list**" is explicitly turned on by this OpenWrt patch "**600-net-enable-fraglist-GRO-by-default.patch**". Seems the patch should be removed. Users compiling their own OpenWrt can remove it from your local repo.
> 
> So it's not related to HW offload as the title may mislead people.
> 
> Except for [@glassd00r](https://github.com/glassd00r) who seems to see "out of order" UDP packets when the flow is offloaded to HW on MT7988A. But I think if "**receive side scaling (RSS)**" is working on his build, packets will always be processed in received order. Hence, no shuffling by the HW engine.

When I initially posted this ISSUE, I didn't realize the problem was related to GRO. Could you please suggest a more appropriate description for the ISSUE title? I'll update it accordingly.

ewangsoft, 2025-12-04:
> To summarise the discussion so far:
>
> This tracker is mainly about multiple users observed issues that turned out to be caused by "rx-gro-list" being "on" by default. "**rx-gro-list**" is explicitly [...]

When I initially posted this ISSUE, I didn't realize the problem was related to GRO. Could you please suggest a more appropriate description for the ISSUE title? I'll update it accordingly.

bleeee, 2026-10-04:
天津电信也是这样，需要手动对端口进行gro off

brada4, 2026-10-05:
Fill a bug report if gro causes regression, in such case it has to be off by default: as a minimum

kvic-z, 2026-10-05:
> 天津电信也是这样，需要手动对端口进行gro off

Keep gro on by default. AND turn on 'rx-udp-gro-forwarding'. Does it resolve your issue?

yveming, 2026-10-08:
The root cause of the packet-loss artifacts in this tracker is now identified and written up in **#25713**.

Summary: the OpenWrt patch `600-net-enable-fraglist-GRO-by-default.patch` turns on `rx-gro-list` (`NETIF_F_GRO_FRAGLIST`) by default. The fraglist UDP GRO path merges consecutive same-size datagrams of a non-local-socket flow into a single oversized skb (e.g. 2×/3× of ~1328 B → 2656/3984 B), keeping `skb_is_gso()` and the DF flag. When such a skb is forwarded by the IPv4 multicast router (`ipmr`), `ipmr_prepare_xmit()` drops it because `skb->len > dst_mtu && IP_DF` — evaluating the **host-internal GRO batch** size, before egress `ip_finish_output_gso()` could segment it back. Multicast cannot be fragmented and cannot get an ICMP error, so the drop is silent (`FragFails` only). Unicast forwarding is unaffected because it reaches the GSO segmentation path.

This matches the observations in this thread exactly:

- "UDP packets larger than 1316 bytes are dropped between WAN and LAN" — those are the GRO-merged datagrams, larger than the 1500 B egress MTU.
- higher bitrate → worse: more consecutive same-flow packets per GRO cycle, so more merging (5 Mbps fine, 10 Mbps pixelated).
- `gro off` / `rx-gro-list off` fixes it (disables the merge at the source); `rx-gro-list off` is the minimal-impact workaround.
- It is not hardware offload / flowtable related — it reproduces with offload disabled, as several people here confirmed.

Note: enabling `rx-udp-gro-forwarding` will not help — the merged skbs still hit the `ipmr` drop. Suggested fixes (details in #25713): revert/skip patch 600, or (if the forwarding benefit is wanted) land the kernel-side fix first: let `ipmr_prepare_xmit()` not drop GSO skbs (they are segmented into complete MTU-sized datagrams at egress). OpenWrt already has precedent for a companion fix to patch 600: `17f07040e7` ("kernel: prevent tunnel drivers from mangling fraglist GSO packets").