openwrt/openwrt #18219: MT6000 WAN interface hangs and cannot get an IP address

samed, 2025-03-10:
### Describe the bug

Since the commit https://github.com/openwrt/openwrt/commit/d8315d5358d59233d9f1ff8353f1ce6f3428e1e5 eth0/eth1 drivers are crashing in the kernel, WAN is down and cannot get an IP address. During the boot following messages are printed:

```
Sun Mar  9 10:56:36 2025 kern.crit kernel: [   38.009918] mtk_soc_eth 15100000.ethernet eth1: NETDEV WATCHDOG: CPU: 2: transmit queue 1 timed out 10180 ms
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.329922]
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.329922] ==================== FE ====================
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.336708] 0x00000000: 810000b0 00000000 03000020 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.342274] 0x00000010: 00000025 00000000 00001818 000fffff
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.347827] 0x00000020: 21021000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.353384] 0x00000030: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.358937] 0x00000040: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.364494] 0x00000050: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.370052] 0x00000060: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.375605] 0x00000070: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.381161] 0x00000080: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.386715] 0x00000090: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.392271] 0x000000a0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.397824] 0x000000b0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.403381] 0x000000c0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.408934] 0x000000d0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:36 2025 kern.info kernel: [   38.414490] 0x000000e0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.420048] 0x000000f0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.425601] 0x00000100: 01f101ff 01fa01f4 00000300 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.431159] 0x00000110: 00000300 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.436712] 0x00000120: 00000000 8000c406 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.442269] 0x00000130: 00000000 00000000 00000000 0000aa9a
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.447822] 0x00000140: 001a000e 01ff001a 000e01ff 000e000e
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.453382] 0x00000150: 000e000e 000e000e 000e000e 000e000e
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.458936] 0x00000160: 000f000a 001a000f 000f001a 01ff000f
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.464494] 0x00000170: 000f000f 0006000f 00060006 00060006
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.470051] 0x00000180: 00000000 00000000 000e0000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.475605] 0x00000190: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.481161] 0x000001a0: 00000000 0000000e 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.486715] 0x000001b0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.492271] 0x000001c0: 84321a43 321a4359 1a436984 30a98432
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.497825] 0x000001d0: 8a432154 321a6439 8435a984 602143a9
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.503382] 0x000001e0: 5cba4312 77770986 003f0003 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.508936] 0x000001f0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.514496] 0x00000200: 090b9a10 00e89000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.520054] 0x00000210: 0b2f8002 17709630 102f8002 01c0967c
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.525607] 0x00000220: 00000000 00000000 00000000 10930000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.531164] 0x00000230: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.536717] 0x00000240: 01de01f1 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.542275] 0x00000250: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.547828] 0x00000260: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.553384] 0x00000270: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.558936] 0x00000280: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.564493] 0x00000290: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.570052] 0x000002a0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.575605] 0x000002b0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.581162] 0x000002c0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.586716] 0x000002d0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.592279] 0x000002e0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.597835] 0x000002f0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.603394] 0x00000300: 00000000 0a800100 0a800100 00000202
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.608949] 0x00000310: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.614507] 0x00000320: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.620064] 0x00000330: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.625617] 0x00000340: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.631174] 0x00000350: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.636728] 0x00000360: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.642285] 0x00000370: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.647838] 0x00000380: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.653395] 0x00000390: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.658948] 0x000003a0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.664506] 0x000003b0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.670064] 0x000003c0: 00000000 00000000 00000000 00000000
Sun Mar  9 10:56:37 2025 kern.info kernel: [   38.675617] 0x000003d0: 00000000 00000000 00000000 00000000
[... 468 lines cut ...]
### OpenWrt target/subtarget

mediatek/filogic

### Device

MT-6000

### Image kind

Official downloaded image

### Steps to reproduce

Try to flash the recent master snapshots

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Djfe, 2025-03-10:
Please retest with https://github.com/openwrt/openwrt/commit/edddbaf79ccfce9ce80079a467a596ec27a2a128
This issue might've been fixed already.

LGA1150, 2025-03-11:
A fix is pending on the mailing list:
https://lists.infradead.org/pipermail/openwrt-devel/2025-March/043829.html

pesa1234, 2025-03-16:
I didn't get the error. Can you tell us the step to reproduce?

aronsky, 2025-03-17:
> A fix is pending on the mailing list: https://lists.infradead.org/pipermail/openwrt-devel/2025-March/043829.html
> 
> You may workaround it by tuning EEE on eth1: `ethtool --set-eee eth1 eee on tx-lpi on tx-timer 1000`

Thank you!

For those who find this issue before it's fixed, and need some more help, here are a couple of suggestions:

You can get `ethtool` from [here](https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/ethtool-full-6.11-r1.apk). Copy it to the router, and in a shell (on the router) run `apk add --allow-untrusted ethtool-full-6.111-r1.apk`.

It seems that once the crash occurs (you can verify by running `dmesg` as suggested in the issue description), the above solution is helpless. To get around it, uncheck the "Bring up on boot" checkbox on the wan/wan6 interfaces in Luci (or, if using text configuration, add `option auto '0'` to these interfaces in `/etc/config/network`). Reboot the router, and verify that the crash hasn't occurred, and the WAN interfaces are down. Then, restart the interfaces in Luci (or with `ifup wan`, `ifup wan6` from the terminal), and immediately after that run `ethtool --set-eee eth1 eee on tx-lpi on tx-timer 1000`. I'm considering adding that command somewhere in the init scripts so that everything is automated, and I don't have to do it manually when the router restarts, but I haven't found the proper place yet. Any ideas are welcome.

Rygle, 2025-03-18:
Great comment @aronsky !!

> You can get `ethtool` from [here](https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/ethtool-full-6.11-r1.apk). Copy it to the router, and in a shell (on the router) run `apk add --allow-untrusted ethtool-full-6.111-r1.apk`.

OR
- I used firmware-selector to include ethtool-full in a snapshot build without issue.
OR
- As [@Enig123](https://forum.openwrt.org/t/gl-inet-flint-2-gl-mt6000-discussions/173524/2416) also suggests, just plug your WAN uplink into a LAN port - you will have to either set LAN IP manually (probably easier) or set LAN to DHCP client from the upstream box DHCP server, then do a software update and download ethtool-full.

- Then follow the steps that @aronsky mentioned and you should be OK to go.

aronsky, 2025-03-18:
So, after looking around, the quick and dirty fix (after disabling bringing the interface on boot) is to put the following in `/etc/rc.local`:

```bash
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.

ifup wan wan6

i=0
while [ ${i} -le 1000 ]; do
  ethtool --set-eee eth1 eee on tx-lpi on tx-timer 1000
  sleep 0.01
  i=$(( i + 1 ))
done

exit 0
```

I'm not sure executing the `ethtool` command a 1000 times is really required, but it seems that it needs to be run *after* the interface is up, and `ifup` is asynchronous (i.e., it returns before the interface is up). So I decided to run it in a loop like that, to make sure it gets executed on time.

systemcrash, 2025-03-19:
Not certain, but wouldn't a workaround script like that be better placed in the hotplug.d hierarchy?

https://openwrt.org/docs/guide-user/base-system/hotplug

There's `iface` available.

aronsky, 2025-03-19:
Quite possible. I haven't researched the exact order of execution of all the init scripts. For a quick and dirty workaround of simply bringing up the WAN interfaces once booted, and setting the EEE timer immediately afterwards, `rc.local` seems to work fine. If the issue is not fixed in the future and the workaround becomes permanently necessary, then a better location should be identified, for sure.

sfx2000, 2025-03-27:
@LGA1150 @dangowrt - wondering if the practical approach would be to handle this in device tree in the short term?

Watching the ML, there seems to be some movement, however slow...

Digging back thru time - there's @LGA1150 [comment about EEE on MediaTek](https://github.com/openwrt/openwrt/issues/17351)...

sfx2000, 2025-03-29:
Interesting insight from the forum thread...

https://forum.openwrt.org/t/gl-inet-flint-2-gl-mt6000-discussions/173524/2441?u=sfx2000

gist of it is - if uplink is 2.5Gbe, everything is fine, as EEE is not supported. If uplink is 1Gbe, whether EEE is enabled or not, the issue happens.

A potential workaround is to disable EEE for eth0 and eth1 (the 2.5Gbe ports) by adding this to the appropriate resources in the mt7986a-glinet-gl-mt6000.dts file - which localizes the work-around specifically to MT6000

```
eee-broken-100tx:
eee-broken-1000t:
```

sfx2000, 2025-03-29:
> I didn't get the error. Can you tell us the step to reproduce?

See my comment above- if one has 2.5Gbe upstream, the issue does not happen, there's a potential DTS fix if you'd like to try on your community build for MT6000

low risk, and much easier to maintain rather than backing out the entire commit that caused this issue - and since you have commit access, I'm ok with you doing the PR and taking credit for the actual fix

sfx2000, 2025-03-30:
> I think the problem might be with the driver for the rtl8221 PHY

It's not the realtek driver - it's a timing issue at the PHY level. Review the ML discussion

Disabling EEE for eth0/eth1 at device tree as suggested resolves the problem in the short term until folks can figure things out with the switch changes.

Djfe, 2025-03-31:
feel free to test the next snapshot build
https://github.com/openwrt/openwrt/commit/324f234966fe7b910c7f25b69334e54591098c18

sfx2000, 2025-04-01:
> feel free to test the next snapshot build [324f234](https://github.com/openwrt/openwrt/commit/324f234966fe7b910c7f25b69334e54591098c18)

Works now...

Thx for folks for looking into it and supporting a fix.

Djfe, 2025-04-01:
that is great to hear ❤️
the fix had been lying on the mailing list for the last few weeks
@samed feel free to test and then maybe close this, if everything works now 😅