openwrt/openwrt #18222: TP-Link Archer C59 v1 radio0 5GHz not working since 24.10

BugsonPL, 2025-03-11:
### Describe the bug

Since version 24.10 radio0 (QCA9886 802.11ac/n) not working, on 23.05.5 works fine.

### OpenWrt version

24.10.0

### OpenWrt release

https://downloads.openwrt.org/releases/24.10.0/targets/ath79/generic/openwrt-24.10.0-ath79-generic-tplink_archer-c59-v1-squashfs-sysupgrade.bin

### OpenWrt target/subtarget

ath79/generic/openwrt-24.10.0-ath79-generic-tplink_archer-c59-v1

### Device

TP-Link Archer c59 v1

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

BugsonPL, 2025-03-11:
Corrected version relese and target

closed 2025-03-11 by BugsonPL

tuxaddict, 2025-03-20:
I fear i have to report the exact same problem:

I used the file linked on: https://openwrt.org/toh/hwdata/tp-link/tp-link_archer_c59_v1, under:

_Firmware OpenWrt Upgrade URL:_ https://downloads.openwrt.org/releases/24.10.0/targets/ath79/generic/openwrt-24.10.0-ath79-generic-tplink_archer-c59-v1-squashfs-sysupgrade.bin

The checksums where correct. same behavior, radio0 not working, flagged as disabled.
Let me know if i can provide more information.

BugsonPL, 2025-03-20:
I have created proper issue report:
https://github.com/openwrt/openwrt/issues/18223
Please write there