openwrt/openwrt #18223: TP-Link Archer C59 v1 radio0 5GHz not working since 24.10

BugsonPL, 2025-03-11:
### Describe the bug

Since version 24.10 radio0 (QCA9886 802.11ac/n) not working, on 23.05.5 works fine.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer c59 v1

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-03-11:
Attach system log.

BugsonPL, 2025-03-12:
[system.log](https://github.com/user-attachments/files/19206036/system.log)
[kernel.log](https://github.com/user-attachments/files/19206035/kernel.log)

There was issue with 5Ghz radio in the past:
https://forum.openwrt.org/t/archer-c59-v1-missing-ieee802-11-a-n-ac-5ghz/2226/3

cis202, 2025-03-14:
after loading 24.10.0 on a TP-Link Archer C6 v2 EU, coming from 23.05.2 , the 5 GHz Wireless was no longer available.

tuxaddict, 2025-03-21:
As mentioned in #18222 i experience the exact same error. and yes, the kernel log looks like the error worked on in  #14541

TP-Link Archer C59 v1
Qualcomm Atheros QCA956X ver 1 rev 0

Error: Network device is not present (phy0-sta0 5ghz)

`[   42.327838] kmodloader: loading kernel modules from /etc/modules.d/*
[   42.925786] jitterentropy: Initialization failed with host not compliant with requirements: 9
[   43.276261] Loading modules backported from Linux version v6.12.6-0-ge9d65b48ce1a
[   43.284054] Backport generated by backports.git v6.1.110-1-35-g410656ef04d2
[   43.970177] PPP generic driver version 2.4.2
[   43.994863] NET: Registered PF_PPPOX protocol family
[   44.001439] urngd: v1.0.2 started.
[   44.097479] ath10k 6.10 driver, optimized for CT firmware, probing pci device: 0x56.
[   44.154703] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   44.161487] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   47.859448] ath10k_pci 0000:00:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   47.869040] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   47.883503] ath10k_pci 0000:00:00.0: firmware ver 10.4b-ct-9888-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 59e741e7
[   48.232055] **ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=20 from ath10k/QCA9888/hw2.0/board-2.bin**
[   48.750162] ath10k_pci 0000:00:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9888/hw2.0
[   48.759967] ath10k_pci 0000:00:00.0: failed to fetch board file: -12
[   48.783863] ath10k_pci 0000:00:00.0: could not probe fw (-12)`

OpenWrt 23.05.5 r24106-10cc5fcd00 / LuCI openwrt-23.05 branch git-24.264.56413-c7a3562 works fine.

thanks for this great software and your help. (-:

BugsonPL, 2025-04-29:
[24.10.1](https://downloads.openwrt.org/releases/24.10.1/)
Still did not solve the issue

tuxaddict, 2025-04-29:
Same here, atlhough i tested the 24.10.0.
Site: https://openwrt.org/toh/hwdata/tp-link/tp-link_archer_c59_v1
File: https://downloads.openwrt.org/releases/24.10.0/targets/ath79/generic/openwrt-24.10.0-ath79-generic-tplink_archer-c59-v1-squashfs-sysupgrade.bin

Same behavior as before: Network device is not present (phy0-sta0 5ghz)
Same error in the logs (although i have to point out that this errors occur with the working 23.05 as well!)

`[   47.766491] ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=20 from ath10k/QCA9888/hw2.0/board-2.bin
[   48.282727] ath10k_pci 0000:00:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9888/hw2.0
[   48.292512] ath10k_pci 0000:00:00.0: failed to fetch board file: -12`

tuxaddict, 2025-05-01:
Sorry, i thought a newer version was alreday officially provided and didn't realize that the 24.10 was already tested by myself - can't delete the post anymore.

@BugsonPL may i ask how you tested the linked packages? there isn't a "upgrade" package to directly load from OpenWRTs Luci, right? Did you use this method:
https://github.com/openwrt/openwrt/issues/14541#issuecomment-2812214184

best, me (-:

tuxaddict, 2025-05-01:
> [24.10.1](https://downloads.openwrt.org/releases/24.10.1/) Still did not solve the issue

@BugsonPL those packages do work! you need to adjust caldata as well, here's a link to the fix/workaround:
https://github.com/openwrt/openwrt/issues/14541#issuecomment-2845285606