openwrt/openwrt #18228: 24.10-on fails to boot on Zimaboard

ab0oo, 2025-03-12:
### Describe the bug

The Zimaboard 8/32 is incapable of booting the kernel included with the 24.10.0 release for x86/64 and all subsequent snapshot.  Kernel hangs during boot for 60 seconds, then dumps more stacktrace to the screen than I can capture.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

x86/64

### Device

Icewhale Zimaboard

### Image kind

Official downloaded image

### Steps to reproduce

Download current x86/64 release image, use dd to flash to USB drive.
Insert USB drive into USB port on Zimaboard
Powerup, press F11 to bring up boot device menu, select USB drive and hit enter.
GRUB will appear, select either OpenWRT or OpenWRT(Failsafe), hit Enter
Observe...

### Actual behaviour

The device will load the kernel into 0x000000...0080, and begin the boot sequence.
Boot begins to initialize PCI devices...
[Pre-fail boot image](https://pasteboard.co/wV1ZAXBb7kje.jpg)

Boot will hang for exactly 60 seconds, then dump a stacktrace into the kernel.  only the last ~25 lines are available.
[Post fail screen image](https://pasteboard.co/Y49lxL8EAd35.jpg)
IT looks like a watchdog of some sort fires at the 60 second mark, and again at the 240 second and 480 second marks.

### Expected behaviour

Expect the Zimaboard to boot cleanly into a working OS.  This is the behavior I get with the versions prior to 24.10:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r23323-6892603efa'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r23323-6892603efa'

### Additional info

I'm open to ideas on how to catch where the actual failure is.  There is no serial console available on the Zimaboard, and even the slo-mo camera feature can't capture the screen output fast enough.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ikazuhiro, 2025-04-19:
My x86-64 device also fails to boot kernel 6.6 (OpenWrt 24.10).  It has no problem with kernel 5.15 (OpenWrt 23.05) and 6.1.  Though I don't know the root problem, enabling PCI_MMCONFIG resolves the issue on my device.

```
--- a/target/linux/x86/64/config-6.6
+++ b/target/linux/x86/64/config-6.6
@@ -345,6 +345,7 @@ CONFIG_MFD_CORE=y
 # CONFIG_MFD_INTEL_LPSS_ACPI is not set
 # CONFIG_MFD_INTEL_PMC_BXT is not set
 CONFIG_MMC=y
+CONFIG_MMCONF_FAM10H=y
 CONFIG_MMC_BLOCK=y
 CONFIG_MMC_CQHCI=y
 CONFIG_MMC_RICOH_MMC=y
@@ -404,7 +405,7 @@ CONFIG_PCIEPORTBUS=y
 CONFIG_PCIE_PME=y
 CONFIG_PCI_HYPERV=y
 CONFIG_PCI_HYPERV_INTERFACE=y
-# CONFIG_PCI_MMCONFIG is not set
+CONFIG_PCI_MMCONFIG=y
 CONFIG_PCI_PASID=y
 CONFIG_PCI_PRI=y
 CONFIG_PCI_XEN=y
```

WROIATE, 2025-04-20:
I got the same problem on my device of Intel cpu N3450
23.05 is well, 24.10 hangs

closed 2025-06-29 by openwrt-bot

ab0oo, 2025-06-29:
Generally speaking, projects usually mark an issue RESOLVED pending confirmation from the original bug filer, do they not?  I'll test this today, but I would have at least liked the courtesy of getting to comment "it works" or "it doesn't" for my particular case, since I'm the original author of the bug.
Regardless, thank you for posting the PC_MMCONFIG fix.  I'm hoping it works precisely as advertised.