openwrt/openwrt #18229: Some packages like i2c-tools and coremark cannot rewrite build_dir cache automatically

Beaverfffan, 2025-03-12:
### Describe the bug

Some packages like i2c-tools and coremark cannot rewrite build_dir cache,needs to be deleted manually to continue making in the second time.

Successfully built smbus-1.1-cp311-cp311-linux_aarch64.whl
cd "/home/beaver/Desktop/xwrt2/build_dir/target-aarch64_cortex-a53_musl/i2c-tools-4.3" && CC="aarch64-openwrt-linux-musl-gcc" CCSHARED="aarch64-openwrt-linux-musl-gcc -DPIC -fPIC" CXX="aarch64-openwrt-linux-musl-g++" LD="aarch64-openwrt-linux-musl-gcc" LDSHARED="aarch64-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/beaver/Desktop/xwrt2/build_dir/target-aarch64_cortex-a53_musl/i2c-tools-4.3=i2c-tools-4.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/home/beaver/Desktop/xwrt2/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/include -I/home/beaver/Desktop/xwrt2/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include -I/home/beaver/Desktop/xwrt2/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include/fortify -I/home/beaver/Desktop/xwrt2/staging_dir/target-aarch64_cortex-a53_musl/usr/include/python3.11" LDFLAGS="-L/home/beaver/Desktop/xwrt2/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/lib -L/home/beaver/Desktop/xwrt2/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro -lpython3.11" _PYTHON_HOST_PLATFORM="linux-aarch64" _PYTHON_SYSCONFIGDATA_NAME="_sysconfigdata__linux_aarch64-linux-musl" PYTHONPATH="/home/beaver/Desktop/xwrt2/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/python3.11:/home/beaver/Desktop/xwrt2/staging_dir/target-aarch64_cortex-a53_musl//usr/lib/python3.11/site-packages:/home/beaver/Desktop/xwrt2/build_dir/target-aarch64_cortex-a53_musl/i2c-tools-4.3/ipkg-install//usr/lib/python3.11/site-packages" PYTHONDONTWRITEBYTECODE=1 _python_sysroot="/home/beaver/Desktop/xwrt2/staging_dir/target-aarch64_cortex-a53_musl" _python_prefix="/usr" _python_exec_prefix="/usr"  CARGO_BUILD_TARGET=aarch64-unknown-linux-musl CARGO_HOME=/home/beaver/Desktop/xwrt2/dl/cargo CARGO_PROFILE_RELEASE_CODEGEN_UNITS=1 CARGO_PROFILE_RELEASE_DEBUG=false CARGO_PROFILE_RELEASE_DEBUG_ASSERTIONS=false CARGO_PROFILE_RELEASE_LTO=true CARGO_PROFILE_RELEASE_OPT_LEVEL=z CARGO_PROFILE_RELEASE_OVERFLOW_CHECKS=true CARGO_PROFILE_RELEASE_PANIC=unwind CARGO_PROFILE_RELEASE_RPATH=false CARGO_TARGET_AARCH64_UNKNOWN_LINUX_MUSL_LINKER=aarch64-openwrt-linux-musl-gcc RUSTFLAGS="-Ctarget-feature=-crt-static -lssp_nonshared -Clink-arg=-fuse-ld=bfd" TARGET_CC=aarch64-openwrt-linux-musl-gcc TARGET_CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/beaver/Desktop/xwrt2/build_dir/target-aarch64_cortex-a53_musl/i2c-tools-4.3=i2c-tools-4.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -mno-outline-atomics" PYO3_CROSS_LIB_DIR="/home/beaver/Desktop/xwrt2/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/python3.11" SETUPTOOLS_RUST_CARGO_PROFILE="release"    /home/beaver/Desktop/xwrt2/staging_dir/hostpkg/bin/python3.11  -m installer --destdir "/home/beaver/Desktop/xwrt2/build_dir/target-aarch64_cortex-a53_musl/i2c-tools-4.3/ipkg-install" --no-compile-bytecode --prefix /usr "/home/beaver/Desktop/xwrt2/build_dir/target-aarch64_cortex-a53_musl/i2c-tools-4.3/py-smbus"/openwrt-build/smbus-1.1-*.whl 
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/beaver/Desktop/xwrt2/staging_dir/hostpkg/lib/python3.11/site-packages/installer/__main__.py", line 98, in <module>
    _main(sys.argv[1:], "python -m installer")
  File "/home/beaver/Desktop/xwrt2/staging_dir/hostpkg/lib/python3.11/site-packages/installer/__main__.py", line 94, in _main
    installer.install(source, destination, {})
  File "/home/beaver/Desktop/xwrt2/staging_dir/hostpkg/lib/python3.11/site-packages/installer/_core.py", line 109, in install
    record = destination.write_file(
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beaver/Desktop/xwrt2/staging_dir/hostpkg/lib/python3.11/site-packages/installer/destinations.py", line 207, in write_file
    return self.write_to_fs(scheme, path_, stream, is_executable)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/beaver/Desktop/xwrt2/staging_dir/hostpkg/lib/python3.11/site-packages/installer/destinations.py", line 167, in write_to_fs
    raise FileExistsError(message)
FileExistsError: File already exists: /home/beaver/Desktop/xwrt2/build_dir/target-aarch64_cortex-a53_musl/i2c-tools-4.3/ipkg-install/usr/lib/python3.11/site-packages/smbus.cpython-311-aarch64-linux-musl.so
make[3]: *** [Makefile:116: /home/beaver/Desktop/xwrt2/build_dir/target-aarch64_cortex-a53_musl/i2c-tools-4.3/.built] Error 1
make[3]: Leaving directory '/home/beaver/Desktop/xwrt2/feeds/packages/utils/i2c-tools'
time: package/feeds/packages/i2c-tools/compile#1.99#0.59#2.97
    ERROR: package/feeds/packages/i2c-tools failed to build.
make[2]: *** [package/Makefile:189: package/feeds/packages/i2c-tools/compile] Error 1

### OpenWrt version

master

### OpenWrt release

master

### OpenWrt target/subtarget

MT798x

### Device

MT798x

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

abajk, 2025-03-14:
https://github.com/openwrt/packages/pull/25569

abajk, 2025-03-14:
Does this fix this issue?

abajk, 2025-07-05:
https://github.com/openwrt/packages/pull/25917